cocotb/cocotb #1947: Cocotb-verilator integration

liza-zoubakina, 2020-06-26:
Hi, I'm currently working on an integration of cocotb and verilator to run simulations.
I am working with Verilator 4.016 and latest version of Cocotb.

I encountered the following issues where:
- Cocotb is having trouble reading certain parameter values from verilator (I think?). For example, my parameters are initialized in the following format in the top level file:
`module example #( parameter EXAMPLE = 1, parameter y = 2 ) ...`
- The simulations, which previously ran on icarus and vcs with success, no longer run as they cite an `AttributeError` where
`AttributeError: top_level_tile contains no object named EXAMPLE`

Is cocotb fully compatible with verilator? And if so, are there certain limitations to parameters or could this be the result of other issues in the cocotb-verilator integration?

- As a way to bypass the issue, I declared the parameters as vectors instead. Such as: `wire [31:0] EXAMPLE = 1;`
- However, this led to additional issues where cocotb began reading the value of `EXAMPLE` as zero instead of 1. 
- I used gtkwave to read the dumped waveforms of the simulation, however the wire signals are read as 1. This led me to suspect that cocotb is not properly reading the registers that are initialized in verilator.

- Finally, through backtracking the bug, I suspect that the GPI interface between cocotb and verilator is likely not sending the correct values of the registers. 
- I eventually ended up in `SimHandle` in `handle.py`, uses the GPI handle to the simulator object to return the object in cocotb.
- Type of GPI handles can be seen in `gpi.h` and the type of handle for `EXAMPLE` was found to be a register. `simulator.REG = 5`
- I found through print statements that the `SimHandle` function still returns 0 for the `EXAMPLE` object.

I am not sure as to why cocotb has trouble reading wire signals from verilator. My second question would be how are signals being read and how are functions being called in the VPI/GPI files? This is because I am not very familiar with this low-level style of coding, so any advice or pointers would be greatly appreciated. Thanks!

elgorwi, 2020-06-26:
Support for accessing parameters through VPI was added very recently to Verilator. I recommend to upgrade to the latest Verilator version from git (v4.037 devel) and compile/install manually or alternatively wait until the next v4.038) Verilator package release.

liza-zoubakina, 2020-06-26:
Thanks a lot! Will try it today. Thank you for your quick response.

regis-c, 2020-06-28:
Hi.
I give another try to access parameter in verilog from cocotb testbench.
I am now in  Running on Verilator version 4.037 devel, Running tests with cocotb v1.4.0rc1.
It seems to still fail. Looks like PR #1901 merge is still pending and needed (not sure ?, probably the changes to  cocotb/share/lib/vpi/...).
When checking on mean example (and bypassing other work in progress issue in #1900 ), I get the following error when accessing parameter BUS_WIDTH:
```
Test Failed: mean_basic_test (result was AttributeError)
Traceback (most recent call last):
File "/home/regisc/cocotb/examples/mean/tests/test_mean.py", line 63, in mean_basic_test
BUS_WIDTH = int(dut.BUS_WIDTH.value)
File "/usr/local/lib/python3.6/dist-packages/cocotb/handle.py", line 282, in __getattr__
raise AttributeError("%s contains no object named %s" % (self._name, name))
AttributeError: mean contains no object named BUS_WIDTH
```
So I guess we have to keep an eye on #1901 and wait for it to be merged before further tests.

liza-zoubakina, 2020-06-29:
Ok, interesting thanks for the update @regis-c . I crossposted the issue to verilator (https://github.com/verilator/verilator/issues/2447) as well and learned that by declaring `parameter EXAMPLE /*verilator public_flat_rd*/ = 1` , this explicitly made the parameter public so as to make them visible by VPI. My parameters are now being properly initialized. 
Will anticipate the #1901 merge

regis-c, 2020-06-29:
Thanks @liza-zoubakina.  
I like the trick for ```/*verilator public_flat_rd*/```.  
Works fine and simplify a bit my bench. Now happy with that extra comment.
(Should you be curious, here is example of use in my last commit https://gitlab.com/regisc/pyrift/-/commit/159798f79db57aaa50683208131501681d77cd71)

ktbarrett, 2020-07-14:
Seems like this issue is now resolved. If not, please reopen.

closed 2020-07-14 by ktbarrett