cocotb/cocotb #1950: Prevent release process from affecting active development

ktbarrett, 2020-07-01:
The current release has more or less paused active development of cocotb for over a month. We stopped bringing in large changes a while ago. Since the release candidate, we have tried to not merge anything else. Since we do not set goals and are willing to break interfaces from release to release, releases are arbitrary. Thus, there is no motivation to pause development, it is simply an effect of our release process.

Instead of doing the release candidate process and releasing on master, we should do that on a version branch. Minimal effort would need to be done to backport bugfixes to the version branch during the release candidate process, but this way we wouldn't need to halt development while releasing occurs. The release process can then take as long as necessary, which should reduce stress on the maintainers that have the authority to do it.

The below flow just makes the steps more explicit:

1. When the decision is made to do a release, pick a recent commit on master (it does not have to be the most recent commit either) to start the release process.
2. A version branch for the to-be-released version is created at the chosen commit.  The branch name should follow the current standard for release branch names (`stable/X.Y`).
3. A commit is added to the version branch updating the version to `X.Y.0rc1-dev0`. Simultaneously, update the version number on the master branch to `X.(Y+1).0-dev0`.
4. The release candidate is tested. Bugfixes or last minutes features can be added, and should be duplicated on the version branch as well as master.
5. Once the release candidate is seemingly stable, an additional commit is added to the version branch to update the version to `X.Y.0rc1`. That commit should be tagged with the release candidate version number.
6. Follow the existing process for publishing release candidate.
7. If additional changes must be made, first add a commit to the version branch that bumps the version to `X.Y.0rc2-dev0`. Otherwise skip to step 9.
8. Repeat this process until the release candidate is deemed acceptable for release.
9. Create another commit on the version branch bumping the version to `X.Y.0`, and tag the commit with the version number.
10. Follow the existing process for publishing final releases.

See [PEP440](https://www.python.org/dev/peps/pep-0440/) for details on version naming.

ktbarrett, 2020-07-01:
Obviously, we should see 1.4 through on the current process since we are already in the thick of it.

imphil, 2020-07-07:
The current approach is by design. We have only very limited maintainer and contributor resources available, which we need to push a release out the door. If we split these resources between "polishing a release" and "doing work on the upcoming release" we risk never getting a release out.

Issue https://github.com/cocotb/cocotb/issues/1948 is a good example: it was one of a handful of tasks left for a 1.4 release, yet nobody found time to implement it for 11 days. That's not to blame anyone: nobody on cocotb is doing it as a day job, we try our best to fit cocotb in between other things. 

Two more things to note:
- We're not the only ones with this approach, with probably the Linux kernel being a very prominent example of a variation of such a development model.
- This model does not prevent contributors and maintainers from writing code, opening issues, creating or reviewing pull requests. We just create an incentive to work on stabilization over feature work, we don't prevent it.

ktbarrett, 2022-05-10:
This isn't really a problem anymore considering the current rate of development.

closed 2022-05-10 by ktbarrett