cocotb/cocotb #1951: Function for comparing simulator version strings?

cmarqu, 2020-07-06:
Icarus Verilog master, reporting ``SIM_VERSION`` "11.0 (devel)", now supports bit selects (see https://github.com/steveicarus/iverilog/issues/323).
Thus, we need a way to selectively expect an error in our tests based on the version.

While I could compare with either the "11.0" string part, or the "10." string of the released version (and assume nobody uses a version before 10), or treat the number part as a float etc., this gets all ugly for the general case.
Since we have to support more such older/newer comparisons in the future (e.g. for upcoming GHDL improvements), does it make sense to provide a version comparison function that would gradually be extended to handle more simulators as needed?

ktbarrett, 2020-07-07:
It is a good idea to encapsulate this into a function, but it should probably be multiple functions, one per simulator, and another function to map the SIM_NAME to a simulator. But how would the version comparison function work? Would it "normalize" a version? Would you feed it a partial version for it to match? Would it support version comparison?

eric-wieser, 2020-07-07:
Can we use `distutils.version.LooseVersion` here?

cmarqu, 2020-07-07:
Indeed, http://epydoc.sourceforge.net/stdlib/distutils.version.LooseVersion-class.html looks like a good start, we probably don't even need to encapsulate this further for now, only when pathological cases show up.

ktbarrett, 2020-07-07:
> we probably don't even need to encapsulate this further for now, only when pathological cases show up

I'd err on the side of caution here. We have a chance to prevent a large refactor in the future. Do all SIM_VERSION strings for all simulators start with the main version? Do some of them contain preambles? How does `LooseVersion` handle version strings like Questa's where they started using both versioning schemes before switching from semver to calver (e.g. `10.7b 2018.06`)? It's a simple as making otherwise empty aliases for now:

```python
class IcarusVersion(LooseVersion):
    pass
```

I think at some point it would be nice to co-locate all the simulator-specific information used by cocotb, and this might be good place to start.

cmarqu, 2020-07-10:
Interestingly, ``LooseVersion`` doesn't like Aldec Active-HDL version numbers (taken from https://github.com/VUnit/vunit/pull/585/files#diff-8e7169ad9a827f7ebe5cb68193d01c63R423-R437):
```
Failed example:
    ActivehdlVersion("10.5a.12.6914") > ActivehdlVersion("10.5.216.6767")
Exception raised:
    Traceback (most recent call last):
      File "/tools/anaconda/2019.10-python3.7/Linux-x86_64/lib/python3.7/doctest.py", line 1329, in __run
        compileflags, 1), test.globs)
      File "<doctest __main__.ActivehdlVersion[0]>", line 1, in <module>
        ActivehdlVersion("10.5a.12.6914") > ActivehdlVersion("10.5.216.6767")
      File "/tools/anaconda/2019.10-python3.7/Linux-x86_64/lib/python3.7/distutils/version.py", line 64, in __gt__
        c = self._cmp(other)
      File "/tools/anaconda/2019.10-python3.7/Linux-x86_64/lib/python3.7/distutils/version.py", line 337, in _cmp
        if self.version < other.version:
    TypeError: '<' not supported between instances of 'str' and 'int'
```
This bug is still present in Python 3.9-dev: https://travis-ci.org/github/cocotb/cocotb/jobs/707016738

eric-wieser, 2020-07-10:
That's unfortunate, thanks for trying it.

ktbarrett, 2020-07-10:
We could give `LooseVersion` some help here, that's why I suggested wrapper classes. But I'm not quite sure I understand the versioning system Active-HDL uses. The example you posted makes no sense to me.

cmarqu, 2020-07-10:
This may be fixed in https://github.com/python/cpython/pull/14952, but that doesn't help us much.

cmarqu, 2020-07-10:
> The example you posted makes no sense to me.

It could be a synthetic example that doesn't occur in the wild.

cmarqu, 2020-07-10:
https://github.com/python/cpython/blob/master/Lib/distutils/version.py#L260-L262 - ha!

ktbarrett, 2020-07-10:
Inheriting from `Version` might still be the right answer. We can embed knowledge about the versioning in a particular simulator to reject invalid version numbers.

cmarqu, 2020-07-10:
I guess I'll make this not a doctest but a regular docstring with a note.
With luck, if we ever need this, we won't have problematic version numbers.

PR #1987 (2020-12-02): Provide <SimName>Version classes to compare simulator versions
Currently, these are directly using `distutils.version.LooseVersion`.
Intended use case is for setting `expect_errors` for testcases.

Closes #1951
commit 38f199c8 (2020-12-02): Provide <SimName>Version classes to compare simulator versions (#1987)

Provide `<SimName>Version` classes to compare simulator versions

Currently, these are directly using `distutils.version.LooseVersion`.
The intended use case is for setting `expect_errors` for testcases.

Closes #1951
closed 2020-12-02 by cmarqu