cocotb/cocotb #1954: AttributeError: 'HierarchyObject' object has no attribute

Divyanirankari, 2020-07-07:
```python
class InputMonitor(BusMonitor):
    """Observes inputs of DUT."""
    _signals = ["_start_sign","_start_exponent","_start_mantissa","_start_convert_unsigned","_start_convert_long","_start_rounding_mode", "_start_flags","_start_issp","EN__start","EN_flush"]

    def __init__(self, dut, callback=None, event=None):
        BusMonitor.__init__(self, dut, None, dut.CLK, dut.RST_N,
                            callback=callback, event=event)
        self.name = "in"

```

```
cocotb/cocotb/handle.py", line 147, in __getattr__
                                                                                                                         return object.__getattribute__(self, name)
                                                                                                                     AttributeError: 'HierarchyObject' object has no attribute '_start_sign'

```

ktbarrett, 2020-07-07:
The documentation for this has not yet been merged AFAIK. Any subhandle name that starts with an underscore is assumed to be a method on handle and not a subhandle. Basically, accessing subhandles that start with underscores is not supported right now. 

In the mean time a quick hack is to call `handle._HierarchyObject__get_sub_handle_by_name(name)`, but obviously this won't work if you are using a `BusMonitor`. You'll have to rename your signals or make a special version of `BusMonitor`. Sorry, this will be fixed in the future.

ktbarrett, 2020-07-07:
Will be fixed following #1955.

Divyanirankari, 2020-07-07:
okay thanks

ktbarrett, 2020-07-07:
Sorry I may have lied a bit, once again it's not documented, but there is a the `_id` method on handles. But still this doesn't work on `BusMonitor`s.

Divyanirankari, 2020-07-07:
i have been tried with directed tests without driver n monitors still showing same error.

```python
receive = dut._start.value
```

cmarqu, 2020-07-07:
Related: #1315, #1639

Divyanirankari, 2020-07-07:
i have created top verilog module and changed signal name removed starting "_"  in top module and testbench now its running.

closed 2020-07-08 by Divyanirankari