cocotb/cocotb #1955: Support index syntax for accessing subhandles

ktbarrett, 2020-07-07:
Using dot syntax to access subhandles in cocotb is "nice", but it results in a name collision between subhandles and methods and attributes on the handles. This has been a known issue for some time (#134). The approach suggested in #134 is a bad one. A better solution is to simply segregate the namespaces. This can be done by moving the subhandle access logic into `__getitem__`. There would then be 2 different syntaxes where previously there was 1. 

* dot syntax for methods and attributes
* index syntax for subhandles (both by name and by numeric index). 

We can even continuing support dot syntax and document that dot syntax is a nicety that won't always work. Which can be easily accomplished by having `__getattr__` alias `__getitem__`.

I feel like this is a low risk change and will improve usability. It may also provide justification for the removal of the casts on handles, which have issues as well (#1222).

eric-wieser, 2020-07-07:
It seems a bit weird (and JavaScript-y) to me to have dot and index notation act as aliases.

Note that we already have `handle._id("subhandle", extended=False)`. The second argument is rather unfortunate, but using a method avoids spending our limited syntax budget on corner cases.

eric-wieser, 2020-07-07:
> but it results in a namespace collision between subhandles and methods and attributes on the handles.

Note that this problem also occurs in the standard library `namedtuple`, where the same approach of prefixing public methods with underscores is used.

ktbarrett, 2020-07-07:
> It seems a bit weird (and JavaScript-y) to me to have dot and index notation act as aliases.

I only mentioned it to placate those worried about backwards-compatibility. Personally I don't mind a 180°, version numbers exist for a reason.

> Note that we already have `handle._id("subhandle", extended=False)`.

So `handle._id("_A")._id("value").value` is preferable to `handle["_A"]["value"].value`?

> Note that this problem also occurs in the standard library namedtuple, where the same approach that we use today of prefixing public methods with underscores is used.

I think this is a false equivalence. In Python private methods are prefixed with underscores, which allows the users unfettered use of the public namespace when creating their namedtuple. There are no such requirements or conventions with HDLs AFAIK, and we are mixing a Python namespace and an HDL namespace, so there is overlap. Even dunder methods overlap as they are valid identifiers.

Adopting the original suggestion makes all possible issues and design considerations moot. There is no need to levy requirements on users or play with private-but-not-really methods. Decreasing code entropy rather than increasing it is beneficial IMO.

eric-wieser, 2020-07-07:
Apologies, I read this as an _additional_ syntax for subhandles, not a _replacement_ syntax. The JavaScript comparison becomes moot for the latter.