cocotb/cocotb #1958: _def_name and _def_file mostly empty

cmarqu, 2020-07-08:
I added ``_def_name`` and ``_def_file`` to ``test_discovery`` to see what it actually contains, and I found that only for the ``HierarchyArrayObject`` and one of the ``HierarchyObject``s the strings are not both empty: 
```
[...]
     0.00ns INFO     cocotb.sample_module                  test_discovery.py:44   in _discover                       Found name=test_and_gate def_name=and def_file= type=(<class 'cocotb.handle.HierarchyObject'>)
[...]
     0.00ns INFO     cocotb.sample_module                  test_discovery.py:44   in _discover                       Found name=stream_in_string def_name= def_file= type=(<class 'cocotb.handle.HierarchyObject'>)
     0.00ns INFO     cocotb.sample_module                  test_discovery.py:44   in _discover                       Found name=genblk1 def_name=sample_module def_file=/home/colin/tools/cocotb/tests/designs/sample_module/sample_module.sv type=(<class 'cocotb.handle.HierarchyArrayObject'>)
     0.00ns INFO     cocotb.sample_module                  test_discovery.py:44   in _discover                       Found name=genblk1[0] def_name= def_file= type=(<class 'cocotb.handle.HierarchyObject'>)
[...]
```
I'm not sure if this is intended or a regression.

garmin-mjames, 2020-07-21:
For VPI, this is because the "definition name" `vpiDefName` property is defined for:
- instance objects (package, interface, program, module)
- ref object (see 37.16 of IEEE 1800-2012 SystemVerilog LRM)
- interface typespec
- primitives (gate, swithc, udp)
- udp definition

It shows up for `HierarchyArrayObject`s because they use the parent module handle.

"definition location" `vpiDefFile` is defined for instance objects and attributes.

cmarqu, 2020-07-21:
Thanks for digging this up, something to put in the docstring.

cmarqu, 2020-09-23:
For VHPI, cocotb is using ``vhpiNameP`` (https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L244) which is the "name of the designated object in unspecified case for basic identifiers (VHDL is case insensitive) or case preserved for extended identifiers.", and ``vhpiFileNameP``.

closed 2020-09-24 by cmarqu

garmin-mjames, 2021-01-19:
For **VPI**, there are also these properties:

> **37.3.3 Object file and line properties**
> Most objects have the following two location properties, which are not shown in the data model diagrams:
>
> -> location
> _int: vpiLineNo_
> _str: vpiFile_

Which may be more useful for objects that are not instance objects.

PR #2077 (2020-09-24): Detail content of _def_name and _def_file
Closes #1958
commit cbe52238 (2020-09-24): Detail content of _def_name and _def_file

Closes #1958