cocotb/cocotb #1960: VCS crash 10ns clock period

Alex-Mann, 2020-07-09:
I'm experiencing an issue where VCS is crashing on a 10ns clock period but is OK when I change the clock period to 6ns. Any idea why this would be happenning?

 Example to reproduce:

**cocotb_bug.sv**
```
module cocotb_bug(
  input          bu_ref_n,
  input          bu_ref_p,
  input          irst_por_ba,
  input          irst_apb4_presetn,
  input          reset
);

logic counter;

always @(posedge bu_ref_p) begin
  if (reset) begin
    counter <= 0;
  end else begin
    counter <= 1;
  end
end

endmodule
```

**test_cocotb_bug.py**
```
@cocotb.test(skip = False)
async def test_mss(dut):

    await Timer(1, units='ns')
    dut.irst_por_ba <= 1
    dut.irst_apb4_presetn <= 1

    cocotb.fork(Clock(dut.bu_ref_p, period=10, units='ns').start(start_high=True)) # this causes the failure, if I change period=6 there is no issue

    await Timer(100, units='ns')
```

ktbarrett, 2020-07-09:
I confirmed it does not crash on Icarus.

Could you upload the output when it crashes?

Alex-Mann, 2020-07-10:
Strangely enough, I can't seem to reproduce this issue now. I am closing but will reopen with more details if I can reproduce in the future.

closed 2020-07-10 by Alex-Mann