cocotb/cocotb #1962: Increase code coverage

ktbarrett, 2020-07-10:
There are a few reasons why coverage is low; the reasons vary. This issue exists to list and track the completion of tasks relating to increasing code coverage. I think the next realistic goal is to take coverage above 80%.

## Dead Code
Coverage is 0% on dead code... Some code is "public" and must go through a deprecation cycle, but the reset can be removed.
- xunit_reporter.py
  - [ ] update_testsuite
  - [ ] update_testsuites
  - [ ] add_log
  - [ ] add_failure
  - [ ] The entire `File` class 
- utils.py
  - [ ] reject_remaining_kwargs
  - [ ] get_python_integer_types
  - [ ] main script part

## Untested, potential code baggage

This stuff isn't tested, likely because it's not a very useful feature and/or undocumented.
- utils.py
  - [ ] pack
  - [ ] unpack

garmin-mjames, 2020-09-18:
Coverage of the C++ code should be achievable with `gcov`, as pointed out by @cmarqu on gitter.
I got something rudimentary using `python setup.py develop` with conda/GCC, using these extra args for compiling and linking:
`["--coverage", "-fprofile-arcs", "-ftest-coverage"]`.

The libraries are built in place so the `*.gcno` and `*.gcda` files show up in the repo, and after running some of the tests I was able to generate some `*.gcov` files with coverage results.

I'm not sure how any of this would translate to our CI builds though. codecov supports C++11: https://docs.codecov.io/docs/supported-languages

ktbarrett, 2020-09-18:
Use lcov. Youll probably need to set up includes/excludes to prevent coverage of external code. Codecov understands lcovs output IIRC.

cmarqu, 2020-09-18:
Out-of-repo seems to be possible with https://gcc.gnu.org/onlinedocs/gcc/Cross-profiling.html

garmin-mjames, 2020-09-18:
@cmarqu 
> You must move the data files to the expected directory tree in order to use ... the gcov tool. 
Doesn't this mean we need a copy of the source anyways?

ktbarrett, 2020-09-19:
```command
CFLAGS="-O0 -g --coverage" LDFLAGS="--coverage" pip install -e .
make
lcov -c -d . --no-external -o cov.info
genhtml cov.info -o coverage
xdg-open coverage/inde.html
```

But VPI coverage isn't being recorded correctly. It might be the object renaming, I'm not sure how gcov works.

cmarqu, 2020-09-29:
The C++ coverage setup is now tracked in #2088.

ktbarrett, 2021-07-22:
Hooks were removed: #2508
The testbenching components were removed: #2289
`_initialize_testbench` is now covered as well as it can be: #2400

ktbarrett, 2024-12-22:
`utils.pack` and `utils.unpack` were removed. `BinaryValue` was removed. `cocotb.memdebug` was removed.

I'm going to mark this as fixed. Another issue can be opened and kept more up to date if need be.

closed 2024-12-22 by ktbarrett