cocotb/cocotb #1963: Kill a coroutine along with all the child coroutines that it has forked (feature request)

benreynwar, 2020-07-10:
I find that it is often useful to kill a coroutine and also at the same time kill all the coroutines that it has forked.

A typical example is if I have a testing coroutine that takes care of forking all the stimulus coroutines and checking coroutines.
In the main test function I fork the clock, send a reset signal and then fork the testing coroutine.  The main test then waits for some arbitrary number of clock cycles before resetting again, killing the testing coroutine and all it's child coroutines, and then forking another testing coroutine.  In this way I can test that the reset is working well interrupting the module at an arbitrary state. 

```python
async def do_test(dut):
    fork(send_stimulus(dut))
    await check_output(dut)

@cocotb.test(): 
def top_level_test(dut)   
    seed = 0
    n_resets = 10
    p_reset = 0.001
    do_test_task = None
    rnd = random.Random(seed)
    fork(start_clock(dut))
    for reset_index in range(n_resets):
        await triggers.RisingEdge(dut.clk)
        if do_test_task is not None:
             do_test_task.kill()
        dut.reset <= 1
        await triggers.RisingEdge(dut.clk)
        dut.reset <= 0
        do_test_task = fork(do_test(dut))
        while True:
             if rnd.random() < p_reset:
                 break
             await RisingEdge(dut.clk)
````

At the moment can't just kill the coroutine, rather I also need to manually kill all the children that the coroutine has created with `fork` or `await`.  Currently I'm taking care of this with a hackish solution (https://github.com/benreynwar/slvcodec/blob/master/slvcodec/cocotb_helper.py), but it would be really nice if this was something that the scheduler could take care of itself.  I'm also curious about how other people are solving these kinds of issues.

eric-wieser, 2020-07-10:
In asyncio this would be achieved with `try/finally` blocks to clean up after the cancellation error. Throwing a similar error I'm cocotb after a `kill` would allow the same pattern.

ktbarrett, 2020-07-10:
I've had a need for this feature in the past, and simply worked around it.

There is #1349, which was intended to raise an exception when a task is killed. We should see that through. As I've mentioned before (https://github.com/cocotb/cocotb/pull/1846#issuecomment-631059941), this means that we should also introduce something that shields a task from cancellation (being killed). It would also free up that WIP PR.

I feel no shame in recommending cocotb just do what asyncio does.

benreynwar, 2020-07-10:
My preference would also just be to follow asyncio as closely as possible.

Somewhat, off topic, are you guys thinking of directly using asyncio in the scheduler at some point?  It feels like that could make everything much simpler if the main loop inherited from `asyncio.AbstractEventLoop` and alternated between a standard asyncio event loop and handing control back to the simulator once the asyncio events were exhausted.

benreynwar, 2020-07-10:
I had a crack at implementing something like that in
https://github.com/benreynwar/slvcodec/blob/master/slvcodec/event.py#L493

ktbarrett, 2020-07-10:
I had thought about it, I think Eric may have tried it (correct me if I'm wrong), but no one have put any serious effort into it. How well would `AbstractEventLoop` handle `cocotb.coroutine`s?

benreynwar, 2020-07-10:
That makes sense.  I don't think it would make sense until the `cocotb.coroutine` approach is deprecated and the coroutines are all using the `async` syntax.  But in the longer term in would make things simpler and it would make it nice and easy to integrate `asyncio` and `cocotb` code since they'd be the same thing.

ktbarrett, 2020-07-10:
Well it is a good motivation to deprecate and remove `cocotb.coroutine` (#1967), but it would completely change the scheduling behavior and interface. Perhaps it's one of those things we should implement alongside of what we have and slowly phase out the old scheduler?

Still we should implement `CancelledError` and `shield` for the current coroutine system. It's not a radical change and makes the current system close enough to feature-complete.

benreynwar, 2025-05-12:
I've just started using try/finally in cocotb for this purpose now that it is supported.  It's working well.  Thank you!  I think this issue can be closed.

ktbarrett, 2025-05-12:
You may also be interested in the TaskManager, which will mirror trios nursery or asynsio's TaskGroup, that I'm hoping to get in 2.1 which will probably be a better fit for what you are trying to accomplish.

closed 2025-05-12 by ktbarrett

benreynwar, 2025-05-13:
That sounds great!