cocotb/cocotb #1967: Deprecate cocotb.coroutine and generator-based coroutine support

ktbarrett, 2020-07-10:
I think it's a good idea to deprecate the `cocotb.coroutine` decorator and `cocotb.decorators.RunningCoroutine` before the 1.5 release, since I would like to see these removed in 2.0. Before we can deprecate them, we need to remove all internal references. This is the biggest problem with the tests (#1731).

## Rationale
`async`/`await` syntax performs the same job as the `cocotb.coroutine` decorated generators; however, `async` does it better.

- Is a stackful coroutine
  - better performance; no need to trampoline to and from the scheduler
  - stack inspection
- Built into Python
  - better performance; implemented in C
  - existing knowledge
  - can leverage a larger ecosystem of existing tools for advanced functionality

Are there any arguments against it's removal besides backwards compatibility? Is backwards-compatibility into 2.0 desired?

imphil, 2020-07-10:
> Are there any arguments against it's removal besides backwards compatibility? 

That's a major point, as it requires a lot of changes to existing code, including existing libraries. Effectively, it requires the whole extension ecosystem to adapt.

We should have not only reasons why the async syntax is better, but also why supporting the generator syntax is causing us major pain to support in addition to the async syntax.

eric-wieser, 2020-07-10:
If we did go this way, I'd recommend we go via a `PendingDeprecationWarning` first

ktbarrett, 2020-07-10:
>  Effectively, it requires the whole extension ecosystem to adapt.

This is a good point. I'm not worried about how this will affect end users, version numbers exist for a reason (a lot of people I know don't seem to appreciate that). But currently that ecosystem is not very big, and (hopefully) it will only get bigger. Delaying removal may mean even more work in the future. 

But removal aside, it's best that we do deprecate it soon, so that even if we don't remove it, extensions and users will be incentivized to use the new syntax or migrate sooner, reducing migration effort once it is truly removed.

> We should have not only reasons why the async syntax is better, but also why supporting the generator syntax is causing us major pain

This is less of a good point. We can work around any problem given enough effort; and it's hard to say what the threshold of "major" is. `async` syntax is certainly the easier option, that's all I can say.

PR #3509 (2023-11-20): Remove cocotb.coroutine and yield syntax
Closes #1967.
closed 2023-11-20 by ktbarrett