cocotb/cocotb #1978: Cocotb with Active HDL

dmzaom4, 2020-07-12:
Hi,

I am trying to use cocotb on Windows 10 with Active HDL 10.2. I installed anaconda for for 64 bit Python 3 and followed the instructions on the wiki.

I am trying to use the simple_dff and adder examples to get my feet wet and cannot seem to get them to work.

I keep on getting the same error:

`<KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\users\textron\miniconda3\lib\site-packages\cocotb\libs\libcocotbvpi_aldec.dll'
PLI:  Cannot load the "C:\users\textron\miniconda3\lib\site-packages\cocotb\libs\libcocotbvpi_aldec.dll" library. The library does not exist or is corrupted.
PLI:  The specified procedure could not be found.>`

When I go to the path specified the libcocotbvpi_aldec.dll is there. Any clue what I am doing wrong?

themperek, 2020-07-13:
Are you sure you use 64bit version of Active HDL?

themperek, 2020-07-13:
Please check notes for both Riviera and Active HDL also: https://docs.cocotb.org/en/latest/simulator_support.html#aldec-riviera-pro

dmzaom4, 2020-07-13:
I have tried it with Active HDL 10.2, 10.5a and 11.1 (all 64 bit):

[cid:image001.png@01D658F2.3577EC70]

[cid:image002.png@01D658F2.3577EC70]

[cid:image003.png@01D658F2.3577EC70]

imphil, 2020-07-13:
> followed the instructions on the wiki.

Which instructions did you follow there? The Wiki at https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#conda-based-installation-all-os has instructions for a 32 bit simulator, and you're using the 64 bit version of Active-HDL?

dmzaom4, 2020-07-13:
These ones:[https://docs.cocotb.org/en/stable/install.html](url)

I realized I had a 64 bit simulator.

dmzaom4, 2020-07-13:
Would I be better off using the 32 bit Active HDL and 32 bit CocoTB?

garmin-mjames, 2020-07-13:
I've been using 32-bit Active-HDL and Conda with no issues, but I haven't tried 64-bit.

There are known issues with 10.4a and 10.5a and cocotb though: #1494

garmin-mjames, 2020-07-13:
> PLI: The specified procedure could not be found.

If you have the Visual Studio command prompt, try `dumpbin /exports libcocotbvpi_aldec.dll` and make sure the DLL has `vlog_startup_routines` exported. You can also see this in other DLL tools like Dependency Walker.
When I was building the DLLs using the MinGW installation that comes with Active-HDL, I had to add some linker flag to make the VPI functions visible.

dmzaom4, 2020-07-13:
OK, I will install a 32-bit version of Active-HDL (not 10.4a or 10.5a) and follow the instructions to setup cocotb for 32 bit.

themperek, 2020-07-13:
Did you try ?

```
On Windows, do not install the C++ compiler, i.e. unselect it during the installation process of Riviera-PRO. (A workaround is to remove or rename the mingw directory located in the Riviera-PRO installation directory.)
```

dmzaom4, 2020-07-13:
Does that apply to Active-HDL too? When I installed Active-HDL I didn’t know about cocoTB yet. I will re-install it and see if that option is there.

garmin-mjames, 2020-07-13:
I don't think there is an installation option for that, but for Active-HDL 11.1 I had to rename the mingw folder.

dmzaom4, 2020-07-14:
Then that doesn’t explain why Active HDL 11.1 isn’t working.

dmzaom4, 2020-07-14:
I tried the 32 bit conda and active hdl, same issue. Then I renamed the MINGW folder it started working!

Thank you!

dmzaom4, 2020-07-14:
Well, the adder example still fails:

(base) C:\Users\Textron\Downloads\cocotb-master\cocotb-master\examples\adder\tests>make SIM=activehdl GUI=1 TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/c/Users/Textron/Downloads/cocotb-master/cocotb-master/examples/adder/tests'
set -o pipefail; GPI_EXTRA= TOPLEVEL_LANG=vhdl \
MODULE=test_adder TESTCASE= TOPLEVEL=adder /c/Aldec/Active-HDL-11.1/bin/vsimsa  -do sim_build/runsim.do | tee sim_build/sim.log

VHDL/Verilog/EDIF/SystemC Simulator build 11.1.263.7352

(c) 1997-2020 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `C:\Users\Textron\Downloads\cocotb-master\cocotb-master\examples\adder\tests\library.cfg', `C:\Users\Textron\Downloads\cocotb-master\cocotb-master\examples\adder\tests\vsimsa.cfg'
VSIMSA: `work' is now working library.
Welcome to VSIMSA!
This message was printed from `startup.do' macro file.

alib work
ALIB: Library "work" attached.
work = C:\Users\Textron\Downloads\cocotb-master\cocotb-master\examples\adder\tests\work\work.lib
set worklib work
VSIMSA: `work' is now working library.
alog -timescale 1ns/1ps  +define+COCOTB_SIM -dbg C:/Users/Textron/Downloads/cocotb-master/cocotb-master/examples/adder/hdl/adder.sv
Pass 1. Scanning modules hierarchy.
Info: VCP2113 Module \$root found in current working library.
Pass 2. Processing instantiations.
Pass 3. Processing behavioral statements.
ELB/DAG code generating.
Info: VCP2113 Module \$root found in current working library.
Unit top modules: adder.
$root top modules: adder.
Compile success 0 Errors 0 Warnings  Analysis time: 0[s].
done
asim   +access +w -interceptcoutput -O2 -dbg -pli "C:/users/textron/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_aldec" adder
ELBREAD: Elaboration process.
ELBREAD: Elaboration time 0.0 [s].
KERNEL: Main thread initiated.
KERNEL: Kernel process initialization phase.
ELAB2: Elaboration final pass...
KERNEL: PLI/VHPI kernel's engine initialization done.
PLI: Loading library 'C:\users\textron\miniconda3\lib\site-packages\cocotb\libs\libcocotbvpi_aldec.dll'
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
ELAB2: Warning: ELAB2_0113 Task or function "$dumpfile" not defined in module "adder" from library "work".
ELAB2: Warning: ELAB2_0113 Task or function "$dumpvars" not defined in module "adder" from library "work".
ELAB2: Create instances ...
KERNEL: Time resolution set to 1ps.
ELAB2: Create instances complete.
ELAB2: Elaboration final pass complete - time: 0.0 [s].
KERNEL: Kernel process initialization done.
Allocation: Simulator allocated 6027 kB (elbread=1280 elab2=4613 kernel=133 sdf=0)
KERNEL: ASDB file was created in location C:\Users\Textron\Downloads\cocotb-master\cocotb-master\examples\adder\tests\wave.asdb
run -all
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Aldec HDL Simulator version 11.1.263
     0.00ns INFO     cocotb                                      __init__.py:209  in _initialise_testbench           Running tests with cocotb v1.4.0 from C:\users\textron\miniconda3\lib\site-packages\cocotb
     0.00ns INFO     cocotb                                      __init__.py:229  in _initialise_testbench           Seeding Python random module with 1594748889
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test_adder.adder_basic_test
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test_adder.adder_randomised_test
     0.00ns INFO     cocotb.regression                         regression.py:463  in _start_test                     Running test 1/2: adder_basic_test
     0.00ns INFO     ..b.test.adder_basic_test.0xfc985f0       decorators.py:256  in _advance                        Starting test: "adder_basic_test"
                                                                                                                     Description: Test for 5 + 10
     2.00ns INFO     cocotb.regression                         regression.py:361  in _score_test                     Test Passed: adder_basic_testsys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=1 mode='w' encoding='cp1252'>
     2.00ns INFO     cocotb.regression                         regression.py:463  in _start_test                     Running test 2/2: adder_randomised_test
     2.00ns INFO     ..t.adder_randomised_test.0xfc8a0d0       decorators.py:256  in _advance                        Starting test: "adder_randomised_test"
                                                                                                                     Description: Test for adding 2 random numbers multiple times
    22.00ns INFO     cocotb.regression                         regression.py:361  in _score_test                     Test Passed: adder_randomised_test
[... cut ...]
    22.00ns INFO     cocotb.regression                         regression.py:255  in tear_down                       Shutting down...
/c/users/textron/miniconda3/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.activehdl:62: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 127
make[1]: Leaving directory '/c/Users/Textron/Downloads/cocotb-master/cocotb-master/examples/adder/tests'
c:\users\textron\miniconda3\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2

garmin-mjames, 2020-07-14:
The tests themselves pass, for some reason `make` is seeing an error. Possibly from the exit code of Active-HDL.
I see the same with 11.1 locally.

dmzaom4, 2020-07-14:
How do I get the simulation to display in the ActiveHDL GUI?

The GUI=1 switch doesn’t seem to do anything. The waves.asdb is blank when I open it.

dmzaom4, 2020-07-14:
Also when I try running the TUN ping example I get this error:

0.00ns CRITICAL cocotb.regression                         regression.py:177  in _discover_tests                 FaiTraceback (most recent call last):ly: No module named 'fcntl'

garmin-mjames, 2020-07-14:
The `GUI` flag is not currently supported for Active-HDL.
You'll have to modify the `Makefile.activehdl` file to log signals to the asdb file by adding the right DO command to the `runsim.do` file generation.

garmin-mjames, 2020-08-12:
Closing this. If you need further help related to this, feel free to re-open or create a new issue.

closed 2020-08-12 by garmin-mjames