cocotb/cocotb #1981: cannot run on old kernels

lisrael1, 2020-07-13:
when running with vcs, with cocotb version 1.4.0, i get this vcs error:
INFO cocotb: Error-[VPI-LOAD] VPI -load library error
INFO cocotb:   Could not open VPI library
INFO cocotb:   /nfs/iil/disks/virtual_env/env/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_vcs.so
INFO cocotb:   : /usr/pkgs/gcc/4.7.2/lib64/libstdc++.so.6: version `GLIBCXX_3.4.21`
INFO cocotb:   not found (`required by
INFO cocotb:   /nfs/iil/disks/python3.6/site-packages/cocotb/libs/libcocotbvpi_vcs.so`)
INFO cocotb:   Please check if your LD_LIBRARY_PATH has the path to the library and if the
INFO cocotb:   library is in a compatible object code format.

note that some vcs users use old linux kernels and they don't have new glibc. 
i have glibc 2.22, and in another machine 2.11, i don't know if it's practical for you to use old glibc as you had in cocotb 1.3.2, but it will be very helpful
thanks!

eric-wieser, 2020-07-13:
Have you read our help page on this, https://docs.cocotb.org/en/stable/troubleshooting.html#glibcxx-3-4-xx-not-found

lisrael1, 2020-07-13:
didn't see this page, and you're right, the gcc version was old: 4.7.2.
i updated gcc to 6.2.0 and it works! thanks!

closed 2020-07-13 by lisrael1