cocotb/cocotb #1983: Error while calling ports from inside a subblock

crusader2000, 2020-07-14:
Error -
[image]

I am facing this issue while calling trying to call ports present inside a sub_block (lut4 , in this case). Am I doing something wrong?

Makefile-
```
SIM = icarus
TOPLEVEL_LANG = verilog

VERILOG_SOURCES = ./cluster.v ./lut4.v ./flipflop.v ./sw2.v ./cfg_data_d16.v ./cfg_data_d1.v
# TOPLEVEL is the name of the toplevel instance in the Verilog or VHDL file
TOPLEVEL=cluster
# instance is the name of the python testfile
instance=test

include $(shell cocotb-config --makefiles)/Makefile.inc
include $(shell cocotb-config --makefiles)/Makefile.sim
```

Test Module- 
```
 Automatically generated by PRGA's RTL generator
timescale 1ns/1ps
module cluster (
    input wire [0:0] clk,
    input wire [3:0] i,
    output wire [0:0] o,
    input wire [0:0] cfg_e,
    input wire [0:0] cfg_clk,
    input wire [0:0] cfg_we,
    input wire [0:0] cfg_i
    output wire [0:0] cfg_o
    );
    
    wire [0:0] _lut__out;
    wire [0:0] _lut__cfg_o;
    wire [0:0] _ff__Q;
    wire [0:0] __sw_o__o;
    wire [0:0] __sw_o__cfg_o;
    
    lut4 lut (
        .bits_in(i),
        .out(_lut__out),
        .cfg_clk(cfg_clk),
        .cfg_e(cfg_e),
        .cfg_we(cfg_we),
        .cfg_i(cfg_i),
        .cfg_o(_lut__cfg_o)
        );
    flipflop ff (
        .clk(clk),
        .D(_lut__out),
        .Q(_ff__Q),
        .cfg_e(cfg_e)
        );
    sw2 _sw_o (
        .i({_ff__Q, _lut__out}),
        .o(__sw_o__o),
        .cfg_clk(cfg_clk),
        .cfg_e(cfg_e),
        .cfg_we(cfg_we),
        .cfg_i(_lut__cfg_o),
        .cfg_o(__sw_o__cfg_o)
        );
    
    assign o = __sw_o__o;
    assign cfg_o = __sw_o__cfg_o;

endmodule

```

test.py - 
```
import cocotb
from cocotb.triggers import Timer,RisingEdge,Edge,First
from cocotb.clock import Clock
import random
import math
from cocotb.result import TestFailure
from cocotb.binary import BinaryValue
from cocotb.scoreboard import Scoreboard

def clock_generation(clk):
    clock_period = 10 #This must be an even number
    test_time = 10000
    
    c= Clock(clk,clock_period)

    cocotb.fork(c.start(test_time//clock_period,start_high = False))
    
@cocotb.test()
def simple_test(dut):
    """Test bench from scratch for 4 input LUT"""
    
    clock_generation(dut.cfg_clk)
    clk = dut.cfg_clk
    # Signals
    input = dut.lut.bits_in
    out = dut.lut.out
    cfg_e = dut.lut.i_cfg_data.cfg_e
    cfg_we = dut.lut.i_cfg_data.cfg_we
    cfg_i = dut.lut.i_cfg_data.cfg_i
    cfg_o = dut.lut.i_cfg_data.cfg_o
    
    # No. of input bits
    # n_input = input()
    n_input = int(math.log(16,2))

    # Setting up LUT
    # Set the value of cfd
    cfg_e <= 1;
    cfg_we <= 1;
    cfd = []
    n_bits = 2**n_input
    for _ in range(n_bits):
        bit = random.choice([0,1])
        cfd.insert(0,bit)
        cfg_i <= bit
        yield RisingEdge(clk)
    
    cfg_e <= 0;
    cfg_we <= 0;
    yield RisingEdge(clk)

    for i in range(0,n_bits):
        input <= i
        yield RisingEdge(clk)
        output = out.value.integer

        if output != cfd[i]:
            raise TestFailure("[ERROR]")  
```

eric-wieser, 2020-07-14:
> `instance is the name of the python testfile`

Where did you get `instance` from? The variable should be `MODULE`, as described in the error message.

Is there anything we could do to make the message more clear?

eric-wieser, 2020-07-14:
Perhaps getting the traceback to appear as part of the log line via `exc_info` would make it clearer.

closed 2020-07-14 by crusader2000

ktbarrett, 2020-07-14:
#1591 is relevant. Not to mention that the version of cocotb used here is out of date. I'll reiterate, throwing an exception when `MODULE` is not defined is a bad idea. We should log an error message and exit cleanly.

In the case of `MODULE` being defined, but the module has no tests, we log an error message and exit cleanly. So throwing in this case is not consistent. Throwing also leads to the big nasty error message "Simulation ended prematurely" which most of the time implies deadlock or some internal cocotb issue. It's too severe.