cocotb/cocotb #1986: GHDL not reporting simulator name and version

ktbarrett, 2020-07-16:
Simulating with GHDL prints the following when it get to the simulator name and version section.

```
     0.00ns WARNING  cocotb.gpi                                  VpiImpl.cpp:80   in get_simulator_product           Could not obtain info about the simulator
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on UNKNOWN version UNKNOWN
```

Investigation should be done on whether this is an artifact of the change introduced by #1737 or if it's a limitation in GHDL.

ktbarrett, 2020-07-17:
GHDL does return the product and version, but not argc/argv, so it is set to always fail. 

https://github.com/ghdl/ghdl/blob/99b542c849311c92e87e2c70d283de133c9d4093/src/grt/grt-vpi.adb#L1561-L1573

closed 2020-07-19 by ktbarrett

eric-wieser, 2020-07-19:
Fixed upstream, I assume.

ktbarrett, 2020-07-19:
@eric-wieser partially, I'm not familiar enough with Ada to finish the last bit #1990.