cocotb/cocotb #1989: Ability to call SystemVerilog task

Alex-Mann, 2020-07-17:
I'm wondering if there is any ability to call a SystemVerilog/UVM Task/Function from Cocotb.

Instead of living in purely Cocotb, or purely UVM, is it possible from the python side to direct and control the simulation? Say I start my test by calling a Cocotb async routine, then I want to call a SystemVerilog function that is exposed at the TB level? Maybe this is not an ideal use case, but I think it would provide a great amount of flexibility and help to drive even more adoption to Cocotb.

I know there is the DPI interface piece in SV, but having a really easy way to call one of these externs from the Cocotb python side would be awesome.

ktbarrett, 2020-07-17:
#1217 is related.

ktbarrett, 2020-07-18:
To expand on the original inquiry about calling SystemVerilog tasks (or VHDL procedures) from cocotb the best to accomplish that is using the SystemVerilog-DPI for Verilog, or VHPIDIRECT in GHDL (if @umarcor get his way it will be a standard). 

There is nothing inherently in the way of using these interfaces simultaneously with cocotb (or any VPI or VHPI cosim). For proper cocotb support though we need these the interface to tie into the simulator since tasks can consume simulation time. The easiest way to accomplish this is using `cocotb.external`. The interface functions could interact with the exported DPI/VHPIDIRECT functions with ctypes if the interface is simple enough, so it could be "easy".

I think we have all the necessary pieces to prototype this, if someone is up to it.

umarcor, 2020-07-19:
AFAIU, the VHDL standard does not define how to execute VHDL functions procedures from a foreign language. There are two possibilities only:

- To execute foreign functions from VHDL (through VHPIDIRECT).
- To use VHPI.

Apart from that, GHDL allows VHDL functions/procedures to be executed from a foreign function. However, this is more of a side effect (because the internal architecture of GHDL is similar to a compiler) than a feature which is implemented on purpose. Hence, it is implementation dependent, not standarized.

I am trying to have the VHDL standard extended/enhanced, to include the details which are currently undefined. These are mostly:

- How should fat pointers look like, or whether tools should provide a header to decode/encode fat pointers.
- How should VHDL functions/procedures be exposed in C; what is the name of the symbol.
  - How to pass a context, so that VHDL functions/procedures executed from C can have `wait` statements.

This is not a new idea. A proposal exists in the standarization group since 2011: http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/DpiProposal. My contribution is being to write down what I have gotten to learn about how GHDL's VHPIDIRECT works:

- https://github.com/umarcor/ghdl-cosim/blob/vhdl202x-lcd-dpi/vhdl202x-lcs-dpi/main.md
- https://ghdl.github.io/ghdl-cosim/

Other users in the VASG are interested on the feature precisely to allow a better integration of VHDL and SystemVerilog and/or Verilog. Hence, the standarized result is unlikely to be an exact copy of how GHDL works today, but I believe it is a solid foundation.

Regarding co-execution of VHDL and Python, the current problems with GHDL are:

- It is a little more verbose than I'd like to, because it seems that an intermediate C file is required. I hope that can be improved (both in VHDL/VHPIDIRECT and in Python/ctypes): ghdl/ghdl#1398.
- Failing simulations exit with an abortion that forces any wrapping Python to close abruptly: ghdl/ghdl-cosim#15. I believe this is a bug in GHDL, but Tristan is currently very busy with synthesis.

Apart from those points, it works and it is currently usable (albeit not completely documented).

Overall, it would be really nice if some developers/maintainers/users of cocotb could take part in the discussion. VASG meets every ~2 weeks. There is also a reflector/mailing-list. Moreover, members and "usual suspects" in the VASG participate in many gitter rooms (VHDL, ghdl...). I think that ghdl/ghdl#1398 is a good first reference to understand what's currently possible and which is my proposal for the future.

Alex-Mann, 2020-07-23:
I can take a crack at prototyping this, but it would be helpful if you could give a general overview of how and where you would structure the calls and integrations.

ktbarrett, 2020-07-24:
@Alex-Mann 

High level you need a few things:
* export a function/task via the DPI from SV
  * importing a Python task will require more work and more thought, so we will skip that for now
* create a shim in C for communicating with that function/task
  * shim will include the svdpi.h header
  * shim will create a wrapper function around the exported SV function/task with the DPI and offer a more regular C interface
  * compile it as a shared library
* from cocotb load the shim library and create a wrapper using ctypes
  * `ctypes.CDLL` to load the shim library
  * wrapper must be decorated with `cocotb.external` so you can await on it

That should be it. Knowledge of the DPI and ctypes should be all you need. Ideally, we can figure out how to work around the creation of the shim, but small steps...

eric-wieser, 2020-07-24:
> The easiest way to accomplish this is using `cocotb.external`.

`cocotb.external` tasks are not allowed to advance the scheduler unless they go through `cocotb.function` to do so, which may cause an issue.

ktbarrett, 2020-07-24:
@eric-wieser Yeah, started looking more into it, it is a problem. We need something that behaves like a trigger so we can allow the simulator to advance and re-enter the scheduler once complete.

eric-wieser, 2020-07-24:
I don't really understand how systemverilog DPI is supposed to work. I assume the systemverilog first calls into the C code, and not the other way around?

ktbarrett, 2020-07-24:
From the little I know about it, DPI is just an FFI, it doesn't define any control. You can write testbench HDL code that calls into an imported C function, or you can call into an exported SV task or function from a cosim application (or the aforementioned C helper function).

eric-wieser, 2020-07-24:
My current understanding of DPI is:
* Calling a task from C results in your C thread being suspended
* Calling a C function from verilog spawns a new thread for your function
* The former can only be done if your thread originated from the latter

Alex-Mann, 2020-07-24:
So I'll try to summarize what I've been reading about the DPI here incase it helps spur some new ideas on how to implement this...

As @ktbarrett said, it really is just an FFI for SV. It allows you to:

- **import** functions or tasks defined in C
- **export** functions or tasks defined in SV

The SystemVerilog DPI supports two types of functions:

### Pure Functions

A nonvoid (must return a value) function which depends on only the input arguments and contains no side effects. They cannot:
- perform any file IO
- read or write anything (ie. no env variables, OS objects, shared memory, etc)
- access any persistent data like global or static variables

### Context Functions
An imported C function which calls exported tasks or functions, or accesses SystemVerilog data objects other than its actual arguments (taken from [Wikipedia](https://en.wikipedia.org/wiki/SystemVerilog_DPI?oldformat=true#Context_tasks_and_functions)). These functions can call the VPI.

---

Without knowing really how the C / Cpp layer of Cocotb works.. it seems like you would need to be aware of _extern_  functions/tasks at compile time, and then provide a cocotb api to call this exported function from the python side since it would have to be able to advance simulation time and fit into the overall Cocotb model.

ktbarrett, 2020-07-25:
@eric-wieser 
I've been reading over the spec [1] and it seems that any caller can call any exported task given they first change their context to the scope where the exported task is defined using `svSetScope` (I'm guessing the idea is that a particular scope is handled by a single thread where re-entrancy is safe?). They even specifically mention calling an exported task from a VPI callback in 35.5.3.

@Alex-Mann 
I don't think any C level changes are required of cocotb, at least not for a prototype.

I think this will require the following:
* export functions/tasks with DPI
  * not testing import functions/tasks as it's a whole other beast that's a lot more wicked
* create a shim library
  * this will do the necessary C things to interface with the DPI and provide a more regular interface to a caller
  * compile this as a shared library
* use the shim from cocotb
  * Use `ctypes` to load the library
  * create a Python wrapper for calling the C function using `ctypes`
  * wrap the Python function in something to make it appear to be a trigger to the scheduler

Most of this is out-of-scope for inclusion into cocotb, although we *definitely* want this documented. One thing cocotb could incorporate is the last point above. This 'thing' would be a `Trigger`, and it would call a Python function with a callback to send the trigger to `scheduler.react` once complete.

[1] You can find it [here](http://ecee.colorado.edu/~mathys/ecen2350/IntelSoftware/pdf/IEEE_Std1800-2017_8299595.pdf), section 35, page 939, if you don't have a IEEE membership on hand. It's crazy they make you pay for access to "open" standards.

eric-wieser, 2020-07-27:
From that same section:

> The behavior of the DPI scope-related APIs and invocation of DPI export subroutines will be simulator
defined and is beyond the scope of the DPI specification.

It sounds like `svSetScope` must either:
* Somehow change the thread of the caller, via `setjmp` / `longjmp` shenanigans
* Force the PLI event handler thread to jump ship to another thread.

Both of these sound likely to make cocotb unhappy.

ktbarrett, 2020-07-27:
There is also an Annex H (starting on page 1222) that gives more detail on the DPI.

>  The behavior of DPI utility functions that manipulate context is undefined
when they are invoked by any subroutine that is not part of a DPI context call chain (see 35.5.3)

Which implies what we'd like to do is undefined.

Section H.9.5 goes over the interaction between DPI and VPI, but completely fails to mention calling exported tasks from VPI callbacks. Not mentioning it usually means it's unsupported.

But from 35.5.3
> A foreign language subroutine supported through some other interface (a VPI callback for example), can
also make a call to svSetScope or to other DPI scope-related APIs. This foreign language subroutine can
also call an export subroutine declared in a specific instantiated scope by first making a call to svSetScope.
The behavior of the DPI scope-related APIs and invocation of DPI export subroutines will be simulator
defined and is beyond the scope of the DPI specification. 

Which is inconsistent with what was said in Annex H.

My best guess is they simply didn't expect people to be doing what we are trying to do and it's not well specified. And I doubt most simulators will support this use case gracefully. So I'm probably wrong, I don't think we can support this in combination with cocotb. DPI functionality can be supported in concert with cocotb ***only*** if the two are totally independent. I don't think signalling between the two would work terribly well either.

Pinging @hackfin, he was asking some questions related to this the other day (of course it was GHDL and VHPIDIRECT, but similar).

garmin-mjames, 2020-11-06:
Also from 35.5.3:
> A DPI import call chain is an inter-language call chain starting from SystemVerilog into a subroutine that is
defined in a DPI supported language. The starting point of the call chain from SystemVerilog is called the
*root of the call chain*.

So the DPI spec only defines how to start from SystemVerilog and go to C through an imported subroutine.
Support for the root of the call chain being in a VPI callback is simulator dependent.

mballance, 2020-12-07:
Apologies for jumping into this thread late... I do see value in being able to call DPI functions/tasks from cocotb (either directly or in the context of a library). Depending on the simulator, cocotb may need to take some action to support this out of the box. Specifically, I believe a DPI-enabled GPI implementation would be required to ensure that cocotb runs within a DPI context. 
I think the actual code involved (on top of the VPI-based GPI implementation) would be fairly minimal. That said, it would be another GPI implementation. Is there openness to having another GPI backend?

ktbarrett, 2020-12-07:
@mballance I'm not so sure how that would work? As stated in this issue, calling a DPI function/task from a VPI callback does not seem to be supported by the standard. We don't have anything against someone developing a DPI/VPI GPI implementation as an extension. Are you or @Alex-Mann willing to push the standards committee to clarify the standard?

mballance, 2020-12-07:
Hi @ktbarrett, the key issue that I've observed is that calls from cocotb back into the simulator via DPI come from VPI-land, and thus do not have a DPI context. If callbacks (sim-event, time-callback, signal-callback) are routed through DPI prior to calling cocotb, it *should* then be fine to make DPI calls from cocotb.
Clarifying the standard would be a good long-term step. Of course the clarification could end up being that VPI->DPI calls are not supported.

ktbarrett, 2020-12-07:
> If callbacks (sim-event, time-callback, signal-callback) are routed through DPI prior to calling cocotb, it should then be fine to make DPI calls from cocotb.

How is that possible?

mballance, 2020-12-07:
@ktbarrett, As they say, the devil is in the details... Some of the details will hinge on exactly what cocotb users that want to use DPI require in terms of simulation access. Is it sufficient to have access to simulation- and time-event callbacks, and use DPI for interacting with the design? Or, is full signal access required?
By the way, I did dig up some interesting work that Jonathan Bromley did on connecting VPI and DPI. Possibly some directions to explore...
https://www.verilab.com/files/snug_2012_presentation71_final.pdf
https://www.verilab.com/files/snug_2012_paper71_final.pdf

My primary interest, though, was understanding the openness of the maintainers to having a new GPI implementation. I wasn't sure if the maintenance burden is sufficient that strongly capping the number is important.

ktbarrett, 2020-12-07:
@mballance Maintenance burden is not an issue if the implementation is not incorporated into this repo. We are trying to avoid becoming a dumping pile of good ideas. We want to provide a solid core of minimum features for users to write testbenches in Python "the cocotb way" and provide good interfaces for extensions.

It's not currently possible to make a new GPI implementation as an extension (though it's not far off), but we would rather work on making that a possibility than maintaining another GPI implementation in-repo. The GPI certainly needs a second pass to redesign it as an end product and not as an implementation detail. There might even be [others](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/13) interested in such a thing.

mballance, 2020-12-07:
@ktbarrett, makes sense and appreciate the feedback.

Alex-Mann, 2020-12-07:
Is there an overview on the implementation/architecture of the GPI or the best thing to do to just dig into the source code?

ktbarrett, 2020-12-07:
@Alex-Mann Unfortunately, digging into the source code and asking questions on the gitter is the best thing to do now. We do plan on documenting it all at some point [here](https://github.com/cocotb/cocotb/wiki/cocotb-Internals).

ktbarrett, 2021-07-26:
Following on this discussion and those papers, it seems the best way of "calling" a DPI export is actually to not call the DPI export, but instead communicate over an asynchronous queue. The VPI callback would add "calls" to the queue (TLM queue?) and toggle a notification noting that there are messages on the queue to consume. A SystemVerilog process would run when that notifier changes, pull messages from the queue and does the actual call.

Alex-Mann, 2021-07-26:
When you refer to "those papers" which ones exactly are you talking about? I'd like to take a look through them if there is some new material/ideas.

ktbarrett, 2021-07-26:
@Alex-Mann The ones that @mballance posted. In the second link on page 6 it goes over how it worked around the inability to call DPI exports from VPI callbacks. I simply worked from there with this specific task in mind.

Alex-Mann, 2024-05-09:
Hi @mballance, I just watched [your talk you gave at Latch Up](https://www.youtube.com/watch?v=-odFakR7h3g) (great stuff) - does this mean my ask in this thread could now be supported in Cocotb? Are there any limitations currently?

I would be interested in putting together a prototype of this which I could back contribute as an example to cocotb if yes.

mballance, 2024-05-09:
Hi @Alex-Mann, yes, that is my understanding. It should now be possible to cross-call between Python+cocotb and SystemVerilog tasks. At the moment, known limitations are around platform (just Linux at the moment, though there are no technical issues supporting OSX and Windows) and testing across all simulators (development has been with one OSS and one closed-source simulator).
My goal would be to have an example of cocotb/SV interaction in both the PyHDL-IF package and in cocotb. I *think* there's the beginnings of one in the PyHDL-IF package, but let me confirm. I'll give an update on this thread, and you can use that example as a starting point for a cocotb-contributed example.

mballance, 2024-05-12:
Wanted to provide a bit of an update. There is now a first example showing cocotb and PyHDL-IF working together here:
https://github.com/fvutils/pyhdl-if/tree/main/examples/call/cocotb/call_sv_bfm

The basic concept is using cocotb for some signal-level interations (wait for reset), while calling the tasks of a BFM implemented directly in SV.

Alex-Mann, 2024-08-03:
Starting to look at this now... can you tell me what Python version and Verilator version you tested with? I am getting a bunch of warnings that are causing an error using:
* Verilator 5.027 devel rev v5.026-97-g37a400209
* Python 3.12.4

```
(example_venv) root@ff06d3679b42:/work/pyhdl-if/examples/call/cocotb/call_sv_bfm# ./runit_vlt.sh 
/work/pyhdl-if/examples/call/cocotb/call_sv_bfm
hdl_if_libs=/work/pyhdl-if/src/hdl_if/entry.cpython-312-aarch64-linux-gnu.so /usr/local/lib/libpython3.12.so
%Warning-PINMISSING: /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:154:29: Cell has missing pin: 'err'
  154 |     WishboneInitiatorBFM    init_bfm(
      |                             ^~~~~~~~
                     /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:9:29: ... Location of port declaration
    9 |     input                   err,
      |                             ^~~
                     ... For warning description see https://verilator.org/warn/PINMISSING?v=5.027
                     ... Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:154:29: Cell has missing pin: 'sel'
  154 |     WishboneInitiatorBFM    init_bfm(
      |                             ^~~~~~~~
                     /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:10:29: ... Location of port declaration
   10 |     output reg[3:0]         sel,
      |                             ^~~
%Warning-WIDTHEXPAND: /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if_call_api.svh:192:40: Operator FUNCREF 'PyLong_FromLong' expects 64 bits on the Function Argument, but Function Argument's VARREF 'obj_id' generates 32 bits.
                                                                                      : ... note: In instance 'call_sv_bfm'
  192 |         void'(PyTuple_SetItem(args, 1, PyLong_FromLong(obj_id)));
      |                                        ^~~~~~~~~~~~~~~
                      /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if.sv:181:1: ... note: In file included from 'pyhdl_if.sv'
%Warning-WIDTHTRUNC: hdl_call_if_api.svh:84:49: Operator ASSIGN expects 32 bits on the Assign RHS, but Assign RHS's FUNCREF 'PyLong_AsLong' generates 64 bits.
                                              : ... note: In instance 'call_sv_bfm.init_bfm'
   84 |                 int unsigned __addr = pyhdl_if::PyLong_AsLong(pyhdl_if::PyTuple_GetItem(args, 0));
      |                                                 ^~~~~~~~~~~~~
                     /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:16:1: ... note: In file included from 'call_sv_bfm.sv'
%Warning-WIDTHTRUNC: hdl_call_if_api.svh:85:49: Operator ASSIGN expects 32 bits on the Assign RHS, but Assign RHS's FUNCREF 'PyLong_AsLong' generates 64 bits.
                                              : ... note: In instance 'call_sv_bfm.init_bfm'
   85 |                 int unsigned __data = pyhdl_if::PyLong_AsLong(pyhdl_if::PyTuple_GetItem(args, 1));
      |                                                 ^~~~~~~~~~~~~
                     /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:16:1: ... note: In file included from 'call_sv_bfm.sv'
%Warning-WIDTHTRUNC: hdl_call_if_api.svh:92:49: Operator ASSIGN expects 32 bits on the Assign RHS, but Assign RHS's FUNCREF 'PyLong_AsLong' generates 64 bits.
                                              : ... note: In instance 'call_sv_bfm.init_bfm'
   92 |                 int unsigned __addr = pyhdl_if::PyLong_AsLong(pyhdl_if::PyTuple_GetItem(args, 0));
      |                                                 ^~~~~~~~~~~~~
                     /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:16:1: ... note: In file included from 'call_sv_bfm.sv'
%Warning-WIDTHEXPAND: hdl_call_if_api.svh:94:36: Operator FUNCREF 'PyLong_FromLong' expects 64 bits on the Function Argument, but Function Argument's VARREF '__retval' generates 32 bits.
                                               : ... note: In instance 'call_sv_bfm.init_bfm'
   94 |                 retval = pyhdl_if::PyLong_FromLong(__retval);
      |                                    ^~~~~~~~~~~~~~~
                      /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:16:1: ... note: In file included from 'call_sv_bfm.sv'
%Warning-WIDTHEXPAND: /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if_call_api.svh:172:40: Operator FUNCREF 'PyLong_FromLong' expects 64 bits on the Function Argument, but Function Argument's VARREF 'trim_elems' generates 32 bits.
                                                                                      : ... note: In instance 'call_sv_bfm'
  172 |         void'(PyTuple_SetItem(args, 2, PyLong_FromLong(trim_elems)));
      |                                        ^~~~~~~~~~~~~~~
                      /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if.sv:181:1: ... note: In file included from 'pyhdl_if.sv'
%Warning-WIDTHEXPAND: /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if.sv:147:18: Operator FUNCREF 'PyTuple_New' expects 64 bits on the Function Argument, but Function Argument's CMETHODHARD 'size' generates 32 bits.
                                                                            : ... note: In instance 'call_sv_bfm'
  147 |         args_o = PyTuple_New(args.size());
      |                  ^~~~~~~~~~~
[... cut ...]
                                                                                      : ... note: In instance 'call_sv_bfm'
  148 |         void'(PyTuple_SetItem(args, 1, PyLong_FromLong(obj_id)));
      |                                        ^~~~~~~~~~~~~~~
                      /work/pyhdl-if/src/hdl_if/share/dpi/pyhdl_if.sv:181:1: ... note: In file included from 'pyhdl_if.sv'
%Error: Exiting due to 12 warning(s)
ERROR
```

I also modified `call_sv_bfm.sv` since Verilator didn't like mixing of blocking and non-blocking assignments.
[image]

I'm running all of this almost as-is inside of a docker container.

mballance, 2024-08-04:
If I recall correctly, I ended up hitting  a snag running this with Verilator due to Verilator's built-in 'main' not supporting VPI. Commercial simulators will work okay.

I've updated the example script and BFM example to address the blocking/non-blocking issue and the error you note above. 

Verilator needs to be linked with `-export-dynamic` in order for DPI libraries to be able to resolve the DPI support-function symbols. The script now passes `-LDFLAGS -export-dynamic` to Verilator.

It should be possible to use the Verilator 'main' provided with cocotb. I just haven't had the time to integrate this into the example script.

Alex-Mann, 2024-08-05:
Thanks. I have it running with cocotb, but I'm a bit stuck now... it seems like the `HdlObjRgy` is not getting populated... I'm tracing through how this gets initialized now but taking me some time to dig through the code and machinery. If you have any tips / outline of how it is all getting initialized that would be great.

The `Multiple matches to pattern .*\.init_bfm: None` is slightly misleading as there is nothing in the Obj instance right now.
```
     0.00ns INFO     cocotb.regression                  entry failed
                                                        Traceback (most recent call last):
                                                          File "/work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.py", line 25, in entry
                                                            init_bfm = rgy.findObj(r".*\.init_bfm", regex=True)
                                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/work/pyhdl-if/src/hdl_if/hdl_obj_rgy.py", line 59, in findObj
                                                            raise Exception("Multiple matches to pattern %s: %s" % (inst_name, str(ret)))
                                                        Exception: Multiple matches to pattern .*\.init_bfm: None
```

<details><summary>Full log</summary>
<p>

```
MODULE=call_sv_bfm TESTCASE= TOPLEVEL=call_sv_bfm TOPLEVEL_LANG=verilog \
         sim_build/Vtop  
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/example_venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.027 devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.0 from /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/example_venv/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1722877494
     0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages
     0.00ns INFO     cocotb.regression                  Found test call_sv_bfm.entry
     0.00ns INFO     cocotb.regression                  running entry (1/1)
entry
/work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.py:21: RuntimeWarning: Timer setup with value 0, which might exhibit undefined behavior in some simulators
  await Timer(0, 'ns')
post-timeout
calling rgy inst
<hdl_if.hdl_obj_rgy.HdlObjRgy object at 0xffffa95590a0>
.*\.init_bfm
Keys: dict_keys([])
dpi_init
calling rgy inst
INFO: starting pyhdl if
-Info: /work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.sv:132: $dumpvar ignored, as Verilated without --trace
[SV] Path: call_sv_bfm.init_bfm.init
     0.00ns INFO     cocotb.regression                  entry failed
                                                        Traceback (most recent call last):
                                                          File "/work/pyhdl-if/examples/call/cocotb/call_sv_bfm/call_sv_bfm.py", line 25, in entry
                                                            init_bfm = rgy.findObj(r".*\.init_bfm", regex=True)
                                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/work/pyhdl-if/src/hdl_if/hdl_obj_rgy.py", line 59, in findObj
                                                            raise Exception("Multiple matches to pattern %s: %s" % (inst_name, str(ret)))
                                                        Exception: Multiple matches to pattern .*\.init_bfm: None
     0.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** call_sv_bfm.entry              FAIL           0.00           0.00          0.34  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                  0.00           0.00          0.20  **
                                                        **************************************************************************************
                                                        
- :0: Verilog $finish
make[1]: Leaving directory '/work/pyhdl-if/examples/call/cocotb/call_sv_bfm'
```

</p>
</details>

ktbarrett, 2024-08-05:
`await Timer(0, 'ns')` should be avoided. It is an illegal transition.

Alex-Mann, 2024-08-05:
I have a draft MR going here: https://github.com/fvutils/pyhdl-if/pull/5.

My current issue is that the read back from Cocotb is not returning the value, but instead the async Event obj. Here is the code snippet:

```Python
    for i in range(64):
        wr_val = (i+1)
        print(f'[Py] writing: {wr_val}')
        await init_bfm.write(0x8000_0000+(4*i), wr_val)
        rd = await init_bfm.read(0x8000_0000+(4*i))
        print(f'[Py] readback: {rd}')
```

And when trying to print `rd` I get below:

```bash
[Py] writing: 1
write
--> write
<-- write
--> read
<-- read (addr,data): 0,1 
[Py] readback: <<Event at 0xffffa6e9fda0>.wait() at 0xffffa6ec4490> # Should just be 1
```

Link to the line of code + MR [here](https://github.com/fvutils/pyhdl-if/pull/5/files#diff-bd7f23cebd88b3c9519bca91f2fd84661028679b864a51438a16904396247d12R39)

ktbarrett, 2024-08-05:
Also, it doesn't seem like this is an issue with cocotb. There is the Gitter and there is pyhdl's issue page...

Alex-Mann, 2024-08-17:
This is now possible with @mballance's library. I have added an example that works with verilator here: https://github.com/fvutils/pyhdl-if/pull/5

Closing this issue.

closed 2024-08-17 by Alex-Mann