cocotb/cocotb #1990: PLUSARGS support on GHDL missing

ktbarrett, 2020-07-19:
GHDL's implementation of `vpi_get_vlog_info` does not return an `argc` and `argv`, so we cannot implement PLUSARGS for GHDL. While the use of plusargs in VHDL is not possible, it is a standard way to pass arguments to a testbench for parameterizing tests, so it is still useful IMO.

This requires knowledge of Ada, see the implementation [here](https://github.com/ktbarrett/ghdl/blob/99b542c849311c92e87e2c70d283de133c9d4093/src/grt/grt-vpi.adb#L1561). It shouldn't be that difficult if you know the language.

garmin-mjames, 2020-11-18:
One step closer: ghdl/ghdl#1511
Plusargs are still rejected as invalid command line options, though.

garmin-mjames, 2020-11-21:
🎉  https://github.com/cocotb/cocotb/runs/1436503460?check_suite_focus=true#step:22:4010

closed 2020-11-21 by garmin-mjames