cocotb/cocotb #1995: Failed to load PLI object file libcocotbvpi_modelsim.dll

you2quan, 2020-07-20:
when i run the example, my command is : make SIM=questa

(base) PS E:\git\cocotb\examples\simple_dff> make SIM=questa
/usr/bin/make results.xml
make[1]: Entering directory '/e/git/cocotb/examples/simple_dff'
set -o pipefail; MODULE=test_dff TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
/c/questasim64_2020.2/win64/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 2020.2

# do sim_build/runsim.do
# QuestaSim-64 vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap -c
# ** Warning: vmap will not overwrite local modelsim.ini.
# QuestaSim-64 vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap work sim_build/work
# Modifying modelsim.ini
# QuestaSim-64 vlog 2020.2 Compiler 2020.04 Apr 19 2020
# Start time: 22:41:39 on Jul 20,2020
# vlog -work work "+define+COCOTB_SIM" -sv -timescale 1ns/1ps -mfcu "+acc" E:/git/cocotb/examples/simple_dff/dff.sv
# -- Compiling module dff
#
# Top level modules:
#       dff
# End time: 22:41:39 on Jul 20,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -pli "C:/users/paulyuan/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" sim_build/work.dff
# Start time: 22:41:39 on Jul 20,2020
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vsim-3865) Due to PLI being present, global +acc is being enabled automatically. This will cause your simulation to run very slowly. Please use vsim -no_autoacc to disable this feature. This feature is now deprecated and will be removed from future releases.
# //  Questa Sim-64
# //  Version 2020.2 win64 Apr 19 2020
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading sv_std.std
# Loading work.dff(fast)
# Loading C:/users/paulyuan/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll
# ** Error: (vsim-3193) Load of "C:/users/paulyuan/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" failed: DLL dependent library not found.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "C:/users/paulyuan/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll".
#    Time: 0 us  Iteration: 0  Root: /
# Error loading design
Error loading design
# End time: 22:41:50 on Jul 20,2020, Elapsed time: 0:00:11
# Errors: 2, Warnings: 1
/c/users/paulyuan/miniconda3/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:132: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 12
make[1]: Leaving directory '/e/git/cocotb/examples/simple_dff'
c:\users\paulyuan\miniconda3\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
(base) PS E:\git\cocotb\examples\simple_dff>

ktbarrett, 2020-07-20:
Possibly related to #1978.

you2quan, 2020-07-21:
> Possibly related to #1978.

thank you, but after i  renamed the MINGW folder, the problem is not solved.

2 folder changed:
C:\MinGW   is changed to  C:\MinGW_bk
C:\questasim64_10.6c\gcc-4.5.0-mingw64vc12\mingw is changed to C:\questasim64_10.6c\gcc-4.5.0-mingw64vc12\mingw_bk

garmin-mjames, 2020-07-21:
> # ** Error: (vsim-3193) Load of "C:/users/paulyuan/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" failed: DLL dependent library not found.

Is there a missing path? You could try something like Process Monitor to see which DLL(s) couldn't be loaded.

you2quan, 2020-07-22:
file is ok：
[image]

how to use Process Monitor

garmin-mjames, 2020-08-12:
You can try using Process Monitor to see which DLLs are not found when starting cocotb.

https://docs.microsoft.com/en-us/sysinternals/downloads/procmon

garmin-mjames, 2020-08-12:
@you2quan Your log shows **/c/questasim64_2020.2** is being used, while you renamed the MinGW folder in **C:\questasim64_10.6c**.

ktbarrett, 2020-09-05:
@you2quan did you resolve the issue?

closed 2020-10-26 by ktbarrett