cocotb/cocotb #1996: Add Verilator to CI suite

garmin-mjames, 2020-07-20:
We have Icarus Verilog already, and we support Verilator as a simulator.

It's free software. Is there any reason we couldn't add it?

ktbarrett, 2020-07-27:
Verilator supplies nightly docker images so implementation should be easy. The issue with verilator is that regressions are terribly terribly slow, I'm not entirely sure why. Someone mentioned sim resolution could be increased to 1ns to increase performance. Travis has no build timeouts except the global one, which is 50 minutes, if that's an issue.

cmarqu, 2020-07-27:
I thought the timescale problem should be solved with https://github.com/verilator/verilator/pull/2253, but maybe there is still something to do?

ktbarrett, 2020-07-27:
The makefiles specify 1ps by default as a commandline argument. I'm not sure if that overrides the time precision set in the file.

PR #2009 (2020-08-20): Verilator regression
Adds Verilator to Travis CI regression. Added just enough exceptions to the regressions to allow them to finish, there are still several failures. Closes #1996.
closed 2020-08-20 by ktbarrett