cocotb/cocotb #2004: Hi @imphil @Matthewar

anuragece, 2020-07-24:
Hi @imphil @Matthewar 
I'm stuck on the same point and also followed the same above steps still I'm getting the same error.
Attached runsim.do.

**AttributeError: Can not find Root Handle (Module_Top)
Cocotb initialisation failed - exiting**

Please help to resolve this issue.

Thanks
Anurag
[runsim.txt](https://github.com/cocotb/cocotb/files/4966054/runsim.txt)

_Originally posted by @anuragece in https://github.com/cocotb/cocotb/issues/758#issuecomment-662998479_

eric-wieser, 2020-07-24:
Your log says "work.ORAN_Top" but your error message says "Module_Top". Is this your problem?

anuragece, 2020-07-24:
Thanks, @eric-wieser for a quick reply, Actually I was edited ORAN_Top to Module_top on the post.
I'm getting an error:

AttributeError: Can not find Root Handle (ORAN_Top)
Cocotb initialisation failed - exiting

anuragece, 2020-07-24:
This type of error i've seen the first time even I 've done cocotb simulation for some other projects previously, That was working fine.
I'm not able to understand what is the exact problem.
Please help me for resolving this issue.
Thanks

eric-wieser, 2020-07-24:
Can you show your makefile?

anuragece, 2020-07-24:
Sure. Please find the attachment.
Thanks

[Makefile.txt](https://github.com/cocotb/cocotb/files/4971943/Makefile.txt)

eric-wieser, 2020-07-24:
Which version of cocotb are you using?

anuragece, 2020-07-24:
(base) [rrhbng1@msrrhbng1 ~]$ cocotb-config -v
1.2.0

eric-wieser, 2020-07-24:
That's two releases behind now, and there have been lots of build-related issues fixed since then. I recommend you try updating.

anuragece, 2020-07-24:
Okay, I will update. but my older project is working fine on the same version.

Thanks

anuragece, 2020-07-24:
Hi @eric-wieser, Is there any direct command to update new version of cocotb?

Thanks

ktbarrett, 2020-07-24:
@anuragece `pip install --upgrade cocotb`

anuragece, 2020-07-24:
Thanks @ktbarrett . after upgrading I'm gating following error:

/home/rrhbng1/anaconda3/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:121: Including Makefile.inc from a user makefile is a no-op and deprecated. Remove the Makefile.inc inclusion from your makefile, and only leave the Makefile.sim include.
/mnt/data0/home/rrhbng1/anaconda3/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:51: *** "Unable to locate command >iverilog<".  Stop.

I'm using questasim simulator. Please help to resolve this issue.
Thanks

ktbarrett, 2020-07-24:
@anuragece As the message said, as of cocotb 1.4 you no longer need to include the `include $(cocotb-config --makefile)/Makefile.inc` line in your project, but that's just a warning. The second warning is because you aren't feeding it a value for `SIM` so it assumes Icarus. Run the project with `make SIM=questa` or add `SIM := questa` to your makefile.

anuragece, 2020-07-24:
Thanks @ktbarrett . I'm getting another error. i'm attaching Makefile. Please have a look and let me know where I'm doing wrong?
[Makefile.txt](https://github.com/cocotb/cocotb/files/4972408/Makefile.txt)

 **Error: (vopt-13130) Failed to find design unit ORAN_Top.
#         Searched libraries:
#             work**

Regards
Anurag

ktbarrett, 2020-07-24:
Well I don't know what your HDL looks like, or the `Makefile.src` you include, so I don't think I can be any more helpful. vopt is not finding that entity in your work library. Did you typo the name? Is it in some other library?

anuragece, 2020-07-24:
In makefile.src, I've put HDL files related to ORAN_Top.
Even my older working project also is showing the same error, After upgrading the cocotb version.

eric-wieser, 2020-07-24:
Are you able to run the examples from the cocotb repo?

anuragece, 2020-07-24:
Not Tried.

anuragece, 2020-07-24:
Hi @eric-wieser , cocotb repo also not working the same error is coming.

**Error: (vopt-13130) Failed to find design unit ORAN_Top.
#         Searched libraries:
#             work**

Thanks

ktbarrett, 2020-07-24:
@anuragece So you are trying to run regression in the cocotb repo by using

```
cd cocotb/
make SIM=questa
```

And it's printing what you posted? That sounds like you set `TOPLEVEL` to be an environment variable and it's polluting your other runs.

anuragece, 2020-07-24:
Hi @ktbarrett , cocotb repo is working actually i did some mistake.
but still my design is not working.
**Error: (vopt-13130) Failed to find design unit ORAN_Top.
#         Searched libraries:
#             work**

Thanks

anuragece, 2020-07-28:
Hi @garmin-mjames , @ktbarrett , @eric-wieser ,

Please help to resolve this issue.
Thannks

ktbarrett, 2020-10-26:
@anuragece Did you ever resolve the issue? Is it possible you could upload more of the code?

closed 2020-12-03 by github-actions[bot]