cocotb/cocotb #2008: Verilator hangs if the simulator exist abnormally

ktbarrett, 2020-07-28:
For example issue_1279 test hangs indefinitely after deadlock is invoked. test_exit_error fails after encountering the syntax error that causes cocotb init to fail.

ktbarrett, 2020-07-28:
I'll consider this fixed when Verilator is fixed, or a work around is added to cocotb, *not* just when test are exempted.

nredd, 2020-08-02:
@ktbarrett Can you add some logs produced during these failures? I'm currently debugging a similar hanging issue with some Verilator testbenches--not sure if this helps, but all hangs have logs similar to the following: 
```640.80ns DEBUG    cocotb.gpi                          simulatormodule.cpp:186  in handle_gpi_callback             Returning to Python
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:373  in _event_loop                     Trigger fired: ReadWrite()
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:434  in _event_loop                     1 pending coroutines for event ReadWrite()
                                                                                                                        Scheduler._do_writes
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:444  in _event_loop                     Scheduling coroutine Scheduler._do_writes
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:757  in schedule                        Scheduling with Value(ReadWrite())
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:764  in schedule                        Coroutine Scheduler._do_writes yielded <<Event at 0x7f269409cd10>.wait() at 0x7f2693d62af0> (mode 1)
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:447  in _event_loop                     Scheduled coroutine Scheduler._do_writes
   640.80ns DEBUG    cocotb.scheduler                           scheduler.py:466  in _event_loop                     All coroutines scheduled, handing control back to simulator
   640.80ns DEBUG    cocotb.gpi                          simulatormodule.cpp:244  in handle_gpi_callback             Returning to simulator
   640.80ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:98   in run_callback                    Generic run_callback done
```

nanduraj1, 2020-08-17:
also, is it possible to return non-zero exit code during error. I see that even on questasim the error code is zero. This will help my jenkins setup.

nredd, 2020-08-17:
@nanduraj1 cocotb doesn't return exit codes--you need to parse the `results.xml` files directly. I would look at `cocotb/bin/combine_results.py` to see how the cocotb team does this internally :slightly_smiling_face:

ktbarrett, 2020-08-20:
@nredd You can see the Verilator results [in CI](https://travis-ci.org/github/cocotb/cocotb/jobs/719530500) now.

PR #2105 (2020-12-06): Update the verilator main control loop.
After some discussion in #2096, here's a stab at improving the verilator main control loop.

Closes #2008. 

Uses solutions for verilator/verilator#2536 and verilator/verilator#2589, so this will bump the minimum supported version of Verilator to 4.106.

- [x] Update supported version in docs
- [x] Newsfragment for better support and version bump

---

- Don't call eval() on every time step, instead skip ahead to the next
registered Timer trigger.

    - All values should be settled after the inner loop, and verilator ignores delays (#) in SystemVerilog  [... cut]
closed 2020-12-06 by ktbarrett