cocotb/cocotb #2010: cocotb/share/lib/embed/gpi_embed.cpp:36:19: fatal error: cassert: No such file or directory

andrewhee, 2020-07-28:
when i run pip install cocotb, there is gcc error occurred such as:
Collecting cocotb
  Using cached cocotb-1.4.0.tar.gz (226 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/python3/bin/python3.8 /usr/local/python3/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpmt9ftbcd
       cwd: /tmp/pip-install-has2fu6a/cocotb
  Complete output (104 lines):
  Modelsim/Questa executable (vdbg) executable not found. No FLI interface will be available.
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.8
  creating build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/scoreboard.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/__init__.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/outcomes.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/scheduler.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/_py_compat.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/clock.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/bus.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/ipython_support.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/memdebug.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/config.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/binary.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/xunit_reporter.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/_version.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/decorators.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/wavedrom.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/handle.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/utils.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/log.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/regression.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/result.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/ANSI.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/triggers.py -> build/lib.linux-x86_64-3.8/cocotb
  copying cocotb/_os_compat.py -> build/lib.linux-x86_64-3.8/cocotb
  creating build/lib.linux-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/__init__.py -> build/lib.linux-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/xgmii.py -> build/lib.linux-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/avalon.py -> build/lib.linux-x86_64-3.8/cocotb/monitors
  creating build/lib.linux-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/__init__.py -> build/lib.linux-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/xgmii.py -> build/lib.linux-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/opb.py -> build/lib.linux-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/avalon.py -> build/lib.linux-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/amba.py -> build/lib.linux-x86_64-3.8/cocotb/drivers
  creating build/lib.linux-x86_64-3.8/cocotb/generators
  copying cocotb/generators/__init__.py -> build/lib.linux-x86_64-3.8/cocotb/generators
  copying cocotb/generators/bit.py -> build/lib.linux-x86_64-3.8/cocotb/generators
  copying cocotb/generators/byte.py -> build/lib.linux-x86_64-3.8/cocotb/generators
  copying cocotb/generators/packet.py -> build/lib.linux-x86_64-3.8/cocotb/generators
  creating build/lib.linux-x86_64-3.8/cocotb/share
  creating build/lib.linux-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.inc -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.sim -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.deprecations -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles
  creating build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.ghdl -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.riviera -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.modelsim -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.xcelium -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.activehdl -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.vcs -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
  copying cocotb/share/makefiles/simulators/Makefile.cvc -> build/lib.linux-x86_64-3.8/cocotb/share/makefiles/simulators
[... 27 lines cut ...]
  creating build/temp.linux-x86_64-3.8/cocotb
  creating build/temp.linux-x86_64-3.8/cocotb/share
  creating build/temp.linux-x86_64-3.8/cocotb/share/lib
  creating build/temp.linux-x86_64-3.8/cocotb/share/lib/gpi_log
  gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Icocotb/share/include -I/usr/local/python3/include/python3.8 -c cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.linux-x86_64-3.8/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -D__STDC_FORMAT_MACROS
  creating build/lib.linux-x86_64-3.8/cocotb/libs
  g++ -pthread -shared -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib build/temp.linux-x86_64-3.8/cocotb/share/lib/gpi_log/gpi_logging.o -Lbuild/lib.linux-x86_64-3.8/cocotb/libs -o build/lib.linux-x86_64-3.8/cocotb/libs/libgpilog.so -Wl,-rpath,$ORIGIN
  building 'cocotb/libs/libcocotbutils' extension
  creating build/temp.linux-x86_64-3.8/cocotb/share/lib/utils
  gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Icocotb/share/include -I/usr/local/python3/include/python3.8 -c cocotb/share/lib/utils/cocotb_utils.cpp -o build/temp.linux-x86_64-3.8/cocotb/share/lib/utils/cocotb_utils.o -std=c++11 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -D__STDC_FORMAT_MACROS
  g++ -pthread -shared -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib -L/usr/local/openssl/lib -Wl,-rpath,/usr/local/openssl/lib build/temp.linux-x86_64-3.8/cocotb/share/lib/utils/cocotb_utils.o -Lbuild/lib.linux-x86_64-3.8/cocotb/libs -lgpilog -o build/lib.linux-x86_64-3.8/cocotb/libs/libcocotbutils.so -Wl,-rpath,$ORIGIN
  building 'cocotb/libs/libcocotb' extension
  creating build/temp.linux-x86_64-3.8/cocotb/share/lib/embed
  gcc -pthread -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DPYTHON_SO_LIB=libpython3.8.so -Icocotb/share/include -I/usr/local/python3/include/python3.8 -c cocotb/share/lib/embed/gpi_embed.cpp -o build/temp.linux-x86_64-3.8/cocotb/share/lib/embed/gpi_embed.o -std=c++11 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -D__STDC_FORMAT_MACROS
  cocotb/share/lib/embed/gpi_embed.cpp:36:19: fatal error: cassert: No such file or directory
   #include <cassert>
                     ^
  compilation terminated.
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

How could I do, my friends, thanks a lot

eric-wieser, 2020-07-28:
What does `gcc --version` give you?

andrewhee, 2020-07-29:
my new version of gcc doesn't work, I will check my gcc install process, thanks guys.

closed 2020-07-29 by andrewhee