cocotb/cocotb #2014: Support custom types in VHDL

garmin-mjames, 2020-08-07:
Add support for bringing VHDL types into Python.

The obvious ones are enumeration and physical types.

**Enumeration type**
```vhdl
type MULTI_LEVEL_LOGIC is (LOW, HIGH, RISING, FALLING, AMBIGUOUS);
```
This would allow using the names LOW, HIGH, etc in Python, rather than 0, 1, etc.

**Physical type**
```vhdl
type DISTANCE is range 0 to 1E16
    units
    -- primary unit:
        Å;               --angstrom

    -- metric lengths:
        nm   = 10 Å;     --nanometer
        um   = 1000 nm;  --micrometer (or micron)
        mm   = 1000 um;  --millimeter
        cm   = 10 mm;    --centimeter
        m    = 1000 mm;  --meter
        km   = 1000 m;   --kilometer

    -- English lengths:
        mil  = 254000 Å; --mil
        inch = 1000 mil; --inch
        ft   = 12 inch;  --foot
        yd   = 3 ft;     --yard
        fm   = 6 ft;     --fathom
        mi   = 5280 ft;  --mile
        lg   = 3 mi;     --league
end units DISTANCE;
```
Something like `distance(1, units="mi")` would be neat.

---
**GPI**

Given an object handle, `vhpi_handle(vhpiBaseType, obj_hdl)` gives the base type.
Next, `vhpi_get(vhpiKindP, type_hdl)` will give us `vhpiEnumTypeDeclK` or `vhpiPhysTypeDeclK` for these types.

For Enumeration types, we then use `vhpi_iterator(vhpiEnumLiterals, type_hdl)` to iterate through the enumeration literals. Each `vhpiEnumLiteralK` object provides a `vhpiStrValP` property containing the string identifier (LOW, HIGH, etc in above example type), as well as a `vhpiPositionP` property containing the integer for the enum literal (this is currently used to read/write to enumeration signals). It also provides a `vhpiNumLiteralsP` property that contains the count of literals.

Physical types are more complicated, and I haven't fully explored them yet. But there is a `vhpiUnitDeclK` object per defined unit, and every secondary unit has a `vhpiPhysLiteralK` object associated with it that defines it in terms of other units. The `vhpiBaseUnit` association from the `vhpiPhysTypeDeclK` object gives the `vhpiUnitDeclK` object representing the primary unit.
So the mapping can be built up.

One limitation that I haven't found a way around is that to get to the type declaration, you have to have an object of the type to start with. So a constant/signal/variable/generic/port of the type you want.

xref #1233

ktbarrett, 2020-08-07:
> One limitation that I haven't found a way around is that to get to the type declaration, you have to have an object of the type to start with.

I think this ties into how you would obtain the type in Python as well. Is there any other reasonable way besides obtaining a handle to an object and then requesting the type?

```python
handle = dut['enum_signal']
enum = handle.get_type()
handle.value = enum.LITERAL
```

garmin-mjames, 2020-08-07:
So, `get_type()` on a handle for a Physical type would return the custom `type` that could then be used to create an object?

```python3
handle = dut['phys_signal']
distance = handle.get_type()
handle.value = distance(12, 'nm')
```

ktbarrett, 2020-08-07:
Yes. The whole thing will require a good bit of plumbing.

ktbarrett, 2020-08-11:
FLI support this via `mti_GetEnumValues` which returns a `char **`, a NULL terminated array of C strings literals.

garmin-mjames, 2020-08-11:
@ktbarrett anything for Physical types?

ktbarrett, 2020-08-11:
`mtiPhysicalDataT *mti_GetPhysicalData(mtiTypeIdT)` returns a linked list that contains all the physical type info.

Each `mtiPhysicalDataT` record
* `next` field is next unit in physical type
* `name` field is a C string
* `position` the multiple of the primary unit

The user is required to use `void mti_Free(void*)` on each struct for cleanup.