cocotb/cocotb #2015: Bad file descriptor logging error on Windows with Modelsim

devnow, 2020-08-10:
Hi. I started to use cocotb with Modelsim and around 50% of the cases I get `--- Logging error ---` with `OSError: [Errno 9] Bad file descriptor` when I run make SIM=modelsim. In this case the powershell output is scrambled but the test cases run successful. The other 50% everything works as expected.

## Installation setup
Windows 10 64-bit (v2004), miniconda 4.8.3 (python 3.8.3) and cocotb 1.4.0 installation according to [https://docs.cocotb.org/en/stable/install.html](https://docs.cocotb.org/en/stable/install.html) all in 64-bit. Modelsim DE-64 vsim 10.7a (64-bit)

## Test setup
I run cocotb\examples\dff wih `make SIM=modelsim` as an example but I experience the same issue with other/own tests as well. The test run in the Anaconda Powershell Prompt
The following section shows the back to back powershell output of two `make SIM=modelsim` calls
1. logging error with scrambled output
2. everything works as expected

The outcome is not predictable, sometime the first run of `make` is successful after clean up the build artifacts. Sometimes it's successful multiple times in a row. Sometimes it scrambles multiple times.

### Powershell output
```
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> make SIM=modelsim
/usr/bin/make results.xml
make[1]: Entering directory '/c/workdir/Arbeitspaket/2020-08-02_cocotb/examples/dff/tests'
mkdir -p sim_build
set -o pipefail; MODULE=dff_cocotb TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=vhdl \
/c/MentorGraphics/modelsim_dlx64_10.7a/win64pe/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading C:/MentorGraphics/modelsim_dlx64_10.7a/tcl/vsim/pref.tcl

# 10.7a

# do sim_build/runsim.do
# Model Technology ModelSim DE-64 vmap 10.7a Lib Mapping Utility 2018.03 Mar 27 2018
# vmap -c
# Copying C:/MentorGraphics/modelsim_dlx64_10.7a/win64pe/../modelsim.ini to modelsim.ini
# Model Technology ModelSim DE-64 vmap 10.7a Lib Mapping Utility 2018.03 Mar 27 2018
# vmap work sim_build/work
# Modifying modelsim.ini
# Model Technology ModelSim DE-64 vcom 10.7a Compiler 2018.03 Mar 27 2018
# Start time: 22:51:45 on Aug 10,2020
# vcom -work work "+acc" C:/workdir/Arbeitspaket/2020-08-02_cocotb/examples/dff/hdl/dff.vhdl
# -- Loading package STANDARD
# -- Loading package TEXTIO
# -- Loading package std_logic_1164
# -- Compiling entity dff
# -- Compiling architecture rtl of dff
# End time: 22:51:45 on Aug 10,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# ** Warning: (vsim-159) Mode option -64 is not supported in this context and will be ignored.
# vsim -onfinish exit -foreign "cocotb_init C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" sim_build/work.dff
# Start time: 22:51:45 on Aug 10,2020
# //  ModelSim DE-64 10.7a Mar 27 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim DE-64 and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
--- Logging error ---
Traceback (most recent call last):
  File "C:\programdata\miniconda3\lib\logging\__init__.py", line 1085, in emit
    self.flush()
  File "C:\programdata\miniconda3\lib\logging\__init__.py", line 1065, in flush
    self.stream.flush()
OSError: [Errno 9] Bad file descriptor
Call stack:
  File "C:\programdata\miniconda3\lib\site-packages\cocotb\__init__.py", line 188, in _initialise_testbench
    _setup_logging()
  File "C:\programdata\miniconda3\lib\site-packages\cocotb\__init__.py", line 86, in _setup_logging
[... 382 lines cut ...]
# End time: 22:53:43 on Aug 10,2020, Elapsed time: 0:00:01
# Errors: 0, Warnings: 1
make[1]: Leaving directory '/c/workdir/Arbeitspaket/2020-08-02_cocotb/examples/dff/tests'
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests>
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> conda --version
conda 4.8.3
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> cocotb-config --version
1.4.0
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> python --version
Python 3.8.3
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> vsim -version
Model Technology ModelSim DE-64 vsim 10.7a Simulator 2018.03 Mar 27 2018
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> Get-ComputerInfo  | select windowsversion,windowsproductname,osarchitecture

WindowsVersion WindowsProductName OsArchitecture
-------------- ------------------ --------------
2004           Windows 10 Pro     64-bit

(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests>
```

## Attachment
sim.log from
1. [bad_sim.log](https://github.com/cocotb/cocotb/files/5053752/bad_sim.log)
2. [good_sim.log](https://github.com/cocotb/cocotb/files/5053754/good_sim.log)

garmin-mjames, 2020-08-11:
https://github.com/cocotb/cocotb/blob/d566d19a8a400e99ab5a298844e9fa7b86940a68/cocotb/__init__.py#L58
This appears to be failing when reopening `stdout`, which then causes an Exception for every logging command.

ktbarrett, 2020-08-11:
@garmin-mjames Seems like Powershell is not reporting `isatty` consistently. Notice on the good results there is no messages saying it reopened the output with buffering.

@devnow Have you tried cmd.exe or another terminal emulator to see if ti fails there as well?

eric-wieser, 2020-08-11:
We ran into this all the time at my old employer, but never came close to finding the cause.

eric-wieser, 2020-08-11:
> Perhaps we should restore them using `sys.__stdout__`/`sys.__stderr__` on failure?

If that function failed, then there is nothing to restore - the very last operation is `setattr`.

devnow, 2020-08-11:
@all thanks for looking into it. I have the same issue when I started with cmd instead of powershell. I had the same issue on my first installation with mingw64 directly (no miniconda) with the bash shell.

I can't reproduce a clean make with the `COCOTB_LOG_LEVEL=DEBUG` enabled it's always scrambles (tested it ~6 times) the output 
[sim_debug.log](https://github.com/cocotb/cocotb/files/5055191/sim_debug.log)
But the simulation is successful in Modelsim.
[results.xml.txt](https://github.com/cocotb/cocotb/files/5055193/results.xml.txt)
[transcript.txt](https://github.com/cocotb/cocotb/files/5055194/transcript.txt)
One I had to hit Ctrl+C to stop because it didn't progress.
[bad_debug_CtrlC_sim.log](https://github.com/cocotb/cocotb/files/5055196/bad_debug_CtrlC_sim.log)

devnow, 2020-08-11:
In a couple of hours I will try to install the newest version of ModelSim DE 2020.3 to check if it has the same behaviour.

devnow, 2020-08-11:
I did run the test with Modelsim 2020.3 (64-bit) and had the same issue.
With `COCOTB_LOG_LEVEL=DEBUG` the output is always scrambled
[sim_debug_modelsim2020.3.log](https://github.com/cocotb/cocotb/files/5055366/sim_debug_modelsim2020.3.log)
Without it can be okay
[sim_good_modelsim2020.3.log](https://github.com/cocotb/cocotb/files/5055376/sim_good_modelsim2020.3.log)
or bad
[sim_bad_modelsim2020.3.log](https://github.com/cocotb/cocotb/files/5055379/sim_bad_modelsim2020.3.log)

It's a minor issue because the internal simulations are working okay. We inherited a design and now we try to automate tests with cocotb/jenkins. It would be nice to have a clean output in the build artefacts when we have to track down a failed test. 

I'm impressed by all the work team cocotb is doing. Writing tests with python is much more pleasant and easier to track. The separation of the test form the framework may allow us to switch simulators in the future.

ktbarrett, 2020-08-11:
I just tried running Questa on Linux and it does reopen the streams with buffering successfully. This is likely a Windows, or a Windows + Modelsim issue.

One last thing to try is run the vsim command that runs the simulation on the command line directly, and not within vsim as the makefile does. If you have already built the library you can run:

```command
MODULE=pythontests COCOTB_LOG_LEVEL=DEBUG vsim -c -onfinish exit -foreign "cocotb_init C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" sim_build/work.dff
```

eric-wieser, 2020-08-11:
> This is likely a Windows, or a Windows + Modelsim issue.

Windows 10 + modelsim was how this manifested for me in the past.

devnow, 2020-08-11:
> One last thing to try is run the vsim command that runs the simulation on the command line directly, and not within vsim as the makefile does. If you have already built the library you can run:
> 
> ```shell
> MODULE=pythontests COCOTB_LOG_LEVEL=DEBUG vsim -c -onfinish exit -foreign "cocotb_init C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" sim_build/work.dff
> ```

I not sure what to do. I tried the follow but modelsim gives an error. 
```
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> $MODULE = "dff_cocotb"
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> $COCOTB_LOG_LEVEL = "DEBUG"
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> vsim -c -onfinish exit -foreign "cocotb_init C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" sim_build/work.dff
Reading C:/MentorGraphics/modelsim_dlx64_10.7a/tcl/vsim/pref.tcl

# 10.7a

# vsim -c -onfinish exit -foreign "cocotb_init C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" sim_build/work.dff
# Start time: 18:59:38 on Aug 11,2020
# //  ModelSim DE-64 10.7a Mar 27 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim DE-64 and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading sim_build/work.dff(rtl)
# Loading C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll
# ** Error: (vsim-3193) Load of "C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll" failed: DLL dependent library not found.
# ** Warning: (vsim-FLI-3160) Failed to load FLI object file "C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll".
#    Time: 0 ns  Iteration: 0  Instance: /dff File: C:/workdir/Arbeitspaket/2020-08-02_cocotb/examples/dff/hdl/dff.vhdl
VSIM 1>

```

I tried  `$MODULE = "pythontests"` but it's the same.
I copied all DLLS from the `libcocotbfli_modelsim.dll` folder into the folder with the `Makefile` and changed to a relative path. The out on the powershell is:
```
(base) PS C:\workdir\Arbeitspaket\2020-08-02_cocotb\examples\dff\tests> vsim -c -onfinish exit -foreign "cocotb_init libcocotbfli_modelsim.dll" sim_build/work.dff
Reading C:/MentorGraphics/modelsim_dlx64_10.7a/tcl/vsim/pref.tcl

# 10.7a

# vsim -c -onfinish exit -foreign "cocotb_init libcocotbfli_modelsim.dll" sim_build/work.dff
# Start time: 19:30:08 on Aug 11,2020
# //  ModelSim DE-64 10.7a Mar 27 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim DE-64 and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'C:\\MentorGraphics\\modelsim_dlx64_10.7a\\win64pe\\vsimk.exe'
  sys.base_prefix = 'C:\\ProgramData\\Miniconda3'
  sys.base_exec_prefix = 'C:\\ProgramData\\Miniconda3'
  sys.executable = 'C:\\MentorGraphics\\modelsim_dlx64_10.7a\\win64pe\\vsimk.exe'
  sys.prefix = 'C:\\ProgramData\\Miniconda3'
  sys.exec_prefix = 'C:\\ProgramData\\Miniconda3'
  sys.path = [
    'C:\\ProgramData\\Miniconda3\\python38.zip',
    '.\\DLLs',
    '.\\lib',
    'C:\\MentorGraphics\\modelsim_dlx64_10.7a\\win64pe',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000007fc (most recent call first):
<no Python frame>
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading sim_build/work.dff(rtl)
# Loading ./libcocotbfli_modelsim.dll
#      -.--ns INFO     cocotb.gpi                         ..ib\fli\FliImpl.cpp:1063 in cocotb_init                     cocotb_init called
#
#
#
#      -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
#
# End time: 19:30:08 on Aug 11,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
```
But then it stops the execution without return to Modelsim or Powershell shell. Ctrl+C doesn't recover as well, but it detects it in the shell with a `# Break key hit`. I had to close the Shell window.

devnow, 2020-08-11:
I just run `make SIM=modelsim COCOTB_LOG_LEVEL=DEBUG`  a couple of times and once it did work without the logging error. See 
[sim.log](https://github.com/cocotb/cocotb/files/5058586/sim.log)

eric-wieser, 2020-08-11:
I just noticed that in your original good vs bad log, none of these lines appear:
```
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading sim_build/work.dff(rtl)
# Loading C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll
#      -.--ns INFO     cocotb.gpi                         ..ib\fli\FliImpl.cpp:1063 in cocotb_init                     cocotb_init called
# 
# 
# 
#      -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# 
#      -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:105  in gpi_print_registered_impl       FLI registered
# 
#      -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
```
These lines come from within modelsim itself, so this is looking like it might be a bug in modelsim.

eric-wieser, 2020-08-11:
It looks to me like the difference between good vs bad is whether modelsim prints the following before or after our logging starts:
```
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading sim_build/work.dff(rtl)
# Loading C:/programdata/miniconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll
```

eric-wieser, 2020-08-11:
@ktbarrett, I think we could perhaps learn something by checking the return code of `fprintf` here: https://github.com/cocotb/cocotb/blob/d566d19a8a400e99ab5a298844e9fa7b86940a68/cocotb/share/lib/gpi_log/gpi_logging.cpp#L131-L136

devnow, 2020-08-11:
@eric-wieser @ktbarrett just let me know when you want me to test an other configuration

manunited10, 2021-10-13:
Hey guys. Just wondering if the issue was ever resolved or not.
I've been having the same issue for months now. The issue randomly appears and disappears.
I use cocotb 1.4 on Windows 10 with modelsim 10.7b.

kcuzner, 2022-02-01:
This may not work for everyone, but I worked around this problem in cocotb 1.6 by simply redirecting the logger to use `mti_PrintMessage`:

```python
import logging
from cocotb.log import SimLogFormatter, SimTimeContextFilter

class ModelsimLogHandler(logging.Handler):
    """
    This works around a very stupid and hard-to-debug issue documented here:
    https://github.com/cocotb/cocotb/issues/2015

    This is a brute-force solution which instead of actually fixing the
    problem merely avoids it by redirecting output into modelsim's window
    via the appropriate FLI call.
    """

    def __init__(self):
        super().__init__()
        import ctypes

        mti = ctypes.WinDLL("mtipli.dll")

        def wrapper(fn, restype, *argtypes):
            fn.restype = restype
            fn.argtypes = argtypes
            return fn

        self._mti_PrintMessage = wrapper(mti.mti_PrintMessage, None, ctypes.c_char_p)

    def emit(self, record):
        try:
            msg = self.format(record)
            self._mti_PrintMessage(msg.encode("latin-1"))
        except RecursionError:  # Copy-paste from logging library
            raise
        except Exception:
            self.handleError(record)

def fix_modelsim_log_bug():
    hdlr = ModelsimLogHandler()
    hdlr.addFilter(SimTimeContextFilter())
    hdlr.setFormatter(SimLogFormatter())
    logging.getLogger().handlers = [hdlr]
```

I call `fix_modelsim_log_bug` at the start of each of my tests. Putting it directly in the file to be executed sequentially wouldn't work since the logger is reset between the time the python is parsed and when the first test case runs.

manunited10, 2022-02-02:
Thanks @kcuzner.
I have put this new class and `fix_modelsim_log_bug` funcion  on top of my main test module. Then I called the funcion before anything else.
I don't see the errors I used to see (a few pages of error as the OP put in this page). However, I still get this error once in a while (almost as frequent as getting the long error):
```
Exception ignored in: <_io.TextIOWrapper name=1 mode='w' encoding='cp1252'>
OSError: [Errno 9] Bad file descriptor
```
This error is in OP's log as well.
The error comes after a simple `print` function. I commented that out and it goes to another `print` (on unsuccessful runs).
While some googling says the error is due to writing to a file which is already closed, I checked my entire code and there is no such thing. Besides 50-60% of the time it runs successfully (not a permanent error).

I'm on modelsim 10.7b and still using cocotb 1.4 and desperate to find a solution.

ktbarrett, 2022-02-02:
A good solution is to set up the cocotb logger to go to a custom file and not stdout. Not sure why anyone would want to two mixed. In ModelSim messages from cocotb and messages from HDL aren't even synchronized.

kcuzner, 2022-02-02:
> This error is in OP's log as well.
The error comes after a simple print function. I commented that out and it goes to another print (on unsuccessful runs).
While some googling says the error is due to writing to a file which is already closed, I checked my entire code and there is no such thing. Besides 50-60% of the time it runs successfully (not a permanent error).

Print by default will write to stdout, which has likely been screwed up by modelsim, so that's why it's failing. You need to replace your `print` calls with calls to `mti_PrintMessage` or as @ktbarrett suggests, write to a file instead (there is a `file` argument to `print` that can be used for this, [see the docs](https://docs.python.org/3/library/functions.html#print)).

kcuzner, 2022-02-02:
> A good solution is to set up the cocotb logger to go to a custom file and not stdout. Not sure why anyone would want to two mixed.

@ktbarrett the reason I want them mixed is because my modelsim runs sometimes occur on a headless server and the log is uploaded to a web interface with a single log view. It would be very unergonomic for me to require my coworkers to read two logs to understand why their simulation failed on the server. Even if the messages aren't correlated, it's still better than requiring users to read two logs.

> In ModelSim messages from cocotb and messages from HDL aren't even synchronized.

Yeah, if both python and modelsim are writing to stdout (file descriptor 1) that makes tons of sense b/c each one has their own separate buffers. However, with my calls to `mti_PrintMessage` I suspect that it just gets thrown into modelsim's own buffer. I wouldn't go so far as to claim it'll be perfectly synchronized, but I can imagine a situation where it's more synchronized than simply writing to stdout in parallel with modelsim.

manunited10, 2022-02-02:
> > This error is in OP's log as well.
> > The error comes after a simple print function. I commented that out and it goes to another print (on unsuccessful runs).
> > While some googling says the error is due to writing to a file which is already closed, I checked my entire code and there is no such thing. Besides 50-60% of the time it runs successfully (not a permanent error).
> 
> Print by default will write to stdout, which has likely been screwed up by modelsim, so that's why it's failing. You need to replace your `print` calls with calls to `mti_PrintMessage` or as @ktbarrett suggests, write to a file instead (there is a `file` argument to `print` that can be used for this, [see the docs](https://docs.python.org/3/library/functions.html#print)). Or, you can replace your `print` with calls to `logging.info` or something.

Thank you for your quick replay @kcuzner 
Is there a way to force all `print()` function to write to a file without editing the source? something system level or something else? The reason is the test benches are a bit large and they use several other classes even from other repos. It's not possible to change all `prints` immediately.

kcuzner, 2022-02-02:
@manunited10 seems like something like this would be your best bet: https://stackoverflow.com/a/4675744/1124529

ktbarrett, 2022-02-02:
I would recommend [these docs](https://docs.python.org/3/library/logging.handlers.html) for moving stdout logs to a file.

closed 2023-02-11 by github-actions[bot]

ktbarrett, 2026-04-09:
Ah, I see. A file that could be used to communicate until we know we have properly spawned the subprocess. That's a reasonable idea. We dont really have something set up for that yet, but we probably should.