cocotb/cocotb #2017: cocotb 1.4 bug in scoreboard with hexdiffs and hexdumps

hjwmedenblik, 2020-08-14:
Since release 1.4 where strings passed to `cocotb.utils.hexdump()` and `cocotb.utils.hexdiffs()` are deprecated, I found out that I experienced issues with the scoreboard results when tests fail. 
Although all my tests are updated to work with bytes instead of string data, it seems that inside the scoreboard code the `got` and `exp` data is still internally converted to strings before the `cocotb.utils.hexdump()` and `cocotb.utils.hexdiffs()` are called. It results in a hanging behaviour of the simulator. The same applies for the log.info where expected and received data is logged inside this piece of code.

eric-wieser, 2020-08-14:
Nice catch, the offending lines are:

https://github.com/cocotb/cocotb/blob/8a37f94188b338f577db19ded92d8ff39afc7fd9/cocotb/scoreboard.py#L120-L155

eric-wieser, 2020-08-14:
Frankly I don't even understand what that code was trying to do... Converting arbitrary objects to `str` and then hex-dumping them is nonsense.

eric-wieser, 2020-08-14:
> It results in a hanging behaviour of the simulator.

That surprises me - can you show a stack trace?

hjwmedenblik, 2020-08-14:
Hi Eric. It does not crash but I do not get a response back. The most common cause for that is the tremendous useless code :
`            if not isinstance(exp, str):
                try:
                    for word in exp:
                        log.info(str(word))
                except Exception:
                    pass
`

hjwmedenblik, 2020-08-14:
btw... the issue also rises in the part next to `        # Compare directly
        if got != exp:`

ktbarrett, 2020-08-14:
@eric-wieser Right. It makes sense in some cases: `BinaryValue` and `str` (assuming it's coming from a fixed length string signal), but in general it's nonsense. Perhaps the scoreboard should be changed to be more general by doing less: it's easy enough to just use `repr()` and let the user compare the values.

If we want to support rich comparison behavior we would need to introduce a comparison API for all values going into scoreboard.

eric-wieser, 2020-08-14:
>Hi Eric. It does not crash but I do not get a response back. 

That surprises me. I don't see any way for the scoreboard to cause a hang without throwing an exception.

eric-wieser, 2020-08-14:
> The most common cause for that is the tremendous useless code :

That code does indeed look dumb, but I don't think it would cause a hang either.

ktbarrett, 2020-08-14:
@hjwmedenblik Which simulator are you using?

meds7, 2020-08-14:
> @hjwmedenblik Which simulator are you using?

Questa

hjwmedenblik, 2020-08-14:
Oops, I have two accounts...
It is Questa
@eric-wieser , @ktbarrett , if you direct me how to trace/debug this behavior then I am willing to help you guys.

ktbarrett, 2020-08-15:
@hjwmedenblik Are you driving some of the design in HDL as well as cocotb? For example, running a clock in HDL? Could you create a minimal example that reproduces the hang and post it here?

hjwmedenblik, 2020-08-15:
@ktbarrett  Hi Kaleb, no, that is just the reason why I switched to cocotb. Everything is handled from cocotb, stimulus, clock, reset, modelling, checking. I will try if I can make a test case

hjwmedenblik, 2020-08-16:
@ktbarrett, @eric-wieser  debugging with gdb reveals that I cannot pass along:
 https://github.com/cocotb/cocotb/blob/8a37f94188b338f577db19ded92d8ff39afc7fd9/cocotb/utils.py#L327-L331

And only when strings are passed there. I isolated it by setting breakpoints before and after.

eric-wieser, 2020-08-16:
How big are the strings it's diffing?

hjwmedenblik, 2020-08-16:
@eric-wieser Maybe it is more convenient to post the data here:

exp : 

b'\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abc\x1a\xcf\xfc\x1d\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abc\x1a\xcf\xfc\x1d\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abc\x1a\xcf\xfc\x1d\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abc\x1a\xcf\xfc\x1d\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f\x10\x11\x12\x13\x14\x15\x16\x17\x18\x19\x1a\x1b\x1c\x1d\x1e\x1f !"#$%&\'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abc\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\ [... cut ...] 00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\''

eric-wieser, 2020-08-16:
What you're seeing is just that your strings are very long (1780 bytes), and `hexdiffs` is rather slow. It's not hanging, it's just taking a very long time:
```python
In [136]: t = time.time(); d = hexdiffs(exp, got); dt = time.time() - t; dt
Out[137]: 19.25763201713562  # seconds
```
This shouldn't be surprising, hexdiffs uses a very naive `N**2` algorithm.
It's copied directly from `scapy` at https://github.com/secdev/scapy/blob/dae5581279db261fc257b1b649c599f2585b2012/scapy/utils.py#L293.

hjwmedenblik, 2020-08-16:
@eric-wieser that is strange because I waited minutes. I would not have come to conclusions after only 19 seconds.

ktbarrett, 2020-08-16:
@hjwmedenblik @eric-wieser Running `hexdiffs` on strings is *much* slower than with bytes, that accounts for the difference in performance. That 19s is probably closer to several minutes to several hours with strings. Strings shouldn't be going to hexdiffs or hexdump, they both expect bytes.

ktbarrett, 2020-08-16:
@hjwmedenblik @eric-wieser The use of `str` is causing serious problems. Passing bytes to `str()` returns the representation, *it does not decode the string*. The resulting string is not equivalent to the original bytes, for example `b'\x00\x01\x02\x03'` becomes the string `"b'\x00\x01\x02\x03'"` which has 19 characters. Your `exp` value as a string is 6083 characters long. This is not only incorrectly, but explodes that O(n**2) algo and make it seem to hang.

We need to fix scoreboard.

hjwmedenblik, 2020-08-16:
@ktbarrett : great to see how you explain what happens and your clarification for the hang behavior. Yes I remember your comments on removing Avalon components as well.

But... why remove all this stuff when they can be very useful. It is just a matter of fixing issues or adding missing options. Several month ago i was notified to use Cocotb for testing all my vhdl blocks and it changed my life seriously. It has been so helpful so far. 
Please be aware that the Cocotb docs also rely heavily on examples with Avalon interfaces and Scoreboard. So removing all that stuff is confusing for someone who just starts with Cocotb.

Of course I am always curious what other options you suggest and in the meantime I will use my patched avalon components and scoreboard.

ktbarrett, 2020-08-17:
> Several month ago i was notified to use Cocotb for testing all my vhdl blocks and it changed my life seriously. 

Glad to hear that 😄. I had a similar experience when I first started using it.

> in the meantime I will use my patched avalon components and scoreboard.

If you have something that works, good, keep using it.

> It is just a matter of fixing issues or adding missing options

We need tests for these interfaces and for the testbenching components, or they will continue to break. The interfaces are not trivial and the maintainers are not experts on them (I have never used any of the interfaces, besides AXI). We just want to give the responsibility of upkeep for those interfaces to someone who understands them well, and can develop them independently of this repo. The fact that the interfaces are in this repo is a crutch, now people expect support when we can't offer it. If you have designs and tests you can open source that exercise an interface, we will gladly take it. Or if you want to take responsibility of an interface that you use often, [there is a process](https://docs.cocotb.org/en/stable/extensions.html). Popular and useful extensions can even be hosted in the cocotb organization.

eric-wieser, 2020-08-17:
> If we want to support rich comparison behavior we would need to introduce a comparison API for all values going into scoreboard.

This feels like a problem we should delegate to pytest, which already has a fair number of comparison tools, source here:

ktbarrett, 2020-08-17:
@eric-wieser Note that the scoreboard allows for accumulation of comparisons, and mismatches don't have to immediately end a test (which is very useful, more than one instance of a mismatch can help diagnose an issue). I'm not sure what pytest offers in this regard. 

The Scoreboard is built like it would be in UVM where it's an independent thread of execution, while pytest seems to be built around the assumption there is a single thread of execution. They are 2 different testing methodologies. We support both testing methodologies at our company with our own framework. That's the main reason I'd like to see a cocotb entry point and segregation of tests from the scheduler, so we can swap testing frameworks.

eric-wieser, 2020-08-17:
I'm not suggesting we use pytest as the test runner any time soon, but it's full of utility functions that are generally useful to testing - so maybe we should start taking a dependency on it and using bits and pieces.

bluecmd, 2020-08-29:
Hi,

This is hitting me as well. I am using scoreboard and AvalonST as per the EndianSwapper example.
A simple repro here: https://gist.github.com/bluecmd/73de5fdffb85ff95733cc13dcb84726a

The interesting part of the output:
```
[..]
 50000.00ns INFO     cocotb.fejkon_fc_debug.st_in                  avalon.py:219  in _monitor_recv                   Received a packet of 7 bytes
 50000.00ns INFO     cocotb.fejkon_fc_debug.st_in                    test.py:26   in model                           b'\x00\x01\x02\x03\x04\x05\x06'
 60000.00ns WARNING  cocotb.scoreboard.fejkon_fc_debug         scoreboard.py:86   in result                          Still expecting 1 transactions on AvalonSTPkts(st_out)
/home/bluecmd/.local/lib/python3.6/site-packages/cocotb/scoreboard.py:89: DeprecationWarning: Passing a string to hexdump is deprecated, pass bytes instead
  (index, hexdump(str(transaction))))
 60000.00ns INFO     cocotb.scoreboard.fejkon_fc_debug         scoreboard.py:89   in result                          Expecting 0:
                                                                                                                     0000   62 27 5C 78 30 30 5C 78 30 31 5C 78 30 32 5C 78   b'\x00\x01\x02\x
                                                                                                                     0010   30 33 5C 78 30 34 5C 78 30 35 5C 78 30 36 27      03\x04\x05\x06'
                                                                                                                     
 61000.00ns ERROR    cocotb.regression                         regression.py:396  in _score_test                     Test Failed: test_generation (result was TestFailure)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/home/bluecmd/fejkon/ip/fejkon_fc_debug/test.py", line 67, in test_generation
                                                                                                                         raise tb.scoreboard.result
                                                                                                                     cocotb.result.TestFailure: Not all expected output was received
[..]
```

FWIW, this patch seems to work fine for me:

```patch
--- /home/bluecmd/.local/lib/python3.6/site-packages/cocotb/scoreboard.py.bak	2020-08-29 15:22:48.180885300 +0200
+++ /home/bluecmd/.local/lib/python3.6/site-packages/cocotb/scoreboard.py	2020-08-29 15:22:52.796778900 +0200
@@ -86,7 +86,7 @@
                                  (len(expected_output), str(monitor)))
                 for index, transaction in enumerate(expected_output):
                     self.log.info("Expecting %d:\n%s" %
-                                  (index, hexdump(str(transaction))))
+                                  (index, hexdump(transaction)))
                     if index > 5:
                         self.log.info("... and %d more to come" %
                                       (len(expected_output) - index - 1))
```

ktbarrett, 2020-09-29:
The `Scoreboard` has been deprecated. We worked out ways to make the scoreboard work for the afflicted users. If you have anymore issues with the scoreboard, ask away on the [gitter](https://gitter.im/cocotb/Lobby).

PR #2041 (2020-09-05): Deprecate Scoreboard and generators
Fixes #2017. I don't have a test, seeing as we aren't currently testing the scoreboard, and may move it out of repo soon, I'm not sure if that's necessary. I will make one if it is desired.

@hjwmedenblik This should fix the issue you were seeing with hexdiff.
@bluecmd This should remove the DeprecationWarning you are seeing.

Could you both please test it?
closed 2020-09-29 by ktbarrett