cocotb/cocotb #2030: Bad DLL format

wdembinski, 2020-08-20:
I'm trying to run simulation of one of examples: dff. The version of coctb is 1.3.2.
Unfortunately I'm getting following errors:
# ** Error: (vsim-3193) Load of "C:/users/wdembins/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" failed: Bad DLL format.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "C:/users/wdembins/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll".
I've followed steps listed in the Installation Guide. Did I miss something? Shall I check anything? Actually the message error doesn't tell me much.

imphil, 2020-08-20:
Can you please re-try with the latest version 1.4.0 of cocotb? Much changed around this topic in the latest release.

wdembinski, 2020-08-20:
Sorry, I did a mistake in the original post. The version of cocotb is 1.4.0. The examples package I was using was for 1.3.2 but I just replaced with 1.4.0. Still getting the same error messages. I think I've installed production version of cocotb, is it possible to enable debug or I need to install debug version of the package? Maybe debug version would print some more logs?

ktbarrett, 2020-08-20:
The only thing I could find on Google is [this](https://forums.xilinx.com/t5/Simulation-and-Verification/Bad-DLL-Format-when-loading-libxil-vsim-dll-in-modelsim/td-p/398549). Which seems to imply the issue is that you build cocotb with a 64 bit compiler and are trying to load it in a 32-bit Modelsim. Could that be the case?

imphil, 2020-08-20:
Can you let us know a bit more about the system you're using?

- Which modelsim version? 32 or 64 bit?
- Which Windows version? 32 or 64 bit?

ktbarrett, 2020-08-29:
@wdembinski Was your problem resolved?

ktbarrett, 2020-09-05:
@wdembinski If you are still having an issue, and can provide more information, feel free to reopen the issue.

closed 2020-09-05 by ktbarrett

mkhazraee, 2020-10-10:
Hi all,

I'm running to a similar issue. I've installed free version of modelsim, which seems to be only available for Windows. I figured how to run the Makefile from bash on windows, and the modelsim runs nicely, but I still get this error:
Load of "C:/home/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_modelsim.so" failed: Bad DLL format.

I'm using latest cocotb (1.4.0), modelsim 10.4a, 64-bit windows. As I understood the problem is modelsim is 32bits but the rest of system and cocotb are 64bits. The only way around is to install i386 architecture and run cocotb on 32bit version of python, or there is another way around? (in issue [396](https://github.com/cocotb/cocotb/issues/396) it says it should be solved now, so thought they might be a better work around. -64 flag doesn't help either). 

Moein

garmin-mjames, 2020-10-10:
@mkhazraee See https://docs.cocotb.org/en/stable/install.html for Windows installation instructions. Install 32-bit conda and use that for the Python environment. https://docs.conda.io/en/latest/miniconda.html

mkhazraee, 2020-10-11:
Thanks a lot, this worked right off the bat on windows cmd, without the bash on windows.