cocotb/cocotb #2043: Verilator SVA support broken?

nredd, 2020-09-03:
I "have" a `cocotb==1.4.0` testbench targeting Verilator (4.099 devel rev v4.040-72-g627d83e8)   
```
module dut(<toplevel>);
    cool_design #(<params>) (<wiring>);
endmodule: dut
```
where `cool_design` contains some basic assertions to e.g. enforce design constraints. Sim-specific variables are disabled by default, as are SV assertions in Verilator, but supplying `COMPILE_ARGS=--assert` (ref #1538) to an otherwise-successful testbench raises an exception like `AttributeError: dut contains no object named cool_design`. Supplying `--assert` only broke a few of the designs I testbenched (the remaining passed) regardless of whether or not they even contained `assert();` statements, which makes me think it's an upstream issue (probably in cocotb-Verilator VPI bindings? not too familiar with those bits)

ktbarrett, 2020-09-04:
Could you give an example of a test in the regression that works without "--assert", but breaks with it? Or provide a test with asserts that does fail?

nredd, 2020-09-04:
Sure, I'll come back to this soon

ktbarrett, 2020-11-17:
@nredd, I would watch #2093.

closed 2020-11-17 by ktbarrett