cocotb/cocotb #2051: Verilator freezes with AvalonSTMonitor on multiple tests after await

bluecmd, 2020-09-07:
Hi,

I am playing with Verilator and cocotb.

I have made this minimal repro: https://gist.github.com/bluecmd/fe70d5279bfcaca470c6556ccd502933.

Log:
```
make results.xml
make[1]: Entering directory '/home/blue/cocotb-freeze'
MODULE=test TESTCASE= TOPLEVEL=thing TOPLEVEL_LANG=verilog \
        sim_build/thing 
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Verilator version 4.100 2020-09-07
     0.00ns INFO     cocotb                                      __init__.py:209  in _initialise_testbench           Running tests with cocotb v1.4.0 from /usr/local/lib/python3.7/dist-packages/cocotb                                                                                                                                                                                  
     0.00ns INFO     cocotb                                      __init__.py:229  in _initialise_testbench           Seeding Python random module with 1599498748
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test.test_a
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test.test_b
     0.00ns INFO     cocotb.regression                         regression.py:463  in _start_test                     Running test 1/2: test_a
     0.00ns INFO     cocotb.test.test_a.0x7b12a02146d8         decorators.py:256  in _advance                        Starting test: "test_a"
                                                                                                                     Description: None
 52000.00ns INFO     cocotb.regression                         regression.py:361  in _score_test                     Test Passed: test_a
 52000.00ns INFO     cocotb.regression                         regression.py:463  in _start_test                     Running test 2/2: test_b
 52000.00ns INFO     cocotb.test.test_b.0x7b12a0228b00         decorators.py:256  in _advance                        Starting test: "test_b"
                                                                                                                     Description: None
[Freeze]
```

- I am using Verilator 4.100
- I set up two identical tests creating two clocks and one AvalonSTMonitor
- If I run a single test using `make TESTCASE=test_a` either test works.
- If I make the test only use a single clock, running both tests works.
- If I initialize the AvalonSTMonitor before calling await on anything both tests work.
- Using `make SIM=icarus` works fine.

I would prefer to be able to initialize AvalonSTMonitor after doing reset to ensure the previous test state has been cleared.

ktbarrett, 2020-09-07:
Could you re-run with `COCOTB_LOG_LEVEL=DEBUG COCOTB_SCHEDULER_DEBUG=1` and upload the log file here?

bluecmd, 2020-09-07:
Here you go: [freeze.log](https://github.com/cocotb/cocotb/files/5184511/freeze.log)

ktbarrett, 2020-09-08:
I had to remove the `timescale` directive for the timing to match up with Icarus's. Icarus doesn't support `timescale`. Still hangs at the same place regardless.

Anyway, the main difference I can see is that Verilator schedules the clock's Timers before the Timer in the `run` Task. This means the clocks are updated before the AvalonMonitor starts. On Icarus, the `run` task's Timer is scheduled first, so the AvalonMonitor starts before the clocks are updated. 

The first thing the AvalonMonitor does is synchronize with the clock with RisingEdge(clk). On Verilator, because the clocks update first, this Risingedge won't trigger until the next clock. On Icarus, because the clocks are updated after Monitor, this causes the RisingEdge(clk) to trigger immediately.

This implies if I add something (say, ReadOnly) to push the Monitor task to run after the clocks are updated that Icarus should fail as well. However, it does not.

I don't know why this scheduling difference is leading to issues. It's almost definitely a Verilator issue. I really can't make head or tails of it.

bluecmd, 2020-09-08:
@ktbarrett Ok, thanks for your debugging! I filled a bug with Verilator to see if they can help debug this further.

garmin-mjames, 2020-12-07:
@bluecmd Can you re-test this with current master?

bluecmd, 2021-03-07:
Let's assume it is.

closed 2021-03-07 by bluecmd