cocotb/cocotb #2053: 1.4.0 not marked as latest release

bluecmd, 2020-09-08:
Hi,

Right now 1.3.2 is marked as the latest release. Is that correct? Technically it's released after 1.4.0 but this leads to that the GitHub landing page for cocotb says the latest release is 1.3.2 - which maybe isn't what is intended?

[image]

imphil, 2020-09-08:
1.3.2 was released after 1.4.0, but it's a separate branch. I have not seen anything on GitHub allowing me to say "this release is newer, but on a stable branch, so don't consider it the latest release". Looks like a GitHub missing feature that we can't do anything about.

themperek, 2020-09-08:
Some ugly hack (not sure if should do this): https://stackoverflow.com/questions/22822586/swap-latest-release-on-github

bluecmd, 2020-09-08:
Seems unreasonable :-/

closed 2020-09-08 by bluecmd

ktbarrett, 2020-09-08:
We will be releasing 1.4.1 not too far in the future, so 1.4.X will be back on top.