cocotb/cocotb #2054: Cached writes in ReadWrite callback

garmin-mjames, 2020-09-08:
There is some weirdness around doing a cached write to a signal while in the `ReadWrite` callback.

`Scheduler._do_writes` unconditionally waits on `ReadWrite`:

https://github.com/cocotb/cocotb/blob/8744492c8af4aaa73681362f86fe62be6665bb4d/cocotb/scheduler.py#L254-L261

Additionally, `Scheduler._do_writes` is always run as the first coroutine during the `ReadWrite` callback, if it is waiting on that trigger and not the `Event`.

I've noticed that running **Active-HDL** in VHPI, the `ReadWrite` isn't called again in the same time step unless something else is scheduled to happen. So if you `await ReadWrite` before doing a cached write, it may not apply the write if nothing else is scheduled to cause an additional delta cycle.

---
Here are two testcases I added to **test_timing_triggers.py**:
```python3

async def write_in_readwrite(dut):
    dut.stream_in_valid.setimmediatevalue(0)
    dut.stream_out_ready.setimmediatevalue(0)

    async def write_delayed():
        await ReadWrite()
        dut.stream_in_valid.value = 1

    @cocotb.coroutine
    async def wait_signal():
        await RisingEdge(dut.stream_in_valid)
        dut.stream_out_ready.value = 1

    write_coro = cocotb.fork(write_delayed())

    await cocotb.triggers.with_timeout(wait_signal(), timeout_time=10, timeout_unit='ns')

    await ReadOnly()
    assert dut.stream_out_ready.value == 1

@cocotb.test()
async def test_write_in_readwrite_no_do_writes(dut):
    """Test doing a normal (delayed) write after awaiting ReadWrite trigger.

    Don't fork _do_writes() task before-hand.
    """
    await write_in_readwrite(dut)

@cocotb.test()
async def test_write_in_readwrite(dut):
    """Test doing a normal (delayed) write after awaiting ReadWrite trigger.

    Fork _do_writes() task by doing a delayed write. This ensures that _do_writes() runs
    before write_delayed() when the ReadWrite() trigger fires.
    """
    dut.stream_in_data.value = 0
    await write_in_readwrite(dut)

```
With **Active-HDL 11.1**,
Both tests pass in VPI:
```
     0.00ns INFO     *******************************************************************************************************************
                     ** TEST                                                       PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     *******************************************************************************************************************
                     ** test_timing_triggers.test_write_in_readwrite_no_do_writes    PASS            0.00          0.00         0.67  **
                     ** test_timing_triggers.test_write_in_readwrite                 PASS            0.00          0.00         1.01  **
                     *******************************************************************************************************************
```
But they both fail in VHPI:
```
    20.00ns INFO     *******************************************************************************************************************
                     ** TEST                                                       PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     *******************************************************************************************************************
                     ** test_timing_triggers.test_write_in_readwrite_no_do_writes    FAIL           10.00          0.08       129.04  **
                     ** test_timing_triggers.test_write_in_readwrite                 FAIL           10.00          0.00      4000.31  **
                     *******************************************************************************************************************
```

garmin-mjames, 2020-09-08:
More succinctly, this fails in VHPI but not VPI:
```python3
@cocotb.test()
async def test_readwrite_in_readwrite(dut):
    """Test that ReadWrite can be awaited while in a ReadWrite callback."""
    @cocotb.coroutine
    async def multi_readwrite():
        await ReadWrite()
        await ReadWrite()

    await cocotb.triggers.with_timeout(multi_readwrite(), timeout_time=1, timeout_unit='ns')
```

ktbarrett, 2020-09-08:
With Questa both the VPI and FLI pass.

ktbarrett, 2020-09-08:
GHDL, Verilator, and Icarus all pass.

garmin-mjames, 2020-09-08:
Hmm, after a bit more investigation, this looks like an Active-HDL bug, but it only happens when these tests are run first. If they are in the middle of other tests they are passing. Weird.

ktbarrett, 2024-10-09:
This should be fixed by applying writes done in the ReadWrite phase immediately. Closed by #4115. If not, reopen.

closed 2024-10-09 by ktbarrett