cocotb/cocotb #2057: Document how to contribute to cocotb

imphil, 2020-09-09:
We should provide documentation how to get started with contributing to cocotb. The audience here are people who have not contributed to cocotb before, potentially not even to open source projects. They might have an "itch to scratch," or might just be looking for an interesting project to work on over the summer.

Examples:
- https://developer.mozilla.org/en-US/docs/Mozilla/Developer_guide/Introduction
- https://rustc-dev-guide.rust-lang.org/getting-started.html
- https://librecores-web.readthedocs.io/en/latest/contributing.html
- https://numpy.org/doc/stable/dev/

ktbarrett, 2020-09-29:
So I'm trying to identify the information missing from CONTRIBUTING.md.

- _what_ to work on
    - link to roadmap?
    - any issue that isn't assigned
    - `good first issue`
    - ask a maintainer
    - "continuous improvement"-type codebases (things that don't need buy-in and can always be improved without worrying about scope)
        - documentation
        - tests / coverage / CI
- getting help
    - gitter
    - ask a maintainer
    - mentoring? (I would be up for that)
- how to set up a development environment (#2058)
- development process
    - running tests / building docs locally 
    - link to newsfragment docs
- breakdown by codebase
    - skill prereqs
    - link to documentation / internal documentation
    - "experts"?

One thing I see mentioned a lot is large "projects". We don't really have such a concept, so I left it out. Our codebase is not that big.

ktbarrett, 2021-01-19:
Can we consider this done?

ktbarrett, 2021-03-07:
On second thought, I feel like some of this could still be improved. Some of the things I mentioned were either glossed over or not mentioned at all. There was also an action to move this into the RTD.