cocotb/cocotb #2059: [tracking] Redesign BinaryValue

imphil, 2020-09-09:
BinaryValue has a fair number of issues which cannot be solved in a backwards-compatible way. We are in the process of coming up with a new design, this issue tracks the progress on it.

ktbarrett, 2020-09-10:
# Minimal Requirements
* Accurately model...
    * (System)Verilog 4-value logic type
    * VHDL std_logic and bit
* Vectors of that type (???)
    * Is this a true requirement? Icarus supports single element access now. This might be a quality of life thing.
    * When getting or setting a vector of logic, we could instead yield or take a list of logic elements.

## Quality of life
* Model...
    * (System)Verilog packed logic arrays
    * VHDL std_logic_vector
* Slicing
* Logic operations
* Arbitrary bounds
* Generic / nestable vectors to support N dimensional arrays (???)
* Convert Python integers to/from 1D vectors of logics
    * Two's complement
    * Unsigned
    * Signed magnitude (???)
    * One's complement (???)
* Convert Python strings to/from 1D vectors of logics

## Antifeatures of BinaryValue
* "endianness"
    * confusing, does it mean to/downto range or byte endianness?
    * unnecessary since the new type won't deal in bytes
* associating integer representation with the object
    * associate with the conversion operator
    * have implicit conversions (like in `handle <= 1`) infer either two's complement or unsigned
* "buff", conversion to/from bytes
    * convert to/from integer and use `int.to_bytes` and `int.from_bytes` instead
* integer operators
    * convert to an integer first and get well defined behavior...

## Backwards compatibility
* COCOTB_RESOLVE_X (???)

eric-wieser, 2020-09-10:
> integer operators
> * convert to an integer first and get well defined behavior...

One think that might be neat is if `b'010X + b'10X0 = b'11XX` - but maybe that's not what verilog does, and maybe that logic belongs elsewhere anyway

ktbarrett, 2020-09-10:
I'd have to look into it. Considering how adders work, all bits more significant than an X should be an X, but that's not how they are typically modeled are they?

And I am trying to stay away from the replacement being a modeling datatype, we can let other libraries handle that. The type just needs to be a way to express logic values and offer conversions to different representations.

eric-wieser, 2020-09-10:
> Considering how adders work, all bits more significant than an X should be an X, but that's not how they are typically modeled are they?

My interpretation was that a computation with `X`s can be performed by computing with every combination of `0` and `1`, and taking the union of all results, replacing conflicts with `X`. In my example, the carry chain can never reach the top bits, even if both Xs on the left are `1`.

I'm surprised to see that 1800-2017 does not appear to specify the behavior of addition on `x` values, or even mention that it can occur. 6.9.1 says that "Vectors shall obey laws of arithmetic modulo-2 to the power n", but that's all I can find.

> The type just needs to be a way to express logic values and offer conversions to different representations.

My thinking was that if possible we want to emulate the verilog /vhdl behavior of the value. Users can always layer higher-level modelling (fixed-point integers, named bits, etc) on top of that in a separate library, but it would be nice if we could match the semantics of the underlying HDL. Maybe we just can't do that, if VPI / VHPI doesn't expose enough to detect the types...

ktbarrett, 2020-09-10:
> I'm surprised to see that 1800-2017 does not appear to specify the behavior of addition on x values, or even mention that it can occur.

That's why I'm hesitant about supporting integer operations on logic vectors. Different language, different language standards, different simulators, etc. can behave differently and I don't want to have to investigate and support all those differences.

We can also choose to be conservative now and add more features later. It's a lot easier to add something than take it away.

> Maybe we just can't do that, if VPI / VHPI doesn't expose enough to detect the types...

They do expose information about types, the limitation is in the GPI.

jshaker000, 2021-01-16:
My biggest requests for a new Binary value replacement would be

1. Define isresolvable() as some top level function that can be called on ints as well as 'BinaryValue2s', rather than being a member function
2. Overload &, |, <<, >>, ^ for BinaryValue, keeping with Verilog FourValue Semantics
3. Perhaps overload arithmetic & comparison, where a nonresolvable value on one side and a resolvable value (or an int) on the other returns an X

I think (1) and (2) would already give a lot of milage. If (3) is too hard I think its less of a want than the others.

eric-wieser, 2021-01-16:
I don't really understand the value of `isresolvable(int)`, which would presumably always return `True`.

ktbarrett, 2021-01-16:
@eric-wieser Presumably for writing generic code that can work with both integer and logic vectors.

dlmiles, 2023-07-21:
Is this issue related to this:

```
assert BinaryValue(63).binstr == "111111"                   # good
assert BinaryValue(6).binstr == "110"                       # good

assert str(BinaryValue(63, bigEndian=True)) == "111111"     # good
assert str(BinaryValue(6, bigEndian=True))  == "110"        # good: agree big-endian, sane default

# default bigEndian=True confirm
assert str(BinaryValue(63)) == "111111"                     # good
assert str(BinaryValue(6))  == "110"                        # good: agree big-endian, sane default

assert str(BinaryValue(63, bigEndian=False)) == "111111"    # bad: probably bad can't tell
assert str(BinaryValue(6, bigEndian=False))  == "110"       # bad: this is not little-endian

assert str(BinaryValue(63, bigEndian=False, n_bits=7)) == "0111111" # bad: this is not little-endian
assert str(BinaryValue(6, bigEndian=False, n_bits=7))  == "0000110" # bad: this is not little-endian

assert str(BinaryValue(63, bigEndian=True, n_bits=7)) == "1111110"  # bad: this is not, it did mirror
assert str(BinaryValue(6, bigEndian=True, n_bits=7))  == "1100000"  # bad: nonsense behaviour

assert str(BinaryValue(BinaryValue(63).binstr)) == "111111"         # good: probably good can't tell
assert str(BinaryValue(BinaryValue(6).binstr))  == "110"            # good: looks big-endian, sane default

assert str(BinaryValue(BinaryValue(63).binstr, n_bits=7)) == "1111110"      # bad: this is not big-endian
assert str(BinaryValue(BinaryValue(6).binstr, n_bits=7)) == "1100000"       # bad: this is not big-endian, nonsense behaviour

assert str(BinaryValue(63, bigEndian=True, n_bits=8)) == "11111100"         # bad: this is not big-endian
assert str(BinaryValue(6, bigEndian=True, n_bits=8))  == "11000000"         # bad: this is not big-endian, more nonsense at 8bits

assert str(BinaryValue(63, bigEndian=True, n_bits=32)) == "11111100000000000000000000000000" # bad: this is not big-endian
assert str(BinaryValue(6, bigEndian=True, n_bits=32))  == "11000000000000000000000000000000" # bad: this is not big-endian

assert str(BinaryValue(63, bigEndian=False, n_bits=8)) == "00111111"        # bad: this is not little-endian
assert str(BinaryValue(6, bigEndian=False, n_bits=8))  == "00000110"        # bad: this is not little-endian

assert str(BinaryValue(63, bigEndian=False, n_bits=32)) == "00000000000000000000000000111111" # bad: this is not little-endian
assert str(BinaryValue(6, bigEndian=False, n_bits=32))  == "00000000000000000000000000000110" # bad: this is not little-endian

```

PR #3634 (2024-01-21): Remove `BinaryValue`
Closes #2059.

Partial rework of #3244. That PR does a *lot* more than simply remove `BinaryValue`. This PR intends only to remove `BinaryValue` and update documentation and tests related to `BinaryValue` and nothing more.

To accomplish that without changing a ton of code I had to add support for comparison with `int` to `LogicArray` and `Logic`. As a consequence this means I had to remove the hashability of `Logic`, which I don't think should affect anyone. Follow-ons might work a little more on this.

I think it's unfortunate, but unfortunately necessary, to do these cr [... cut]
closed 2024-01-21 by ktbarrett