cocotb/cocotb #2062: [Tracking] Improve internal documentation

ktbarrett, 2020-09-09:
This is related to #2057.

Cocotb's guts are fairly large and complicated and this complexity with no documentation is off putting to new contributors. We need in-depth design justifications and details for internal components. The best place for this information to reside is on the Wiki, since it does not require pull requests and reviews to contribute.

Some internal sections that need more detail:
- [ ] the GPI and it's implementations
- [ ] the coroutine system (scheduler, triggers, "decorators")
- [ ] Python simulator object handles / GPI simulator object types
- [ ] the build system
- [ ] how the C++ libraries fit together

cmarqu, 2020-09-10:
We should probably write the Wiki pages in Restrctured Text instead of the default Markdown so that it's easier to move to Sphinx later in case we want to.

garmin-mjames, 2020-09-10:
Relevant wiki page: https://github.com/cocotb/cocotb/wiki/cocotb-Internals