cocotb/cocotb #2070: save simulator state (rewind to a previoiusly set point)

bedros, 2020-09-12:
is there a way to save simulator state, and then later restore it

Thanks

ktbarrett, 2020-09-12:
What are you trying to achieve?

bedros, 2020-09-12:
Some TBs require long time to setup the Dut, if we save state of simulator after setup, we can restore and iterate quickly over random seeds

ktbarrett, 2020-09-12:
You could recursively discover and dump the values of all signals in a design, then apply all of those signals back. See the [`test_recursive_discover`](https://github.com/cocotb/cocotb/blob/4f4f6a9eda034678007f958ceea4385d94cead61/tests/test_cases/test_discovery/test_discovery.py#L37-L46) test for example of recursive discovery.

themperek, 2020-09-14:
@bedros Some inspiration for you: https://github.com/themperek/apbi2c_cocotb_example/blob/parallel/tb/checkpoint.py . Not sure if this works still.
You can use pickle or anything else to save to file. 

Maybe worth to make an extension.

closed 2020-11-17 by bedros