cocotb/cocotb #2071: Undefined function 'Py_Initialize'

ktbarrett, 2020-09-13:
Hello, I would like to ask for help in this issue as well. My system settings:

Ubuntu 18.04
Questasim 64-bit 10.5b_1
cocotb 1.4.0
Python 3.6

* cocotb was installed using: pip3 install cocotb
* Installation of cocotb was done after the installation of questa

I get the exact same error when trying to run the adder example (make SIM=questa):

```
# Loading sv_std.std
# Loading work.adder(fast)
# Loading /tmp/unknown@b5cd0c757113_dpi_2237/linux_x86_64_gcc-4.7.4/STUB_SYMS_OF_libcocotbvpi_modelsim.so
# Loading /usr/local/lib/python3.6/dist-packages/cocotb/libs/libcocotbvpi_modelsim.so
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# ** Fatal: (vsim-12005) Undefined function 'Py_Initialize' introduced from '/usr/local/lib/python3.6/dist-packages/cocotb/libs/libcocotb.so' is being called. Exiting ... 
# End time: 21:42:33 on Sep 13,2020, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
```

_Originally posted by @jrri in https://github.com/cocotb/cocotb/issues/1758#issuecomment-691732378_

ktbarrett, 2020-09-13:
@jrri Could I get you to run the following and post the output?

```command
cd /usr/local/lib/python3.6/dist-packages/cocotb/libs/
ldd ./libcocotb.so
```

jrri, 2020-09-14:
Sure @ktbarrett , and thanks for the reply. Here is the output:
```
eda_dev@e6b382de9c2d:/usr/local/lib/python3.6/dist-packages/cocotb/libs$ ldd ./libcocotb.so
	linux-vdso.so.1 (0x00007ffe594b9000)
	libpython3.6m.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0 (0x00007ffa0c558000)
	libgpilog.so => /usr/local/lib/python3.6/dist-packages/cocotb/libs/./libgpilog.so (0x00007ffa0c355000)
	libcocotbutils.so => /usr/local/lib/python3.6/dist-packages/cocotb/libs/./libcocotbutils.so (0x00007ffa0c153000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ffa0bf34000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ffa0bb43000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007ffa0b911000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ffa0b6f4000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ffa0b4f0000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007ffa0b2ed000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ffa0af4f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ffa0ce0c000)
eda_dev@e6b382de9c2d:/usr/local/lib/python3.6/dist-packages/cocotb/libs$ 
```

ktbarrett, 2020-09-14:
I do see *a* problem, I'm not sure if it's *the* problem?

This line has libpython pointing to the system Python, but you seem to be running a user-managed Python installation out of `/usr/local`. What kind of installation is this? Did you compile it yourself?
```
libpython3.6m.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0 (0x00007ffa0c558000)
```

jrri, 2020-09-14:
No, I did not compile it myself, installed it through "sudo apt-get python3" after that the directory /usr/local/lib/python3.6/dist-packages/ (empty) is created and that seems to be the location that cocotb chooses. That is the only installation of Python I have done.

Looking at comments in the last answer of:

https://unix.stackexchange.com/questions/476971/ldd-shows-no-location-after-arrow-library-does-not-exist-on-system

Mentions the dynamic linker (/lib64/ld-linux-x86-64.so.2) not being in the same path as the other libraries, i.e.,  /lib/x86_64-linux-gnu/

Do you think something similar may have something to do?

ktbarrett, 2020-09-15:
The dynamic linker is a system library, that's not a problem. 

I just checked, that implies you installed cocotb using sudo? `sudo pip install cocotb`? That's bad practice, but it shouldn't be causing an issue. Maybe you gave your user permissions on that directory? On my Debian box, `pip install cocotb` does a user install into `~/.local/`. None of that should matter.

It's not complaining about not being able to find a libpython to load, but it's not resolving the names. Could you run the following on a simple test and upload the output?

```command
LD_DEBUG=libs make SIM=questa
```

jrri, 2020-09-15:
Yes, you are right, I installed with a root user. If I use non-root user it goes into ~/.local as you said. 

Now, for the output of LD_DEBUG=libs make SIM=questa, see attached file, as it is too big to copy it here.
[log.txt](https://github.com/cocotb/cocotb/files/5228210/log.txt)

ktbarrett, 2020-10-26:
@jrri Did you ever resolve your issue? I haven't had time to look into it further.

jrr-xena, 2020-11-17:
Hi, Unfortunately I didn't resolve the issue

themperek, 2020-11-17:
Can you try local conda installation?

ktbarrett, 2020-12-18:
@jrri I suggest trying again after #2268 is merged, it should resolve the issue.

closed 2020-12-18 by ktbarrett