cocotb/cocotb #2072: Verilator Double-Triggering

davidlenfesty, 2020-09-17:
I've run into a bug where rising edges of clocks will double-trigger, when using Verilator as a simulator.

Here is a minimum reproduction of the design that causes the issue:

```Verilog
module test(input wire clk);
    reg clkdiv2;
    always @(posedge clk) begin
        clkdiv2 <= ~clkdiv2;
    end
endmodule
```

See [this gist](https://gist.github.com/davidlenfesty/0bdc98ac7f5fa031a446f04f6ab0ca9f) for my full test setup.

I can't seem to get the example to trigger at all in iverilog, so I don't entirely know if it's Verilator or CocoTB. I'm assuming iverilog is optimizing out `clkdiv2` but I'm not sure how to fix that.

gretzteam, 2020-09-18:
Could you try making clkdiv2 an output to debug? It would get rid of the "simulator optimizing wires" part of the equation.

davidlenfesty, 2020-09-18:
No dice.

```verilog
module test(input wire clk, output reg clkdiv2);
    always @(posedge clk) begin
        clkdiv2 <= ~clkdiv2;
    end
endmodule
```

Also does not give me anything in iverilog. (Also realised I may not have put enough info in the main issue, so I'm editing it for more info)

ktbarrett, 2020-09-18:
After looking at your example I see you aren't assigning a valid initial value to `clkdiv2`. `~X` is `X`. There is no edge, which is why Icarus isn't working. After adding `initial clkdiv2 = 1'b0`, both Icarus and Verilator work.

ktbarrett, 2020-09-18:
I am seeing an issue though.

This is one time step, see how the `value_test test.clkdiv2 v[0]=1/0 0x55d0406947ca 0x55d04079d0c0` test is run *twice* in the same eval step causing the callback `value_callback 0x55d04075cec8 test.clkdiv2 v[0]=1` to be called twice.

```
-V{t0,1624}+++++TOP Evaluate Vtop::eval
-V{t0,1625}+    Vtop::_eval_debug_assertions
-V{t0,1626}+ Clock loop
-V{t0,1627}+    Vtop::_eval
-V{t0,1628}+    Vtop::_combo__TOP__1
-V{t0,1629}+    Vtop::_sequent__TOP__3
-V{t0,1630}+    Vtop::_change_request
-V{t0,1631}+    Vtop::_change_request_1
-V{t0,1632}- vpi: value_test test.clkdiv2 v[0]=1/0 0x55d0406947ca 0x55d04079d0c0
-V{t0,1633}- vpi: value_callback 0x55d0407b4388 test.clkdiv2 v[0]=1
-V{t0,1634}- vpi: vpi_get_value 0x55d04074b708
-V{t0,1635}- vpi: vpi_get_value 0x55d04074b708
-V{t0,1636}- vpi: vpi_remove_cb 0x55d0407b4388
-V{t0,1637}- vpi: vpi_register_cb 1 0x55d04075cec8
-V{t0,1638}- vpi: value_test TOP.clk v[0]=1/1 0x55d0406947c8 0x55d0407184e0
-V{t0,1639}- vpi: value_test test.clkdiv2 v[0]=1/0 0x55d0406947ca 0x55d04079d0c0
-V{t0,1640}- vpi: value_callback 0x55d04075cec8 test.clkdiv2 v[0]=1
-V{t0,1641}- vpi: vpi_get_value 0x55d04074b708
-V{t0,1642}- vpi: vpi_get_value 0x55d04074b708
-V{t0,1643}- vpi: vpi_remove_cb 0x55d04075cec8
-V{t0,1644}- vpi: vpi_register_cb 1 0x55d040791b68
-V{t0,1645}- vpi: value_test TOP.clk v[0]=1/1 0x55d0406947c8 0x55d0407184e0
-V{t0,1646}- vpi: value_test test.clkdiv2 v[0]=1/1 0x55d0406947ca 0x55d04079d0c0
```

davidlenfesty, 2020-09-18:
Unfortunately can't share code for this, but I also run into an infinite loop in an error state in the actual application I'm trying to write tests for. Running `COCOTB_LOG_LEVEL=DEBUG COCOTB_SCHEDULER_DEBUG=1 make` gets me essentially to [this state](https://gist.github.com/davidlenfesty/e7d21e1ccd10ac20cab03d9daad5fd77)

I believe these are related errors, but I can't verify because I'm unsure how to replicate the issue outside of my design.

For this run I moved the clock divider (as in the above minimal example) up and out of the module, and am generating the divided clock manually.

garmin-mjames, 2020-11-30:
@davidlenfesty Can you check if this issue is still occuring? verilator/verilator#2644 may have fixed this.

davidlenfesty, 2020-11-30:
I can confirm that the issue has been fixed, running a fresh build: `Verilator 4.105 devel rev v4.104-81-gd21b4e3f`.

The infinite loop issue I believe was related has also been fixed!

closed 2020-11-30 by davidlenfesty