cocotb/cocotb #2073: to_gpi_objtype is missing vpiInterconnectNet

wimiko, 2020-09-18:
to_gpi_objtype, the function that maps the VPI object types to GPI types, doesn't decode vpiInterconnectNet (value 533).

I ran in to this problem trying to get a handle to a 'real' port of a .vams module.
Cadence Excelium returns type vpiInterconnectNet for this object.

I'm not sure what to_gpi_objtype should return here though. Is there a way to find out that it is a real?

ktbarrett, 2020-09-18:
Is it only returned as a vpiInterconnectType? That type, in and of itself, tells us nothing about the value type, just the net type. The same object can have multiple types. It might be the same issue we see with https://github.com/cocotb/cocotb/pull/1128#issuecomment-659139832.

cmarqu, 2020-09-29:
@wimiko Do you have a small example that produces a ``vpiInterconnectNet``?

wimiko, 2020-11-12:
Sorry, i will not be able to work on this for the time being (tape-out).
I will get back on this in february...