cocotb/cocotb #2074: Implement better way to extend Questa runsim.do file?

TC01, 2020-09-21:
I see from #1495 that there is a proposal to deprecate the simulator-specific ```SCRIPT_FILE``` environment variable, as part of a general cleanup of simulator-specific directives.

For Questa, ```SCRIPT_FILE``` makes it possible to insert additional directives into runsim.do between the vlog and vsim stages.  I'm currently working a project at the moment where colleagues would like to extend runsim.do in a different place-- to insert directives after the vsim stage (in order to modify the waveform generation when ```WAVES=1``` is set, which is also a simulator-specific variable).

I should say here I'm not a Questa expert-- it might be that there's a way to modify the waveform generation setup than modifying the script file itself. But assuming for the moment that this is something someone might want to do... I'm wondering what the best way to do it would be. Without actually touching cocotb's makefile infrastructure, it seems like the best way to do this at present from a project makefile would be:

* I can "make sim_build/runsim.do" to generate this script without actually running the simulation.
* So in my makefile I can create a new target (say, ```modify_runsim```) that first creates runsim.do and then modifies it by inserting additional directives as appropriate.
* Then I just change the default targets around in order to ensure that my new target runs first, and then the simulation runs without recreating the runsim script (I think).

This seems to be pretty fragile, so I feel like it's not the best way to proceed, but I also don't know what a better way would be. Any thoughts? I think it could be useful to support modifying this script (without needing to manually edit Makefile.questa), and ```SCRIPT_FILE``` doesn't necessarily cover all possible use cases, so perhaps we could come up with a standard way to do this and document it (and perhaps also get rid of this simulator specific environment variable in the process). An alternative of course would be to add separate environment variables for each stage of the simulation, but that would increase rather than decrease the number of simulator-specific variables...

ktbarrett, 2020-09-22:
IMO the makefiles are a conceptual dead end and I don't think we intend on seriously expanding the functionality before first replacing it with a Python-based solution or some third-party project automation software like VUnit or edalize. The best thing to do right now is to copy and modify the makefiles to your liking.

TC01, 2020-09-22:
That's fair. Copying Makefile.questa is actually what has done at the moment, but then if/when that file changes due to a cocotb upgrade (as was recently the case from 1.3.1 to 1.4) that needs to be updated, so I was wondering if there was some way to improve this upstream.

If 1.4 is likely to be the last major release that uses the makefiles, or at least makes major changes to them, then I'm less worried about future-proofing this (obviously changes will be required when the makefiles go away but that was inevitable anyway). I guess "support extending the sequence of commands ran when launching the simulation" is a useful target feature for the new way of launching simulations, though, whatever that ends up being?

themperek, 2020-09-22:
I expect we will keep makefiles for longer time in any case. When 1.4 was major change in makefiles I do not expect much change anymore to them.

PS.  _Maybe something interesting for you ;-) https://github.com/themperek/cocotb-test/_

themperek, 2020-09-22:
There is also "secrete" feature. You can copy/modify `Makefile.questa` and include instead `Makefile.sim`.

TC01, 2020-09-22:
> There is also "secrete" feature. You can copy/modify `Makefile.questa` and include instead `Makefile.sim`.

Oh, interesting, that hadn't occurred to me! That would probably make things quite a bit simpler, yeah. Looking at what's actually in ```Makefile.sim```, I guess the bare minimum would be to set ```all: sim``` in a project makefile before including the modified version of ```Makefile.questa```?

Actually... I wonder if it would be useful to have an include guard in the simulator makefiles to make overriding them easier? Then we could customize one without needing to drop support for other simulators by just including the modified makefile before ```Makefile.sim```.

themperek, 2020-09-24:
>I guess the bare minimum would be to set all: sim in a project makefile before including the modified version of Makefile.questa?

Looks like. I missed this. Maybe could move this to `Makefile.inc`

>Actually... I wonder if it would be useful to have an include guard in the simulator makefiles to make overriding them easier? Then we could customize one without needing to drop support for other simulators by just including the modified makefile before Makefile.sim.

Maybe.

closed 2021-01-17 by ktbarrett