cocotb/cocotb #2076: `ConstantObject.value` does not return BinaryValue anymore

corna, 2020-09-22:
Starting from ef3c212f21f35caeb1d246079d4521a0ed2ca505, accessing `value` on a `ConstantObject` now returns (in my test) an `int`, while it previously returned a `BinaryValue`.

This breaks [this test in my PR](https://github.com/cocotb/cocotb/pull/1248/files#diff-ff33e1ce7b82e9b297fda32cde23ee41R25-R28), as the returned `int` object does not have a `integer` attribute.

Is it an expected result of ef3c212f21f35caeb1d246079d4521a0ed2ca505?

garmin-mjames, 2020-09-22:
Paging @ktbarrett

There was also previous discussion on deprecating `ConstantObject` and instead providing some const property on the other handle classes (https://github.com/cocotb/cocotb/pull/1949#discussion_r451944598).

ktbarrett, 2020-09-22:
Given the fact that `ADDR_WIDTH` in your test is an integer parameter, returning an integer is expected. I would consider the old behavior a bug and I'm surprised that it didn't break more simulators (calling `vpi_get_value` with `vpiBinStrVal` on an integer parameter seems like it's UB, but maybe most simulators handle this as a QoL thing).

corna, 2020-09-22:
OK thanks, I will update the code.
Is there some doc regarding the return type of `value` for the various objects?

ktbarrett, 2020-09-22:
No, there currently isn't, but there should be.

High-level there is:
* logic and bit vector like objects return `BinaryValue`
* integer objects return `int`
* real objects return `float`
* string objects return `bytes`
* multi-dimensional vectors of logic or bits return arrays of `BinaryValue` (@garmin-mjames added this recently)
* other hierarchy objects don't implement `.value` and will raise an exception

The `.value` types should be the same for both `ConstantObject` and the subclasses of `ModifiableObject`.

corna, 2020-09-23:
Thanks @ktbarrett, this helps a lot.

closed 2020-09-23 by corna