cocotb/cocotb #2084: Cocotb failed to find python shared library macOS Catalina python3 homebrew

cysys, 2020-09-27:
Hi, 
I am facing this issue where the python shared library is not being found. 

The errors shown:
`-.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib libpython3.8.so: dlopen(libpython3.8.so, 9): image not found`

` -.--ns ERROR    cocotb.gpi                         ..mbed/gpi_embed.cpp:127  in embed_init_python               Failed to find Python shared library`

Could you please tell me how to fix these ?

cmarqu, 2020-09-27:
Is there a virtual environment involved?
Which cocotb version is this, and how did you install it?

cysys, 2020-09-28:
Version 1.4

I tried with and without virtual environment

themperek, 2020-09-28:
Maybe @mtikekar has an idea?

mtikekar, 2020-09-28:
Hi @cysys, I see the same error messages but the simulation runs to completion regardless.

```
     -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib libpython3.8.so: dlopen(libpython3.8.so, 9): image not found

     -.--ns ERROR    cocotb.gpi                         ..mbed/gpi_embed.cpp:127  in embed_init_python               Failed to find Python shared library

     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb                                      __init__.py:208  in _initialise_testbench           Running tests with cocotb v1.4.0 from /Users/mtikekar/Library/Python/3.8/lib/python/site-packages/cocotb
 ```

If your simulation isn't running, maybe there is an error later on that is related to your code? You can check if your installation is fine by running one of the examples in this repo. This is what I ran:

```
git clone https://github.com/cocotb/cocotb
cd cocotb/examples/dff/tests
make
```

cysys, 2020-09-28:
Hi @mtikekar , the examples seem to be working fine.

Is it safe to ignore these errors? I would be using cocotb for building out my final college project.

ktbarrett, 2020-09-28:
@cysys That error always occurs on MacOS. It is safe to ignore. Is it the only error you are seeing?

cysys, 2020-09-28:
Hi @ktbarrett, yes thats the only error I am getting when running the examples.

ktbarrett, 2020-09-28:
@cysys Don't worry about it. It's an artifact of how libpython is being loaded on different platforms. We try loading it as a dependency as well as runtime loading it using dlopen. The first works, but the second fails on MacOS, which is why you are seeing that error. If cocotb tests run, then libpython was successfully loaded, meaning the error is null.

Have fun with your project! If you need any help feel free to ask on the [gitter](https://gitter.im/cocotb/Lobby).

closed 2020-09-28 by ktbarrett