cocotb/cocotb #2088: Increase C++ code coverage

cmarqu, 2020-09-29:
Coverage of the C++ code should be achievable with `gcov`, as pointed out by @cmarqu on gitter.
I got something rudimentary using `python setup.py develop` with conda/GCC, using these extra args for compiling and linking:
`["--coverage", "-fprofile-arcs", "-ftest-coverage"]`.

The libraries are built in place so the `*.gcno` and `*.gcda` files show up in the repo, and after running some of the tests I was able to generate some `*.gcov` files with coverage results.

I'm not sure how any of this would translate to our CI builds though. codecov supports C++11: https://docs.codecov.io/docs/supported-languages

_Originally posted by @garmin-mjames in https://github.com/cocotb/cocotb/issues/1962#issuecomment-694673986_

cmarqu, 2020-09-29:
In https://github.com/cocotb/cocotb/issues/1962#issuecomment-695146917, @ktbarrett noticed that VPI (and VHPI) coverage is not collected. GPI and ``cocotb_utils`` etc. are covered.

This appears to be happening because we compile the VPI and VHPI files several times for the different simulators, and gcov is quite finicky about the internally-recorded timestamps of its ``*.gcno`` (the "notes files", generated during compilation) and ``*.gcda`` (the "data files", updated during code execution) files. https://github.com/numpy/numpy/issues/11790 provided the crucial hint.
gcov reports "stamp mismatch with notes file" (a hard error) for the VPI/VHPI files, but for me only when dumping the intermediate information with ``gcov -i <vpi gcda file>``.
I hacked around this problem by removing compilation for all simulators except my own in ``cocotb_build_libs.py``.

Some related info:
* gcovr (http://gcovr.com/) is a nice pip-installable tool similar to lcov. 
* ``gcov-dump`` can show the raw content of gcno and gcda files.
* ``gcov-tool`` can merge multiple gcda files.
* some gcov versions (e.g. the one from gcc 7.3.0) did not work at all for me, saying that there were no executable lines in those VPI and VHPI files.
* the gcov version must match the gcc version the files were compiled with.

cmarqu, 2020-09-29:
The question is now, how do we best handle this in CI? One way would be to enhance ``cocotb_build_libs.py`` to allow building just a single simulator.

ktbarrett, 2020-09-29:
> One way would be to enhance cocotb_build_libs.py to allow building just a single simulator

Or a limited set. I would like this, building all the simulator interfaces takes a while on my old thinkpad and I only use a couple simulators anyway. Flags could be used to control the build process (or just reuse `SIM`) to only build certain simulators, or just one for coverage.

That being said. I'm wondering if there is some other way that plays with the timestamps. Why are the timestamps so important to GCOV?

cmarqu, 2020-09-29:
> Why are the timestamps so important to GCOV?

Here is some rationale: https://github.com/gcc-mirror/gcc/blob/590480417bc16d5710dacdb459bd4c3a509b89ba/gcc/gcov-io.h#L62-L66

ktbarrett, 2020-09-29:
So let me get this right. The .gcno file is generated by GCC during compilation of one instance of the VPI library. It writes out VpiImpl.cpp.gcno. The compilation occurs for libcocotbvpi_modelsim.so with the same tag as was written to VpiImpl.cpp.gcno. Later, VPI is compiled again, overwriting VpiImpl.cpp.gcno with a new version with a different tag, and it finishes compiling libcocotbvpi_icarus.so. So when Modelsim is run, the tag doesn't match up to the tag in VpiImpl.cpp.gcno?

garmin-mjames, 2020-09-29:
This may be relevant: https://gcc.gnu.org/onlinedocs/gcc/Cross-profiling.html#Cross-profiling

ktbarrett, 2020-09-30:
I'm more convinced of reusing `SIM` since we use it to select a simulator with tox.

cmarqu, 2020-10-01:
The solution with ccache mentioned here may help: https://gcovr.com/en/stable/cookbook.html#how-to-collect-coverage-for-c-extensions-in-python

ktbarrett, 2020-11-24:
@cmarqu Can this be closed?

closed 2020-11-24 by cmarqu