cocotb/cocotb #2089: Make variables as target dependencies

garmin-mjames, 2020-09-29:
You've hit a problem in make: arguments passed to Make are not considered a dependency by default. We have many cases where this is problematic, the most annoying example (I find) is a change of `SIM`, which breaks in interesting ways. 

There are solutions out there which are reasonably widely used, e.g. https://cr.yp.to/redo/honest-script.html.  

I'd prefer a solution which doesn't only target Active-HDL, but eliminates the problem for all the other simulators at the same time, by properly tracking the necessary make variables as dependency.

Even though it would be nice, I don't think we need an efficient solution, but we should aim for a correct solution. So running `make clean` whenever an argument changes would be "brutal", but sufficient IMO.

_Originally posted by @imphil in https://github.com/cocotb/cocotb/pull/1873#issuecomment-636468661_

---

When variables change between invocations of `make`, the targets don't take account of this.
So switching `TOPLEVEL_LANG=verilog` to `TOPLEVEL_LANG=vhdl` won't do what you expect (such as update the `tcl` script to compile the VHDL sources)

ktbarrett, 2020-09-29:
One thing to note is that the way `combine_results.xml` works is means you have to clean all the `result.xml`s when switching `TOPLEVEL_LANG` since some tests only run on one language and you don't want them counted in the merged results.

garmin-mjames, 2020-10-05:
One way to implement this might be:
1. Always create a file `_new_vars` with the values of all the dependency variables printed in it.
2. If `_vars` doesn't exist or the contents are different than `_new_vars`, overwrite `_vars` with `_new_vars`
3. Make `_vars` a dependency of the target that should be re-run

ktbarrett, 2020-10-06:
Is it more reasonable to instead replace the Makefiles with Python? That's been on the docket for a while now #1562. From the user survey, people are very unhappy with the Makefiles. Even with this issue solved, the Makefiles are still not advanced enough for a lot of people's use cases. And in general, Makefiles aren't very ergonomic.

It instead might be better to provide a simple set of abstractions in Python, so users can use alternative project automation software (which is yet to exist), or build it themselves. At that point we can refactor our Makefiles to use those abstractions and move it out of the main repo as an optional dependency. At that point it can be developed separately without warrant of scope. We would likely still depend upon it for regressions.

garmin-mjames, 2020-10-06:
@ktbarrett Yeah, I almost didn't create this issue as I see the Makefiles as a dead end, or at the very least have their scope of functionality restricted. I only use them for regression tests locally, as we have a Python/Tcl/batch script process for running tests at work.