cocotb/cocotb #2093: Support for SystemVerilog Assertions (SVA)

gtaylormb, 2020-10-01:
Cocotb seems to have very limited support for SVA at this point. If I create an assertion:

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
    );
```

The simulator would normally log the hierarchical name of the scope in which the assertion failed as well as the ``__FILE__ `` and `` __LINE__ `` to give you context. Cocotb gives you:

`# 101990.00ns ERROR    *** NO MESSAGE PROVIDED ***`

No context for the assertion failure and the test passes.

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else
      $error("%m in %s at line %0d", `__FILE__, `__LINE__); 
```

Cocotb gives you:

`# 101990.00ns ERROR    %m in %s at line %0d`

$error doesn't handle formatting input (as reported in #130), test still passes.

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else begin
      $error("Assertion failure");
      $display("%m in %s at line %0d", `__FILE__, `__LINE__);      
    end
```

Cocotb gives you:

```
# 102340.00ns ERROR    Assertion failure
# instance.ERROR_condition in path/module.sv at line 176
```
Now we've hacked our way to getting the assertion context but the test still passes.

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else begin
      $display("%m in %s at line %0d", `__FILE__, `__LINE__);      
      $fatal("Assertion failure");
    end
```

This crashes the whole simulator (I'm using Questa), so the test will fail in a regression suite. Unfortunately if you want to use the GUI to debug the assertion failure you're out of luck.

The main 2 issues I see are: 

- Cocotb does not log any context for errors during simulation, and 

- errors during simulation do not cause the test to fail (possibly reported in #911)

ktbarrett, 2020-10-01:
We are willing to accept any recommendations or changes to better support SVA's. There was another user interested in it (#2043).  I have no idea what the requirements are, or how assertions are instrumented. Please advise.

garmin-mjames, 2020-10-01:
IEEE 1800-2012 Clause 39 has details on assertion API, including VPI callbacks you can register.
40.5 has details on assertion coverage.

jjj11x, 2020-10-03:
> The main 2 issues I see are:
> 
> * ...
> * errors during simulation do not cause the test to fail (possibly reported in #911)

I think #911 might be a separate issue, since those are python assertions, not systemverilog assertions. Systemverilog assertions and `$error()` not causing the test to fail is a pretty serious bug (or maybe just oversight) I've run into as well.

jmerdich, 2022-07-30:
I haven't seen any progress here for a long while, but ignoring (systemverilog) asserts seems like a critical bug, and I'm certainly not comfortable switching to cocotb while that's broken.

I'm new to this code and seeing what it will take to fix this. In the meantime here's a failing test: https://github.com/jmerdich/cocotb/tree/jmerdich/sva. Fails on icarus, passes on ghdl, so it'd be interesting to see what other sims do for this. I have *some* access to others, but not easy access.