cocotb/cocotb #2094: Introduce new system tasks to end the test with pass or fail

ktbarrett, 2020-10-01:
Following the issues in #2093 and discussion #1722, I think we need to introduce two new system tasks, whose names _don't_ overload builtin system tasks and whose purpose is to end the currently running cocotb test with either a pass or fail. Example names could be `$cocotb_pass` and `$cocotb_fail`.

This can be added onto the current system of system task declarations in VpiImpl.cpp. They would issue `gpi_sim_event` with appropriate error codes and messages that would be fed up to the regression manager / scheduler.

gtaylormb, 2020-10-01:
Additionally, it would be useful if these behaved similarly to the SystemVerilog severity system tasks, from IEEE Std 1800-2012 20.20:

> Each of the severity system tasks can include optional user-defined information to be reported. The userdefined message shall use the same syntax as the $display system task (see 21.2.1) and thus can include
> any number of arguments.
> All of the severity system tasks shall print a tool-specific message, indicating the severity of the exception
> condition and specific information about the condition, which shall include the following information:
> — The file name and line number of the severity system task call. The file name and line number shall
> be the same as `__FILE__ and `__LINE__ compiler directives, respectively.
> — The hierarchical name of the scope in which the severity system task call is made.
> — For simulation tools, the simulation run time at which the severity system task is called.
> The tool-specific message shall include the user-defined message if specified.

ktbarrett, 2020-10-01:
@gtaylormb I recommended removing cocotb's severity system task overloads instead (#2095) and allowing the simulator to handle severity reporting as it normally would. The only thing the user would have to do to make the cocotb test fail is issue `$cocotb_fail` in an else statement, like you were originally doing with `$fatal` (but it won't crash the sim).

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else
      $cocotb_fail(); 
```

Would that work for you?

gtaylormb, 2020-10-02:
Yes I think that would work. If $cocotb_fail can report context where the error occurred in addition to failing the test, all the better. Otherwise I could do something like this:

```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else begin
      $error();
      $cocotb_fail(); 
    end
```

which in theory would give the context from the default simulator $error implementation and then fail the test. Probably have to wrap $cocotb_fail in a `define to allow the RTL to be parsed outside of the cocotb environment.

garmin-mjames, 2020-11-13:
We may be able to catch `$error()` with the `cbError` callback.

eric-wieser, 2020-11-16:
Perhaps exposing system tasks to communicate with the python logger would be useful too.

IMO, there is little value to a "end test with pass" system call - we didn't have one before, and none of the examples in this thread suggest one would be useful either.

ktbarrett, 2020-11-16:
@eric-wieser There has been interest in using cocotb as a slave to HDL testbenches. The HDL testbenches would judge pass/fail and end the cocotb "test". A task that ends a test with a pass is necessary for that, otherwise the responsibility for determining test end is split between the two, which sounds like a nightmare to me. But maybe only failure is necessary assuming the test writer just ignores `results.xml`?

eric-wieser, 2020-11-16:
It's not clear to me what the point of cocotb is in that picture - is any python code running at all?

ktbarrett, 2020-11-16:
Yes. A cocotb test could set up stimulus, monitor and do checking against Python or MATLAB models, mocking out un-implemented firmware with a model, basically anything cocotb can currently do. But instead of the cocotb test determining test end conditions and judging pass/fail, that comes from the HDL. People like SystemVerilog's functional coverage and assertions and can use those for test end and pass/fail, in concert with cocotb as a general-purpose Python cosimulation environment. Running Python or MATLAB models and mocking firmware with a software model is kind of hard in an HDL.

But you are correct this has eclipsed the original issue of getting cocotb to work with SV assertions at a minimum.

ktbarrett, 2020-11-16:
A cocotb test in this case would spin up a bunch of tasks and then sit in an infinite `Timer` wait, or maybe do heartbeat logs, etc.

garmin-mjames, 2020-11-17:
A test like that could currently use (ugly) `expect_error=SimFailure` to pass.

ktbarrett, 2025-02-21:
Registering system tasks is not fun. I don't think we should do this anymore.

closed 2025-02-21 by ktbarrett