cocotb/cocotb #2095: Deprecate and remove the current set of system tasks

ktbarrett, 2020-10-01:
The currently registered system tasks in verilog code are `$info`, `$warning`, `$error`, and `$fatal`. These system tasks are also defined by the Verilog standard. Many simulators don't allow you to overload certain system tasks.

From Icarus [README](https://github.com/steveicarus/iverilog/blob/c4883da334344da016e5af9cc5ce58e2f02d9582/README.txt#L414-L415)
> 
    Builtin system functions

	Certain of the system functions have well-defined meanings, so
	can theoretically be evaluated at compile-time, instead of
	using runtime VPI code. Doing so means that VPI cannot
	override the definitions of functions handled in this
	manner. On the other hand, this makes them synthesizable, and
	also allows for more aggressive constant propagation. The
	functions handled in this manner are:

		$bits
		$signed
		$sizeof
		$unsigned

	Implementations of these system functions in VPI modules will
	be ignored.

Aldec simulators don't like it (from [here](https://github.com/cocotb/cocotb/issues/1722#issuecomment-702362301))
> The Aldec simulators ignore the cocotb overloading and use the built-in ones.

And Mentor doesn't allow overloading `$fatal` (from [here](https://github.com/cocotb/cocotb/issues/2093#issue-713085580).

> This crashes the whole simulator (I'm using Questa), so the test will fail in a regression suite.
```
  ERROR_condition:
    assert property (@(posedge clk) 
      condition
      )
    else begin
      $display("%m in %s at line %0d", `__FILE__, `__LINE__);      
      $fatal("Assertion failure");
    end
```

While `$fatal` is designed to fail the current test, the rest existed solely to overload logging... And does so poorly (#130). The overloaded implementations also get in the way of the use of the system tasks for SVA (#2093).

These system task overloads should be removed.

garmin-mjames, 2020-10-01:
You can see the Aldec behavior in the logs in #1859.

cmarqu, 2020-10-09:
One more datapoint about the behaviour of VCS:
https://github.com/cocotb/cocotb/blob/744dd2f46cec7e6619da3117ffc0fd254934753b/tests/designs/close_module/Makefile#L52

PR #2133 (2020-10-23): Remove the builtin system task overloads for VPI
Closes #130. Closes #2095.

These system tasks overload tasks defined by in the Verilog LRM. This
makes many simulators very unhappy; they do not allow you to overload
some or none of the functions and either do not work, or protest
(#2095). These system tasks never properly parsed additional arguments
correctly (#130, #2093). Also they were never supported with either the
VHPI or FLI interfaces. Their main purpose was to hook HDL reporting
into the Python logger. Since the overload is global, this is fairly
dubious design decision to begin with [... cut]
closed 2020-10-23 by ktbarrett