cocotb/cocotb #2096: Simulation ends although it should wait 1000 Cycles

markusdd, 2020-10-04:
I'm currently having a weird issue with cocotb 1.4 and verilator 4.100

I'm verifying an asynchronous FIFO design and have a basic 'manual, section which works perfectly, where I test reads and writes and the empty/full flags etc. then, I tried to do this to generate some more stress:

```python
    # randomize
    cocotb.log.info(f"Test random read/write.")
    cocotb.fork(write_fifo(dut, vals))
    cocotb.fork(read_fifo(dut, vals))
    await ClockCycles(dut.clk_r_i, 1000)
```

So I am launching 2 coroutines, one of which is writing, the other one is reading and checking, both with random delays.

The coroutines look like this:

```python
async def write_fifo(dut, vals):
    w_cnt = 0
    while True:
        cocotb.log.info(f"In write proc.")
        for _ in range(random.randrange(1, 32)):
            await FallingEdge(dut.clk_w_i)
        if dut.full_o == 0:
            cocotb.log.info(f"Random write number {w_cnt}.")
            dut.wen_i <= 1
            dut.wdata_i <= vals[w_cnt]
            await RisingEdge(dut.clk_w_i)
            dut.wen_i <= 0
            w_cnt = w_cnt + 1

async def read_fifo(dut, vals):
    r_cnt = 0
    while True:
        cocotb.log.info(f"In read proc.")
        for _ in range(random.randrange(0, 12)):
            await FallingEdge(dut.clk_r_i)
        if dut.empty_o == 0:
            cocotb.log.info(f"Random read number {r_cnt}.")
            dut.ren_i <= 1
            await RisingEdge(dut.clk_r_i)
            dut.ren_i <= 0
            await FallingEdge(dut.clk_r_i)
            assert dut.rdata_o == vals[r_cnt], f"Read Data mismatch at value {r_cnt}, got {hex(int(dut.rdata_o))} but expected {hex(int(vals[r_cnt]))}"
            r_cnt = r_cnt + 1
```

The weird thing is, the simulation does not wait for the 1000 cycles of the read clock, it just ends:

```
0.00ns INFO     Running test 1/2: test_afifo
     0.00ns INFO     Starting test: "test_afifo"
                     Description:  Test that data is being passed across clock domains.
     0.00ns INFO     Reset AFIFO on both sides.
     5.00ns INFO     Check empty and not full.
     5.00ns INFO     Fill FIFO with random values.
   285.00ns INFO     Test write error flag.
   305.00ns INFO     Read back previously written data.
  1535.50ns INFO     Test read error flag.
  1609.50ns INFO     Test random read/write.
  1609.50ns INFO     In write proc.
  1609.50ns INFO     In read proc.
  1625.00ns INFO     Random write number 0.
  1628.00ns INFO     Test Passed: test_afifo
```

You can see from the log that both coroutines do start and that the first write is apparently attempted.
I can also confirm that from the waveform, the wen_i signal goes high, but everything ends before the RisingEdge of the write clock clk_w_i does occur.

Either I am overlooking something very stupid or something is wrong here.
In my understanding, even if the coroutines were garbage, shouldn't at least the waiting for the 1000 clocks work?

ktbarrett, 2020-10-04:
Could you post the full code?

Side note. Instead of waiting some arbitrary amount of time in the main coroutine, wait until both tasks are finished using `Combine`.

```python
    # randomize
    cocotb.log.info(f"Test random read/write.")
    writer = cocotb.fork(write_fifo(dut, vals))
    reader = cocotb.fork(read_fifo(dut, vals))
    await Combine(writer, reader)
```

markusdd, 2020-10-04:
Hi, yes I know I can use Combine, just treat this arbitrary wait as a temporary test for now, I just wanted to see if my coroutine setup works, because if you look at them you can see that they do not finish by themselves.

Here is the full testcode, everything until the "randomize" comment works perfectly:

```python
import random
import cocotb
import subprocess
from cocotb.clock import Clock
from cocotb.triggers import FallingEdge
from cocotb.triggers import RisingEdge
from cocotb.triggers import ClockCycles
from cocotb.triggers import Timer
from cocotb.result import TestError

import asyncio

async def write_fifo(dut, vals):
    w_cnt = 0
    while True:
        cocotb.log.info(f"In write proc.")
        for _ in range(random.randrange(1, 32)):
            await FallingEdge(dut.clk_w_i)
        if dut.full_o == 0:
            cocotb.log.info(f"Random write number {w_cnt}.")
            dut.wen_i <= 1
            dut.wdata_i <= vals[w_cnt]
            await RisingEdge(dut.clk_w_i)
            dut.wen_i <= 0
            w_cnt = w_cnt + 1

async def read_fifo(dut, vals):
    r_cnt = 0
    while True:
        cocotb.log.info(f"In read proc.")
        for _ in range(random.randrange(0, 12)):
            await FallingEdge(dut.clk_r_i)
        if dut.empty_o == 0:
            cocotb.log.info(f"Random read number {r_cnt}.")
            dut.ren_i <= 1
            await RisingEdge(dut.clk_r_i)
            dut.ren_i <= 0
            await FallingEdge(dut.clk_r_i)
            assert dut.rdata_o == vals[r_cnt], f"Read Data mismatch at value {r_cnt}, got {hex(int(dut.rdata_o))} but expected {hex(int(vals[r_cnt]))}"
            r_cnt = r_cnt + 1

@cocotb.test()
async def test_afifo(dut):
    """ Test that data is being passed across clock domains. """

    clk_w = Clock(dut.clk_w_i, 10, units="ns")  # Create a 10ns period clock on port clk
    cocotb.fork(clk_w.start())  # Start the write clock
    clk_r = Clock(dut.clk_r_i, 37, units="ns")  # Create a 37ns period clock on port clk
    cocotb.fork(clk_r.start())  # Start the read clock
    vals = [random.randrange(1, 2**41) for _ in range(2**16)]

    # reset, disable divs
    cocotb.log.info(f"Reset AFIFO on both sides.")
    dut.rst_w_an_i <= 0
    dut.rst_r_an_i <= 0
    dut.cycle_i <= 0
    await FallingEdge(dut.clk_w_i)

    # deassert reset, check non full, but empty
    dut.rst_w_an_i <= 1
    dut.rst_r_an_i <= 1
    cocotb.log.info(f"Check empty and not full.")
    assert dut.empty_o == 1, f"Status empty_o was not asserted after reset."
    assert dut.full_o == 0, f"Status full_o was asserted after reset."
    assert dut.werr_o == 0, f"Status werr_o was asserted after reset."
    assert dut.rerr_o == 0, f"Status rerr_o was asserted after reset."

    # fill
    cocotb.log.info(f"Fill FIFO with random values.")
    for i in range(16):
        await FallingEdge(dut.clk_w_i)
        assert dut.full_o == 0, f"Status full_o was asserted already after {i} samples."
        dut.wen_i <= 1
        dut.wdata_i <= vals[i]
        await RisingEdge(dut.clk_w_i)
        dut.wen_i <= 0
        if i == 0:
            # check after first write that the empty flag goes low on read side (wait for sync)
            for i in range(3):
                await FallingEdge(dut.clk_r_i)
            assert dut.empty_o == 0, f"Status empty_o was still asserted after {i} writes."
            await FallingEdge(dut.clk_w_i)
    await FallingEdge(dut.clk_w_i)
    assert dut.full_o == 1, f"Status full_o was not asserted after filling up AFIFO completely."

    # test write error flag
    cocotb.log.info(f"Test write error flag.")
    dut.wen_i <= 1
    dut.wdata_i <= 0xAFFEAFFE
    await RisingEdge(dut.clk_w_i)
    dut.wen_i <= 0
    await FallingEdge(dut.clk_w_i)
    assert dut.werr_o == 1, f"Status werr_o was not asserted after trying to write to a full AFIFO."
    await FallingEdge(dut.clk_w_i)
    assert dut.werr_o == 0, f"Status werr_o was not deasserted after one cycle."

    # read until empty
    cocotb.log.info(f"Read back previously written data.")
    for i in range(16):
        await FallingEdge(dut.clk_r_i)
        assert dut.empty_o == 0, f"Status empty_o was asserted already after {i} samples."
        dut.ren_i <= 1
        await RisingEdge(dut.clk_r_i)
        dut.ren_i <= 0
        await FallingEdge(dut.clk_r_i)
        assert dut.rdata_o == vals[i], f"Read Data mismatch at read {i}, got {hex(int(dut.rdata_o))} but expected {hex(int(vals[i]))}"
        if i == 0:
            # check after first read that the full flag goes low on write side (wait for sync)
            for i in range(3):
                await FallingEdge(dut.clk_w_i)
            assert dut.full_o == 0, f"Status full_o was still asserted after {i} reads."
            await FallingEdge(dut.clk_r_i)
    await FallingEdge(dut.clk_r_i)
    assert dut.empty_o == 1, f"Status empty_o was not asserted after emptying AFIFO completely."

    # test read error flag
    cocotb.log.info(f"Test read error flag.")
    dut.ren_i <= 1
    await RisingEdge(dut.clk_r_i)
    dut.ren_i <= 0
    await FallingEdge(dut.clk_r_i)
    assert dut.rerr_o == 1, f"Status rerr_o was not asserted after trying to read from an empty AFIFO."
    await FallingEdge(dut.clk_r_i)
    assert dut.rerr_o == 0, f"Status rerr_o was not deasserted after one cycle."

    # randomize
    cocotb.log.info(f"Test random read/write.")
    cocotb.fork(write_fifo(dut, vals))
    cocotb.fork(read_fifo(dut, vals))
    await ClockCycles(dut.clk_r_i, 1000)
```

ktbarrett, 2020-10-04:
So what you are seeing is the read or write tasks not finishing? Have you tried using another simulator like Icarus or a commercial simulator?

There have been a rash of bugs recently when using Verilator; all of them related to VPI callback scheduling. Neither us or the Verilator devs have tracked it down yet.

markusdd, 2020-10-04:
Hi, as I'm doing this as part of a (future) open source effort I do not have a commercial simulator at hand for this.

I can try later if this will work with icarus, although this usually has some issues with systemverilog, so I have to see.

As far as the finishing goes: The read and write processes should not finish. After starting, one polls for not full and tries to write with random gaps. The other one watches for not empty and tries to read with arbitrary gaps.
I can then just wait for an arbitrary amount of clocks in the main test to decide foir how long I want to run this random read/write scenario and then kill the coroutines as I see fit. I know there is a thousand other methods I could use for this.

The requested log is attached. To me this looks that after the forking, simulation time does not advance anymore (stuck at 1628ns) and all read-clock events happen at once.

[log.txt](https://github.com/cocotb/cocotb/files/5323248/log.txt)

markusdd, 2020-10-04:
Also just tried with icarus, just as I expected, it refuses to compile my Systemverilog code, it craps out at elaboration time checks that use $error

and it also cannot deal with always_ff and the like it seems

ktbarrett, 2020-10-04:
Thanks for posting the log.

```
  1628.00ns DEBUG    Scheduling coroutine test_afifo
  1628.00ns DEBUG    Scheduling with Value(RisingEdge(ModifiableObject(svb_afifo_tb.clk_r_i)))
  1628.00ns DEBUG    Coroutine test_afifo yielded RisingEdge(ModifiableObject(svb_afifo_tb.clk_r_i)) (mode 1)
  1628.00ns DEBUG    Scheduled coroutine test_afifo
  1628.00ns DEBUG    All coroutines scheduled, handing control back to simulator
  1628.00ns DEBUG    Returning to simulator
  1628.00ns DEBUG    Returning to Python
  1628.00ns DEBUG    Trigger fired: RisingEdge(ModifiableObject(svb_afifo_tb.clk_r_i))
  1628.00ns DEBUG    1 pending coroutines for event RisingEdge(ModifiableObject(svb_afifo_tb.clk_r_i))
                     	test_afifo
```

This gets repeated 1000 times at 1628.00ns and then the test ends. This is your `ClockCycles` call. It completes the `RisingEdge(clk_r_i)` in that call immediately. This implies there are bugs in the Verilator scheduler. There might be a few ways around this, depending entirely on what the actual problem is, but these kind of problems will be common until the problem is located and fixed in Verilator. It's up to you to decide how you want to proceed. I'd recommend against using SystemVerilog for FOSS projects at this point, but that is just my opinion.

Since you mentioned this is a FOSS effort, could you link the project so I can run tests?

#### Bug Analysis

The scheduler does not segregate newly-scheduled triggers from old triggers. New triggers are added to the same collection of triggers to service, which will be seen later after continued iteration over that collection. The scheduler then runs the callback on that new trigger in the same eval step.

markusdd, 2020-10-04:
Hi, so I see we come to the same conclusion. It just unrolls all clock edge triggers at once without advancing the simulation.Funny enough, this does not seem to be an issue with the clocks themselves, which are also coroutines being forked right at the start.

The project has  not been put out yet because especially with this module I wanted it to be well tested, because there is literally nothing worse than a broken AFIFO being used somewhere.

The decision went with SV for several reasons, mainly I just feel it is 2020 now and I'm sick of using verilog 2001 until eternity. If there are issues with the open source ecosystem this is our chance to iron it out.
Also we use V2001 at the company I work for, so I wish to avoid any conflict of interest by using the exact same language and coding styles.

I will look into sharing this project with a few people here privately so we can advance in debugging this.

ktbarrett, 2020-10-04:
Pinging @wallento.

ktbarrett, 2020-10-04:
@markusdd I don't expect the clock generator to fail since it is using `Timer`. There is semantic implications that it *must* consume time. Signal values *can* change in the same eval step, so not consuming time is possible in their evaluation system.

I *think* bug is that they mistakenly decide to run a callback on an event that has already been processed. They need to keep track of that. The step-based simulation they do is supposedly faster, but HDLs are still event-oriented and they need to make it appear to behave as such.

I understand the frustration with having to write Verilog-2001; it's not a very ergonomic language. I use VHDL at work and probably would at home as well. There are also a multitude of other HDL languages out there: MyHDL, Chisel, SpinalHDL, nmigen, clash, and probably a lot more I'm not familiar with. They are designed to generate Verilog or VHDL that a simulator or synthesizer can easily consume. I'm not telling you shouldn't use SystemVerilog, I'm just saying there are other HDLs out there, so don't feel limited. And of course, these opinions are my own and not cocotb's organizations ;)

markusdd, 2020-10-04:
Ok so I set up a private mirror to guthub here which I shared with you @ktbarrett .
If other members from the core cocotb team need access let me know, I will add them

Instructions are simple, you need verilator, python 3.7+ and cocotb 1.4 in your environment.

Then navigate to test_lib/svb_afifo_tb and run make.

You should see that the test passes (dangerously) although the random part of the test never gets executed properly.

You can also use gtkwave dump.vcd to look at the waves if this helps in any way.

markusdd, 2020-10-04:
> @markusdd I don't expect the clock generator to fail since it is using `Timer`. There is semantic implications that it _must_ consume time. Signal values _can_ change in the same eval step, so not [...]

At my workplace we also use heavy code generation practices to do a lot of the inconvenient V2001 stuff.
As far as the other stuff that is out there goes: I'm not a big fan. Those approaches all very look and feel like a software guy trying to do hardware. Even with VHDL I sometimes feel that way. Verilog and VHDL and their sematics exist for a reason.

That's why I love cocotb so much. It gives you the efficiency of python in the testbench world, and for the hardware you use  the languages which are good for hardware. My 2 cents :D

markusdd, 2020-10-04:
OK, picking up on your Timer statement, you are absolutely correct:

I replaced the 1000 clock cycles wait with this: `await Timer(10000, units="us")`

Now the simulation hangs up completely after forking the coroutines, not even Ctrl+C works, I had to kill it from within htop.

Log:

```
1665.00ns INFO     Random write number 0.
  1665.00ns DEBUG    Coroutine write_fifo yielded RisingEdge(ModifiableObject(svb_afifo_tb.clk_w_i)) (mode 1)
  1665.00ns DEBUG    Scheduled coroutine write_fifo
  1665.00ns DEBUG    1 pending coroutines for event <<Event at 0x10759c7d0>.wait() at 0x107adb050>
                     	Scheduler._do_writes
  1665.00ns DEBUG    Scheduling coroutine Scheduler._do_writes
  1665.00ns DEBUG    Scheduling with Value(<<Event at 0x10759c7d0>.wait() at 0x107adb050>)
  1665.00ns DEBUG    Coroutine Scheduler._do_writes yielded ReadWrite() (mode 1)
  1665.00ns DEBUG    Scheduled coroutine Scheduler._do_writes
  1665.00ns DEBUG    All coroutines scheduled, handing control back to simulator
  1665.00ns DEBUG    Returning to simulator
```

markusdd, 2020-10-05:
Hi, in order to simplify things, I made a small example repo we can use to get to the bottom of this:

https://github.com/markusdd/verilator_coco_debug

markusdd, 2020-10-06:
Given the things you @ktbarrett have written in the verilator thread, one question:

Is it correct to call eval multiple times or just once before the loop?

markusdd, 2020-10-06:
I dug a bit into verilators docs, this section seems relevant:

[image]

garmin-mjames, 2020-10-06:
I looked over IEEE 1800-2012 Section 4 Scheduling semantics, and what I could find of the Verilator `eval` flow.
- https://www.veripool.org/wiki/verilator/Manual-verilator#EVALUATION-LOOP
- https://github.com/verilator/verilator/blob/master/docs/internals.adoc#verilated-flow

According to the Verilator FAQ, it's not a standards compliant sim:

> Most simulators have to be compliant with the complete IEEE 1364 (Verilog) and IEEE 1800 (SystemVerilog) standards, meaning they have to be event driven. This prevents them from being able to reorder blocks and make netlist-style optimizations, which are where most of the gains come from.

> You should not be scared by non-compliance. Your synthesis tool isn't compliant with the whole standard to start with, so your simulator need not be either. Verilator is closer to the synthesis interpretation, so this is a good thing for getting working silicon.

This should be okay for cocotb though, as we only really care about being able to call the callbacks at the right time.
My understanding is that the `eval` function basically runs all of the *iterative regions* until no more changes are seen.

@ktbarrett is correct that the verilator code just adds the new `cbValueChange` callback to the end of the collection of callbacks.
The callback is only supposed to be called if a stored previous value `prevDatap` doesn't match the current value `varDatap`. But these aren't updated until all of the registered callbacks are iterated through, and registering a new callback doesn't update `prevDatap` if it already is set.
https://github.com/verilator/verilator/blob/c5da072ff680dd64eca10f18bd8e97f6a1643073/include/verilated_vpi.cpp#L492-L519
So verilator iterates through and calls all callbacks on the same edge.

As far as freezing on the `Timer`, I think it has to do with the timescale vs a `Timer` of 10000us. If `main_time++;` increments by one each time through the loop, you're going to be waiting a long time for the `Timer`, doing an eval and checking for callbacks each time step.

I think there's a fundamental issue with how we control Verilator because of this. If we have to step through each time step it's potentially slow compared to other simulators that jump to the next time step that has scheduled events. `VerilatedVpi::cbNextDeadline()` gives the next time that a `cbAfterDelay` is scheduled. We might be able to skip to that since Verilator ignores `#delay_value` constructs, and we've settled all values for the time step.

markusdd, 2020-10-06:
I'm not claiming to get it 100% (as I am not a VPI expert, but I understand simulation and synthesis semantics), but your explanation makes sense to me.

garmin-mjames, 2020-10-06:
You could try `main_time = VerilatedVpi::cbNextDeadline()` in the cocotb verilator loop, but if there's no `cbAfterDelay` registered it returns `~(0)` (all bits set).

markusdd, 2020-10-06:
not sure if I get your point, where exactly would you place it? instead of main_time++?

ktbarrett, 2020-10-06:
> This should be okay for cocotb though, as we only really care about being able to call the callbacks at the right time.
My understanding is that the eval function basically runs all of the iterative regions until no more changes are seen.

I'm not sure it's okay. It evaluates all deltas at once, which violates assumptions of expected values of signals after a value change trigger. After a RisingEdge(clk), clock-sensitive processes have not run yet, so sympathetic signals are still at their old values. It effectively means that value change triggers and ReadWrite callbacks can only occur at the same simulation time. 

> I think there's a fundamental issue with how we control Verilator because of this. If we have to step through each time step it's potentially slow compared to other simulators that jump to the next time step that has scheduled events.

This is my experience. Verilator is *much* slower than Icarus, and it's supposed to be much faster.

If you have an idea on how to change the eval loop to improve behavior, don't be shy. I've tried and was unsuccessful, so it won't likely be me.

markusdd, 2020-10-06:
I do not know about their willingness, but my feeling is cocotb and verilator are both very popular projects, that is why I asked in the other thread, if someone from verilator team could check on the cocotb implementation.

I mean you are basically interfaceing using standard VPI, so they should be able to make a recommendation?

garmin-mjames, 2020-10-06:
@ktbarrett 
> I'm not sure it's okay. It evaluates all deltas at once, which violates assumptions of expected values of signals after a value change trigger. After a RisingEdge(clk), clock-sensitive processes [...]

That's true, the synthesis-style compilation means value change triggers happen after everything has settled. It also means the callbacks happen in whatever order they are registered, not necessarily the order things change (ie, you might process a `RisingEdge` trigger for a signal before one on the clock that caused the signal to change.)

It definitely requires a deeper knowledge and careful construction of testbenches. I would restrict my cocotb test to interact only with ports on the top level module.

garmin-mjames, 2020-10-06:
@markusdd  

> not sure if I get your point, where exactly would you place it? instead of main_time++?

Yes I would try replacing `main_time++;` in the cocotb `verilator.cpp` file.

markusdd, 2020-10-06:
Hi, yes this is definetly a 'limitation', although I would claim that any 'sane' synthesizable verilog code should be written in a way that this does not matter.

You will run into trouble when there are non-blocking assignments in combinational logic etc, but this is actually a big no-no.

markusdd, 2020-10-06:
> @markusdd
> 
> > not sure if I get your point, where exactly would you place it? instead of main_time++?
> 
> Yes I would try replacing `main_time++;` in the cocotb `verilator.cpp` file.

Ok, tried that. "Linear" Testbenches keep functioning as designed (I think even better, as the timescale is now correct?) but the bug remians exactly the same for the forked coroutine scenario.

ktbarrett, 2020-10-06:
@garmin-mjames Supporting that is possible (in fact I wrote something like this last night, which seemed "wrong" at the time), but it is different from other simulators. This is kind of a big deal; portability is a major asset and I don't want to confuse users by having triggers mean different things on different simulators. If that truly is the only way we can make this work, so be it. I was hoping we could avoid that... I think we should see what the other maintainers and some of the verilator people think about it and then decide how to move forward. But having something working and not buggy is good regardless 😄.

I write a lot of my stuff to emulate event-based HDL code and this would break a lot of my stuff. For example, I wait on `RisingEdge(clk)` then sample old values and compute new values, just like a process with non-blocking statements would do. I can't do that now, reading sympathetic signals will give me new values and not old values. I could rewrite it to work off that assumption, but it's a rewrite.

markusdd, 2020-10-06:
Got your point @ktbarrett .

I've inherited the habit to usually operate on FallingEdge in the testbench, so directly between rising sample edges.
This would I think also enable your workflow and is also a nice way to yield waveforms that are rather easy to read.

garmin-mjames, 2020-10-07:
@markusdd There is a bug in Verilator with how callbacks are scheduled so I wouldn't expect your example to work until that's fixed.

@ktbarrett I can look at the Verilator loop at some point, but I don't have it installed to test.

markusdd, 2020-10-07:
Thanks so much for diving into this.

Can I rely on you driving this between cocotb and verilator?

Because i have to admit this definetly left my area of expertise now^^

garmin-mjames, 2020-10-07:
There's a description of the problem in the Verilator issue. I probably won't be making a PR for the fix, but there are enough eyeballs on it for something to happen soon. I can probably find some time if nobody takes it on in the near future.

ktbarrett, 2020-10-07:
Opened verilator/verilator#2588.

markusdd, 2020-11-15:
I have locally built a latest master verilator and used it with cocotb 1.4.0

Still the problem persists. So I assume the verilator side fix was not enough?

ktbarrett, 2020-11-15:
I don't think so. The logic in the verilator main loop is not quite right.

markusdd, 2020-11-15:
yep, seems like it.

Timer gets still stuck, waiting for clock cycles rushes through instantly.

garmin-mjames, 2020-11-16:
@ktbarrett Your PR didn't fix it. the **end** iterator moves as things are added to the list.

markusdd, 2020-11-16:
I'm a little confused. How can it in this instance when it is 'const'? Or are we just losing the remaining events here?

garmin-mjames, 2020-11-16:
`std::list::end()` returns an iterator to a place-holder element _past_ the last element of the list. Adding to the end of the list doesn't invalidate the iterator, it just inserts the new element between the last element and the end iterator.

The real solution is to track the last element and break the loop after processing it.

garmin-mjames, 2020-11-16:
This is my fix:
```diff
diff --git a/include/verilated_vpi.cpp b/include/verilated_vpi.cpp
index e1a2977a..371151b4 100644
--- a/include/verilated_vpi.cpp
+++ b/include/verilated_vpi.cpp
@@ -477,16 +477,21 @@ public:
     static bool callCbs(vluint32_t reason) VL_MT_UNSAFE_ONE {
         VpioCbList& cbObjList = s_s.m_cbObjLists[reason];
         bool called = false;
-        const auto end = cbObjList.end();  // prevent looping over newly added elements
-        for (auto it = cbObjList.begin(); it != end;) {
+        if (cbObjList.empty()) { return called; }
+        const auto last = std::prev(cbObjList.end());  // prevent looping over newly added elements
+        bool is_last;
+        for (auto it = cbObjList.begin();;) {
+            is_last = it == last;
             if (VL_UNLIKELY(!*it)) {  // Deleted earlier, cleanup
                 it = cbObjList.erase(it);
+                if (is_last) { break; }
                 continue;
             }
             VerilatedVpioCb* vop = *it++;
             VL_DEBUG_IF_PLI(VL_DBG_MSGF("- vpi: reason_callback %d %p\n", reason, vop););
             (vop->cb_rtnp())(vop->cb_datap());
             called = true;
+            if (is_last) { break; }
         }
         return called;
     }
@@ -496,10 +501,14 @@ public:
         bool called = false;
         typedef std::set<VerilatedVpioVar*> VpioVarSet;
         VpioVarSet update;  // set of objects to update after callbacks
-        const auto end = cbObjList.end();  // prevent looping over newly added elements
-        for (auto it = cbObjList.begin(); it != end;) {
+        if (cbObjList.empty()) { return called; }
+        const auto last = std::prev(cbObjList.end());  // prevent looping over newly added elements
+        bool is_last;
+        for (auto it = cbObjList.begin();;) {
+            is_last = it == last;
             if (VL_UNLIKELY(!*it)) {  // Deleted earlier, cleanup
                 it = cbObjList.erase(it);
+                if (is_last) { break; }
                 continue;
             }
             VerilatedVpioCb* vop = *it++;
@@ -518,6 +527,7 @@ public:
                     called = true;
                 }
             }
+            if (is_last) { break; }
         }
         for (const auto& ip : update) { memcpy(ip->prevDatap(), ip->varDatap(), ip->entSize()); }
         return called;
```
@markusdd your test completes on master (b5ff50addbfe69657125ef6dab507d8f6d8f457f):

```
 37155.00ns INFO     Random write number 202.
 37166.50ns INFO     Random read number 148.
 37203.50ns INFO     Read Data: 01110010011000010101010001010100
 37240.50ns INFO     Random read number 149.
 37277.50ns INFO     Read Data: 01100110110010111011100110001010
 37314.50ns INFO     Random read number 150.
 37351.50ns INFO     Read Data: 11000110011010011001101010101011
 37425.50ns INFO     Random read number 151.
 37462.50ns INFO     Read Data: 10010110001110111001001011110000
 37465.00ns INFO     Random write number 203.
 37481.00ns INFO     Test Passed: test_afifo
 37481.00ns INFO     Passed 1 tests (0 skipped)
 37481.00ns INFO     *****************************************************************************************
                     ** TEST                             PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     *****************************************************************************************
                     ** debug_verilator_coco.test_afifo    PASS        37481.00         10.49      3571.77  **
                     *****************************************************************************************

 37481.00ns INFO     *************************************************************************************
                     **                                 ERRORS : 0                                      **
                     *************************************************************************************
                     **                               SIM TIME : 37481.00 NS                            **
                     **                              REAL TIME : 10.51 S                                **
                     **                        SIM / REAL TIME : 3566.85 NS/S                           **
                     *************************************************************************************

 37481.00ns INFO     Shutting down...
- :0: Verilog $finish
```

markusdd, 2020-11-16:
You mean with the patch you just posted?

Can check locally as well.

garmin-mjames, 2020-11-16:
With that patch to verilator, the test appears to run to completion.

markusdd, 2020-11-16:
Thanks, I will patch locally and run all the tests that I have.

markusdd, 2020-11-16:
Beautiful. Runs like a charm now.

[image]

ktbarrett, 2020-11-16:
I guess it makes sense, kind of insane that was the only issue, the main verilator loop just seems intuitively off.

@garmin-mjames You could make the logic in that loop look more like a conventional loop by keeping what was there, changing `end` to `last`, and changing the loop condition to `std::prev(it) == last`. `std::prev` on a `std::list` or a `std::vector` is O(1).

markusdd, 2020-11-16:
I guess we probably have not ironed out any possible kink, but at least this unblocks the fundamental flaw that starting any parallel prcesses would basically not work.

garmin-mjames, 2020-11-16:
@ktbarrett I don't think that will work, because `last` may have `erase` called on it, invalidating the iterator.
I had something similar to that, updating `last = std::prev(last)` if it was going to be erased, but I couldn't solve the segfault I was getting.

My in-progress verilator loop update is also MUCH faster. The example test runs in ~0.6s vs 10.5s

markusdd, 2020-11-16:
Yep I think that is most likely the actual benefit, because currently we are running slower than icarus, which is very unusual for verilator, which is normally blazing fast.

themperek, 2020-11-16:
@markusdd BTW: From my experience, slow Verilaotr may come from not optimal timescale settings. Timescale should be of the fastest clock period. For example, if the timescale is 1ns/1ns the fastest clock should be 2ns period + https://github.com/cocotb/cocotb/issues/1676

markusdd, 2020-11-16:
> @markusdd BTW: From my experience, slow Verilaotr may come from not optimal timescale settings. Timescale should be of the fastest clock period. For example, if the timescale is 1ns/1ns the fastest clock should be 2ns period + #1676

yes that is obvious because verilator is a cycle-based simulator.
Still we have a few other inefficiencies here like @garmin-james has described.

garmin-mjames, 2020-11-17:
Created verilator/verilator#2644.

ktbarrett, 2020-11-30:
@markusdd Can you confirm your simulation works with the most recent Verilator?

markusdd, 2020-11-30:
Hi, I in fact are  currently working on extending this afifo TB to much more randomization and so far it does look pretty good.

I am starting and stopping the read/write processes and stop/reconfigure/restart the clocks.

I'm running 500 different  clock ratios now and no simulator related issues so far.

For now, I consider this as solved.

Also: I stepped down from 1ps/1ps timescale to 1ns/1ns in verilator compile options and the speed-up is insane.
It is really, really quick, but of course limits you in the rations you can select. 1ns//500ps would be ideal, but verilator does not accept 500ps, although it would be legal. I will raise this as a ticket.

cmarqu, 2020-11-30:
> 1ns//500ps would be ideal, but verilator does not accept 500ps, although it would be legal. I will raise this as a ticket.

Sure it would be legal? IIRC I looked that up a while ago and found it wasn't; but would love to be wrong :)

markusdd, 2020-11-30:
FYI: https://github.com/verilator/verilator/issues/2659

markusdd, 2020-11-30:
> > 1ns//500ps would be ideal, but verilator does not accept 500ps, although it would be legal. I will raise this as a ticket.
> 
> Sure it would be legal? IIRC I looked that up a while ago and found it wasn't; but would love to be wrong :)

I have seen many modules and testbench models that use scales like 10ns/1ns or so. I have not seen any indication it isn't allowed. Also if you think about it, only being able to shift resolution by 3 orders of magnitude at once is quite a limitation.

ktbarrett, 2020-11-30:
@markusdd @garmin-mjames is working on making improvements to the main verilator loop that should drastically improve performance. However, I think the original issue is resolved, so this question can be closed. I would enable notifications for https://github.com/cocotb/cocotb/pull/2105 if you wish to kept up-to-date on verilator performance improvements.

cmarqu, 2020-11-30:
> I have seen many modules and testbench models that use scales like 10ns/1ns or so. I have not seen any indication it isn't allowed. Also if you think about it, only being able to shift resolution by 3 orders of magnitude at once is quite a limitation.

Yes, 1, 10, 100, 1000 are all legal, but 500 is not, that's what I meant.

markusdd, 2020-12-01:
Oooooops. Interesting. I did not know that.

garmin-mjames, 2020-12-04:
Closing this as solved

closed 2020-12-04 by garmin-mjames

UmairKhan345, 2025-05-23:
Bro I would suggest use VCS Simulator.i can do very easily.Brother don't worry. I would understand each and every step. This is my Gmail muhammadumair12345khan@gmail.com.Share asynchronous fifo and related Cocotb test  with me.if you are interested