cocotb/cocotb #2110: SPICE co-simulation for Analog Mixed Signal (AMS)

bluecmd, 2020-10-08:
This is an overall feature-request to make cocotb drive a workbench in a SPICE environment.

Note: This is quite different from what https://docs.cocotb.org/en/latest/examples.html#mixed-signal-analog-digital talks about.
The currently supported way requires the use of a simulator that can simulate e.g. Verilog-AMS. That is useful but also quite restricted in the number of people having access to those kind of simulators.

Instead, this is about interacting with a SPICE simulation straight away using the cocoTB python library.

This can be done in various ways, e.g. using [PySpice](https://pyspice.fabrice-salvaire.fr/releases/v1.4/examples/ngspice-shared/external-source.html) or using a SPICE simulator capable of VPI like Xyce[[1](https://xyce.sandia.gov/)][[2](https://www.osti.gov/servlets/purl/1483152)].

A proposed goal is to be able to take a SPICE netlist and a Python cocoTB testbench and run those together to produce a result. Just like you today can take a cocoTB testbench and a Verilog module and get a result.
Notable differences are that SPICE is of course analog but that will hopefully prove a surmountable problem.

Imagine this simple inverter as the DUT:
[image]

```
**.subckt test out vcc gnd in
*.opin out
*.ipin vcc
*.ipin gnd
*.ipin in
XM1 out in vcc vcc sky130_fd_pr__pfet_01v8 w=1 l=1 m=1
XM2 out in gnd gnd sky130_fd_pr__nfet_01v8 w=1 l=1 m=1
**.ends
.end
```

And this as the testbench:
```python
@cocotb.test()
async def test(dut):
    # Set voltage to logical high
    dut.in <= V(1.8)
    await Timer(2, units='ns')
    # Check that voltage is in the zone of a logical 0
    assert dut.out.value < 1.8/2, "Inverter result is incorrect"
```

Given that

garmin-mjames, 2020-10-08:
Updated application note for Xyce 7.1: https://xyce.sandia.gov/downloads/_assets/documents/AppNote-MixedSignal.pdf

ktbarrett, 2020-10-08:
After further research I found that Xyce does not, in fact, have a VPI interface. And this makes sense, if Verilog supported AMS out of the box, there would be no need for Verilog-AMS extensions. The Verilog-AMS standard did not extend the VPI to support AMS designs. This was seen in the example provided by @cmarqu. The VPI + Xyce paper mentioned uses Xyce in cosimulation with Icarus, which is run with Icarus's VPI. 

Without a VPI, VHPI, or some other standard interface the chance of implementing support is essentially 0%. I am also not certain that the concepts within cocotb map well to an AMS simulation. I think this feature is out of scope.

ktbarrett, 2020-10-08:
What I think *could* be useful, is a way to run Icarus + cocotb + Xyce. This most definitely belongs in an extension module. Changes that cocotb would accept would be reactor/cleanup to better support this method of cosimulation.

garmin-mjames, 2020-10-08:
Someone at Sandia also used GHDL + Xyce, but the project repo is not public: https://www.osti.gov/biblio/1488489-digital-analog-cosimulation-using-cocotb-xyce

ktbarrett, 2020-10-08:
This is yet another good case for supporting arbitrary GPI extensions, which further re-enforces the need to remove the embed library (#1994) and replace it with a service-handler-style interface in the GPI, much like the VPI and VHPI.

ktbarrett, 2021-06-27:
Closing as the original suggestion is out of scope.

closed 2021-06-27 by ktbarrett