cocotb/cocotb #2111: Support remote Python debugging

garmin-mjames, 2020-10-08:
After thinking about this some more, this approach locks cocotb into debuggers that use the [Debug Adapter Protocol](https://microsoft.github.io/debug-adapter-protocol/) such as VSCode or Theia (used by GitPod). There are a variety of ways to do remote debugging:

- VSCode/GitPod uses **debugpy**.
- Pycharm Professional uses **pydevd-pycharm**.
- PyDev in Eclipse uses plain **pydevd**.
- Our [docs](https://docs.cocotb.org/en/stable/troubleshooting.html#python) recommend **remote-pdb**.

This may be a good candidate for cocotb extensions that provide a setuptools [entry point](https://setuptools.readthedocs.io/en/stable/setuptools.html#dynamic-discovery-of-services-and-plugins) that can be selected by `COCOTB_PY_ATTACH`.

So for example, a new extension **cocotbext-debugpy** could provide an entry point:
```py3
setup(
    name = 'cocotbext-debugpy',
    entry_points = {"cocotb.remote_debugger": "debugpy = cocotbext.debugpy:debug_attach"}
    ...
)
```
And the user would select it with `COCOTB_PY_ATTACH=debugpy,localhost:5678`

_Originally posted by @garmin-mjames in https://github.com/cocotb/cocotb/pull/2103#issuecomment-705050127_

---

This issue is to discuss the best approach to support remote debugging.

garmin-mjames, 2020-10-08:
If we go with extensions, I think it makes sense for these to live in the cocotb GitHub organization.

eric-wieser, 2020-10-08:
Do we actually need our entrypoints to be specific to debugging, or would any plugin system with setup/teardown phases do the job?

garmin-mjames, 2020-10-08:
Are you talking about a more general plugin system where cocotb provides hooks that the plugins can use?

ktbarrett, 2020-10-08:
There is #1225. This could fit into a more robust overloadable startup customization, but I haven't thought much about exactly what that looked like.

garmin-mjames, 2020-10-12:
1. Is supporting remote debugging on Python initialization desirable?
2. If so, can we settle on an approach to enable it?

I like the simplicity of a single environment variable, but it may make sense to split it up.
- `COCOTB_PY_DEBUGGER`
- `COCOTB_PY_ATTACH`

ktbarrett, 2020-10-12:
> 1. Is supporting remote debugging on Python initialization desirable?

Yes IMO.

> 2. If so, can we settle on an approach to enable it?

Of course!

> I like the simplicity of a single environment variable, but it may make sense to split it up.
>
> * `COCOTB_PY_DEBUGGER` - **debugpy**, **pydevd-pycharm**, etc
> * `COCOTB_PY_ATTACH` - **host:port** or just **port** if localhost

The number of point solutions and environment variables we are using for those point solutions is already a lot IMO, and adding more is only going to decrease maintainability. I think it's time to step back and abstract the problem. To me the best solution to abstract all the behaviors we want at startup, as well as support user customization points, are "custom entry points" (#1225). To me, these are more of a startup routine. Attaching a Python debugger can be one of those startup routines, and so could other existing aspects of `_initialize_testbench`.

For example, in your PR you have encapsulated the behavior into a function. Now we can do `COCOTB_STARTUP_ROUTINES += ',cocotb:_check_debug_attach'`. The remote debugger initialization could live in this repo, or not. Whether it can exist or not is then divorced from everyone else's approval. I think this is the path forward for new features that must run during initialization. There are also a host of other issues that can be solved using this generalized approach.

garmin-mjames, 2020-10-12:
An alternative approach is to support entry points, such as `cocotb.startup`, and run all of the provided entry point functions, allowing each one to determine it's own behavior through whatever configuration/enablement it uses.

themperek, 2020-10-13:
Not sure if related but could use this? https://docs.pylonsproject.org/projects/pylons-webframework/en/latest/advanced_pylons/entry_points_and_plugins.html

garmin-mjames, 2020-10-14:
@themperek I saw that, as well as some other examples using **entry_points**.

Pros/Cons for the two approaches as I see them:

| Approach | `COCOTB_STARTUP_ROUTINES` | entry_point
| ---------- | -------------------------------- | ---------------
| Pros | Don't need `pip` installed package<br>Can assume functionality is enabled | Install once, always available<br>Plugin/extension can use multiple entry points easily (startup, teardown, event, etc.)
| Cons | Need to explicitly add each function to call<br>May need envars anyway to pass information | Must be installed to register entry point<br>Needs way to enable functionality (envars, plusargs, ?)

eric-wieser, 2020-10-14:
> EDIT: Thre are probably better/newer examples.

My vote would be for `pluggy`, since that's the plugin manager used by `pytest` and `tox`.

eric-wieser, 2020-10-14:
> There is #1225. This could fit into a more robust startup customization, but I haven't thought much about exactly what that looked like.

I think we should treat this as distinct from #1225. In my mind, that issue tracks "I want to run arbitrary code inside a simulation environment with a python GPI, which may or may not be the cocotb regression manager or scheduler".

While that's obviously sufficient to solve custom startup procedures, I think we want our default startup procedure to support a plugin mechanism to allow multiple startup procedures to interoperate (eg coverage, debug, log customization, etc)

eric-wieser, 2020-10-14:
@garmin-mjames: Note in principle that cocotb could support only `entry_points`, and we could provide a `cocotb-plugin-hacks` package that provides an entry point that reads `COCOTB_STARTUP_ROUTINES` (or vice versa)

ktbarrett, 2020-10-14:
@garmin-mjames I had considered syntax like so: `path.to.module1:entry.point(arbitrary argument string that can be parsed by the plugin);path.to.module2:entry_point2()` that can support passing arguments to plugins. It could take something like JSON for structured data, for example. I'm not sure if "Need to explicitly add each function to call" is a negative or just an alternative to auto discovery.

I'm not a fan of auto-discovery in general. Things that aren't immediately visible or explicitly agreed upon changing behavior is a bad default IMO. Especially if the code to be run has arbitrary privileges. @eric-wieser's suggestion of pluggy is a bit better in that regard because it offers a limited API through which to communicate requests to the framework. It is a dependency though; and we would have to add code and documentation for every behavior we wish to support. All to save the user from having to add a single line to their makefiles. I'm for an arbitrary priviledge, but opt-in, solution.

@eric-wieser I don't think these things have to be distinct, `cocotb:_initialize_testbench` can be a startup routine. It could be set by default, or the user can totally overwrite variable to select a different Python entry point. I think this code would exist either in gpi_embed, or a utility Python library that is loaded instead of directly calling `_initialize_testbench`.

ktbarrett, 2020-10-14:
Note, an explicitly enabling plugins which are auto-discovered is cool by me too. The environment variable isn't the only solution.

garmin-mjames, 2020-10-20:
@ktbarrett It gets complicated if `_initialize_testbench` is just another startup routine in a list, because it needs to be called last (as things are currently).

ktbarrett, 2020-10-20:
Well, "typically". And I'm not sure how that's an unavoidable iddue? Assuming we are using env variables. The Makefiles could define the following.

```makefile
COCOTB_STARTUP_ROUTINES := "cocotb:_initialize_testbench();"
COCOTB_PLUGINS := ""  # for users
```

In user code
```makefile
COCOTB_PLUGINS += "my_plugin:main(123, 'example text');"
```

At simulator call site
```makefile
COCOTB_STARTUP_ROUTINES += $(COCOTB_PLUGINS)
export COCOTB_START_ROUTINES
```

Users should use `COCOTB_PLUGINS` and we can ensure `_initialize_testbench` is always executed last. Unless of course you want to override that entirely, in which you set `COCOTB_STARTUP_ROUTINES`.

But this does bring up another issue with the automatic approach. Plugins will be loaded in arbitrary order. If the plugins have side effects on cocotb's environment (that's kind of the whole point), then certain orders _may_ break or not work as intended. And the issue might not be repeatable in others environments (depending on how they are autodiscovered). This might eventually require the ability to specify load order; in which case we have negated the whole point of autodiscovery.