cocotb/cocotb #2117: Cleanup @cocotb.test(expect_error=True)

garmin-mjames, 2020-10-08:
#2107 revealed how `expect_error=True` can mask a problem, because any `Exception` raised by the test will mark it as PASS.

This test passes even though the wrong error is causing the failure:

https://github.com/cocotb/cocotb/blob/89e7bd77b1e9ceb3172b4659e267bc4c42217a0f/tests/test_cases/test_discovery/test_discovery.py#L101-L106

These tests should be cleaned up to use `assert_raises` style or provide the expected `Exception` type.

markusdd, 2020-10-09:
This test is a duplicate or very similar to what is done in test_handle.py

We agreed to merge/move/remove this in PR #2107

markusdd, 2020-10-10:
@garmin-mjames could you please link PR #2107 to this issue, so this is closed when it gets merged? Thanks.

garmin-mjames, 2020-10-10:
@markusdd There are other uses that should be modified. The test from #2107 is mentioned as the motivating example that showed how using `True` instead of the expected `Exception` type can lead to an inadvertent PASS. The point of this issue is to investigate the uses of `expect_error=True` and be more explicit where it makes sense to do so.

markusdd, 2020-10-10:
Ah ok, sorry, I didn't interpret this issue to be a general cleanup task.

ktbarrett, 2020-11-24:
Passing `True` to `except_error` can also be deprecated.

PR #2223 (2020-12-04): Clean up and deprecate passing bools to expect_error option to cocotb.test
Closes #2117. Cleans up all usages of `expect_error=True` option in the test and also adds a `DeprecationWarning` for it's use.
closed 2020-12-04 by ktbarrett