cocotb/cocotb #2121: test_discovery "failure" with recent icarus verilog

risicle, 2020-10-11:
Using icarus verilog revision `d8556e4c86e1465b68bdc8d5ba2056ba95a42dfd` (in Nix we call this `unstable-2020-08-24`) and `cocotb` 0.3.2 on python 3.7, we get test failures of `test_discovery.access_single_bit` and `test_discovery.access_single_bit_assignment`. I say test failures, but these look to actually be unexpected passes:

```
                                                                                                                     Description:
                                                                                                                         Access a single bit in a vector of the DUT

                                                                                                                         Icarus v0.96 doesn't support single bit access to vectors

     0.06ns INFO     cocotb.sample_module                  test_discovery.py:117  in access_single_bit               00000000 = 8 bits
     0.07ns ERROR    cocotb.regression                         regression.py:296  in handle_result                   Test passed but we expected an error: access_single_bit (result was TestSuccess)
     0.07ns INFO     cocotb.regression                         regression.py:344  in execute                         Running test 7/12: access_single_bit_assignment
     0.07ns INFO     ..gle_bit_assignment.0x7ffff6969be0       decorators.py:252  in _advance                        Starting test: "access_single_bit_assignment"
                                                                                                                     Description:
                                                                                                                         Access a single bit in a vector of the DUT using the assignment mechanism

                                                                                                                         Icarus v0.96 doesn't support single bit access to vectors

     0.08ns INFO     cocotb.sample_module                  test_discovery.py:137  in access_single_bit_assignment    00000000 = 8 bits
     0.09ns ERROR    cocotb.regression                         regression.py:296  in handle_result                   Test passed but we expected an error: access_single_bit_assignment (result was TestSuccess)
     0.09ns INFO     cocotb.regression                         regression.py:344  in execute                         Running test 8/12: access_single_bit_erroneous
     0.09ns INFO     ..ngle_bit_erroneous.0x7ffff6969e20       decorators.py:252  in _advance                        Starting test: "access_single_bit_erroneous"
```

Assuming this means "single bit access" now _is_ supported in recent icarus? If so, I guess we (NixOS packagers) can safely ignore this test?

cmarqu, 2020-10-11:
Yes indeed, see #1952.

ktbarrett, 2020-10-13:
Seems this has been resolved.

closed 2020-10-13 by ktbarrett