cocotb/cocotb #2126: Fails to read negative integer values from VHDL testbenches

jwprice100, 2020-10-15:
When reading integer values from a VHDL testbench, negative values are treated as if they are unsigned. Note that reading an integer value does not read a BinaryValue but instead a ConstantObject. I've attached a simple example to reproduce the issue.

[integer_read_fail.zip](https://github.com/cocotb/cocotb/files/5384567/integer_read_fail.zip)

ktbarrett, 2020-10-15:
Discussion starts [here](https://gitter.im/cocotb/Lobby?at=5f87449424a20801a8e8b272). @garmin-mjames notes that newer version of vhpi_user.h distributed by the VASG have changed the typedef to a signed integer, which would prevent the issue.

garmin-mjames, 2020-10-15:
2008 standard:

> ### 22.2.3 vhpiIntT and vhpiLongIntT
> A value of type `vhpiIntT` or `vhpiLongIntT` represents a value of a VHDL integer type. A value of
type `vhpiIntT` shall be represented with at least 32 bits, and a value of type `vhpiLongIntT` shall be
represented with at least 64 bits. The value represented by a given value of either type is the given value,
interpreted as a signed twos-complement binary number.

So regardless of the internal typdef for each simulator, they should all be returning a twos-complement bit pattern that we can turn into a signed 32-bit integer.

eric-wieser, 2020-10-15:
Pasting the code links from gitter:

https://github.com/cocotb/cocotb/blob/6be15a7fea1b0c18a0d394eae031e9e1ccfdf99a/cocotb/handle.py#L497

https://github.com/cocotb/cocotb/blob/6be15a7fea1b0c18a0d394eae031e9e1ccfdf99a/cocotb/share/lib/simulator/simulatormodule.cpp#L618-L623

https://github.com/cocotb/cocotb/blob/6be15a7fea1b0c18a0d394eae031e9e1ccfdf99a/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L781-L793

eric-wieser, 2020-10-15:
Note that this bug will _not_ reproduce on windows, as on windows `long` is the same size as `int32_t`.

garmin-mjames, 2020-10-15:
@eric-wieser The test passes on Windows with 32-bit Conda, Python 3.7.9, Active-HDL 11.1 32-bit.

garmin-mjames, 2020-10-15:
@ktbarrett regarding compilation warning for this:
> #### -Wconversion
> ... Warnings about conversions between signed and unsigned integers are disabled by default in C++ unless `-Wsign-conversion` is explicitly enabled.

When I add `-Wsign-conversion` I get:

```
cocotb\share\lib\vhpi\VhpiCbHdl.cpp: In member function 'virtual long int VhpiSignalObjHdl::get_signal_value_long()':
    cocotb\share\lib\vhpi\VhpiCbHdl.cpp:792:24: warning: conversion to 'long int' from 'vhpiIntT {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
         return value.value.intg;
                            ^
```

ktbarrett, 2020-10-15:
@garmin-mjames That would only show on Windows due to `long` being 32 bits and an implicit case causing the value to change. The error occurs when that is not the case.

There are a couple issues here IMO. One is `long` is probably a bad type to use since it's different depending on the OS. We probably should be specifically calling out `int64_t` as the return type instead. Our minimum supported standards are C99 and C++11, so we can use stdint.h. In that case the warning would never occur and the bug would always occur if `vhpiIntT` is an unsigned type.

PR #2129 (2020-10-19): Fix VHPI cast issue
Closes #2126.

`vhpiIntT` before the 2008 standard was defined as an unsigned integer in
vhpi_user.h. However, the standard states that all integers are 32-bit
integers with two's complement representation.

The issue occurs when `vhpiIntT` is defined as `uint32_t` and `long` is a
signed 64 bit integer. Negative values are represented as large unsigned
values in two's complement. When this value is returned as a long, it is
implicitly converted ("safely") to the larger signed representation.

To prevent this issue we convert the `vhpiIntT` into a 32-bit two's
complement int [... cut]
closed 2020-10-19 by ktbarrett